esp8266/Arduino #2532: deep sleep and REASON_EXT_SYS_RST (6) on wake

jjssoftware, 2016-09-17:
#### Hardware
Hardware:		ESP-12
Core Version:      	v2.3.0 (sdk v1.5.3)

### Description
Is described here:
http://www.esp8266.com/viewtopic.php?p=55248#p55248

and here:
http://www.esp8266.com/viewtopic.php?p=55273#p55273

I'm trying to understand why when the device is in deep sleep and I reset the device with a switch by pulling the RST pin to ground, the rst_reason gets flagged as REASON_DEEP_SLEEP_AWAKE = 5 and not REASON_EXT_SYS_RST = 6. 

The idea here is I'd like to be able to detect a manual reset with REASON_EXT_SYS_RST on wake so code can act accordingly but it seems like a manual reset via the RST pin always results in the rst_reason being flagged as REASON_DEEP_SLEEP_AWAKE = 5 if the device was in deep sleep.

If the device is not in deep sleep and I reset the device with a switch by pulling the RST pin to ground then the rst_reason does get flagged as REASON_EXT_SYS_RST = 6 as expected.

Is this expected behavior? 

Thanks
Joe

igrr, 2016-09-17:
This is the expected behavior. Due to a hardware bug, we can not wake the chip from deep sleep cleanly. We have to abuse external reset circuit to do that (I.e. connect gpio16 and rst). So upon exit from deep sleep, the whole SoC is reset, except for the RTC memory. There is no way of knowing whether that happened because RTC timer caused wake up, or because reset pin was toggled via some other signal. Upon startup (after CPU reset) software checks whether "entered deep sleep" flag in RTC memory is set. If that is the case, software assumes that reset cause is deep sleep wake up. Otherwise, reset cause is external reset.

jjssoftware, 2016-09-17:
Great thanks for confirming good to know. I'll update the http://www.esp8266.com/viewtopic.php?p=55273#p55273 thread linking to this issue and close this down.

I will indeed need to rearrange where my reset switch is positioned so it disconnects power completely so I can get the desired rst_reason behavior I need.

closed 2016-09-17 by jjssoftware