esp8266/Arduino #2535: WiFiClient.connect() does always fails

xetra11, 2016-09-17:
### Basic Infos

I have one ESP(1) opening an AP and running a server on it. The other ESP(2) is running a client on it and is connecting to the AP and after that trying to connect to the host server. The connection always fails and I cannot figure out what the problem is here. Testing the server via netcat and packetsender works fine. Connecting to it via telnet fails because the connections gets closed by the "foreign host".

#### Hardware
Hardware:		?ESP-7?
Core Version:      	?2.3.0?

### Description

Problem description

### Settings in IDE

doesn't matter

### Sketch

Client ESP (2)
```
#include <ESP8266WiFi.h>

const char* ssid = "artex11";
const char* pw = "getmenow11";
const char* host = "192.168.4.1";
const int port = 80;

void setup() {
  Serial.begin(115200);
  WiFi.begin(ssid, pw);
  Serial.println("Connecting to AP");
  while(WiFi.status() != WL_CONNECTED){
    delay(500);
    Serial.print("."); 
   }
  Serial.println(" connected");
}

void loop() {
  WiFiClient client;
  Serial.println("Connecting to host");
  if(!client.connect(host, port)){
    Serial.println("...connection failed!");
    Serial.println("Retrying in 5 seconds...");
    delay(5000);
    return;  
  }
  client.print("A");
  Serial.println("...TCP message fired!");
  delay(2000);
}
```

Server ESP (1)

```
#include <ESP8266WiFi.h>

const char* ssid = "artex11";
const char* pw = "getmenow11";
WiFiServer server(80);
WiFiClient client;

void setup() {
  Serial.begin(115200);
  //setup AP
  WiFi.softAP(ssid, pw);
  Serial.println("AP started at: ");
  Serial.print(WiFi.softAPIP());
  server.begin();
  Serial.println("Server started...");
}

void loop() {
  client = server.available();

  if(client){
    Serial.println("Client connected!");
    Serial.print("HasClient: ");
    Serial.println(server.hasClient());
    char message = client.read();
    Serial.printf("Message: %c\n", message);
  }
}

```

### Debug Messages

none

Andrey-ew, 2016-10-07:
A similar problem

xetra11, 2016-10-07:
@Andrey-ew 
I fixed this by defining the Client as a Station! That was the missing piece:

`void setup_connection(){
  Serial.begin(115200);
  WiFi.mode(WIFI_STA);
  WiFi.begin(SSID, PW);
  Serial.println("Connecting to AP");
  while(WiFi.status() != WL_CONNECTED){
    Serial.print(".");
    load_blinker(200);
  }
  Serial.println(" connected");
}

The server simply opens a SoftAP`

closed 2016-10-07 by xetra11

abhinay92, 2017-12-26:
I have the similar problem, but couldn't resolve it by setting the   WiFi.mode(WIFI_STA); below is my code. My ESP8266 nodemcu development board always fails to connect to the defined host on port 80. Below is my code

#include <ESP8266WiFi.h>

const char* ssid = "Abhinay";
const char* password = "Abhinay92";

const char* host = "http://jsonplaceholder.typicode.com/";

void setup()
{
  Serial.begin(115200);
  Serial.println();

  Serial.printf("Connecting to %s ", ssid);
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED)
  {
    delay(500);
    Serial.print(".");
  }
  Serial.println(" connected");
}

void loop()
{
  WiFiClient client;

  Serial.printf("\n[Connecting to %s ... ", host);
  if (client.connect(host, 80))
  {
    Serial.println("connected]");

    Serial.println("[Sending a request]");
    client.print(String("GET /") + " HTTP/1.1\r\n" +
                 "Host: " + host + "\r\n" +
                 "Connection: close\r\n" +
                 "\r\n"
                );

    Serial.println("[Response:]");
    while (client.connected())
    {
      if (client.available())
      {
        String line = client.readStringUntil('\n');
        Serial.println(line);
      }
    }
    client.stop();
    Serial.println("\n[Disconnected]");
  }
  else
  {
    Serial.println("connection failed!]");
    client.stop();
  }
  delay(5000);
}

please advice !

blubbi321, 2018-01-31:
Had the same problem and could solve it by connecting another power supply on a Wemos D1 mini

raymarkdelpuerto, 2018-11-14:
You have to set both the client and the server.
Set the client to WIFI_STA and the server to WIFI_AP.

CLIENT:
[image]

SERVER:
[image]

m7mdrahil, 2019-02-12:
I have a similar problem. I tried uploading data that is given directly (static). It worked perfectly. Now, I am getting the data using serial communication and parsing it into different fields (a requirement for Thingspeak). Now the same client.connect returns a value of 0 and it doesn't work. Because of this I feel like it may be a power issue. Here is my code.

#include <ThingSpeak.h>
#include <ESP8266WiFi.h>
void setup() {
  // put your setup code here, to run once:
Serial.begin(115200);
connectWifi();
}

void loop()
{
     if(Serial.available()>0)
        {
          buffStr = Serial.readString();
          parseData();
        }
} 
void uploadData()
{  
  
   WiFiClient client;
   int c = client.connect(server, 80);
  Serial.println(c);
   if (client.connect(server, 80)) { // use ip 184.106.153.149 or api.thingspeak.com
   Serial.println("WiFi Client connected ");
   
   String postData = writeAPIKey;
   postData += "&field1=";
   postData += instCurrent;
   postData += "&field2=";
   postData += instVoltage;
   postData += "&field3=";
   postData += instPower;
   postData += "&field4=";
   postData += totalPower;
   postData += "&field5=";
   postData += totalEnergy;
   postData += "&field6=";
   postData += Latitude;
   postData += "&field7=";
   postData += Longitude;
   postData += "\r\n\r\n";
   
   client.print("POST /update HTTP/1.1\n");
   client.print("Host: api.thingspeak.com\n");
   client.print("Connection: close\n");
   client.print("X-THINGSPEAKAPIKEY: " + writeAPIKey + "\n");
   client.print("Content-Type: application/x-www-form-urlencoded\n");
   client.print("Content-Length: ");
   client.print(postData.length());
   client.print("\n\n");
   client.print(postData);
   delay(1000);
   
   }//end if
 client.stop();
}

void connectWifi()
{
  //Serial.print("Connecting to "+*MY_SSID);
  WiFi.mode(WIFI_STA);
  WiFi.begin(MY_SSID, MY_PWD);
  while (WiFi.status() != WL_CONNECTED) {
  delay(1000);
  Serial.print(".");
  }
  
  Serial.println("Connected");  
}//end connect

MattLogue, 2019-04-04:
I have the same issue. I already had WiFi.mode set to station. I tried many hosts, which I verified on my PC telnet application as working, I tried many sketches that implemented WiFiClient. I also tried getting the most current version of WiFiClient. The funny thing is that HTTPClient works fine. I am powering it through the PC USB port, since power is a possible issue...

m7mdrahil, 2019-04-24:
> I have a similar problem. I tried uploading data that is given directly (static). It worked perfectly. Now, I am getting the data using serial communication and parsing it into different fields (a [...]

The problem got solved when I updated the Arduino and did a new install. I know it sounds stupid but it works. And, I started powering it from an external 3V source as well. It could be one of these things.

teeho88, 2020-10-16:
> ### Basic Infos
> I have one ESP(1) opening an AP and running a server on it. The other ESP(2) is running a client on it and is connecting to the AP and after that trying to connect to the host server.
> 
> #### Hardware
> Hardware: ESP-7
> Core Version: 2.3.0
> 
> ### Description
> The conncetion to the AP does succeed but the connection to the host always fails and I cannot figure out what the problem is here. Testing the server via netcat and packetsender works fine. Connecting to it via telnet fails because the connections gets closed by the "foreign host".
> 
> ### Settings in IDE
> doesn't matter
> 
> ### Sketch
> Client ESP (2)
> 
> ```
> #include <ESP8266WiFi.h>
> 
> const char* ssid = "artex11";
> const char* pw = "getmenow11";
> const char* host = "192.168.4.1";
> const int port = 80;
> 
> void setup() {
>   Serial.begin(115200);
>   WiFi.begin(ssid, pw);
>   Serial.println("Connecting to AP");
>   while(WiFi.status() != WL_CONNECTED){
>     delay(500);
>     Serial.print("."); 
>    }
>   Serial.println(" connected");
> }
> 
> void loop() {
>   WiFiClient client;
>   Serial.println("Connecting to host");
>   if(!client.connect(host, port)){
>     Serial.println("...connection failed!");
>     Serial.println("Retrying in 5 seconds...");
>     delay(5000);
>     return;  
>   }
>   client.print("A");
>   Serial.println("...TCP message fired!");
>   delay(2000);
> }
> ```
> 
> Server ESP (1)
> 
> ```
> #include <ESP8266WiFi.h>
> 
> const char* ssid = "artex11";
> const char* pw = "getmenow11";
> WiFiServer server(80);
> WiFiClient client;
> 
> void setup() {
>   Serial.begin(115200);
>   //setup AP
>   WiFi.softAP(ssid, pw);
>   Serial.println("AP started at: ");
>   Serial.print(WiFi.softAPIP());
>   server.begin();
>   Serial.println("Server started...");
> }
> 
> void loop() {
>   client = server.available();
> 
>   if(client){
>     Serial.println("Client connected!");
>     Serial.print("HasClient: ");
>     Serial.println(server.hasClient());
>     char message = client.read();
>     Serial.printf("Message: %c\n", message);
>   }
> }
> ```
> 
> ### Debug Messages
> none

You can check your PC host again.
const char* host = "192.168.4.1"; - it must be your PC private IP (IPV4 Adress).