esp8266/Arduino #2538: SPI ESP8266 slave not working correct.

littleboot, 2016-09-19:
Hello, 
I have problems using the ESP8266 as SPI slave. 
First problem when installing ESP8266 arduino using the board manager. the examples from the SPISlave are not available inside the arduino IDE. (\libraries\SPISlave\examples)

I downloaded this repository and copied the SPIslave library to "Documents\Arduino\libraries\SPISlave"
Now the SPIslave library and examples are available inside the arduino IDE. 
This code compiles without issues and after downloading the code to the ESP8266 all seems oke. but it just doesn't work properly.

The slaves receives data from the master and it can read the question but it fails to send a response. the PISlave.setData("") function doesn't work. (i confirmed using a logic analyser MISO line never changes.

ESP8266 version 2.3.0 installed using board manager
arduino 1.6.11 installed using windows installer
system info: windows 10 home 64bit
Hardware: ESP8266 nodemcu dev board V1 (esp8266 esp12-e).

aj-ptw, 2017-03-23:
I am having this same problem! Did you fix your @littleboot ?

littleboot, 2017-03-23:
I didn't fix it sry but I can't help you

aj-ptw, 2017-03-23:
I think I figured out I need to send a `0x03` to read data not a `0x02` Like when I write data

devyte, 2017-10-12:
@littleboot is this issue still valid with latest git?
@aj-ptw did you get this working?

littleboot, 2017-10-12:
@devyte I did not get it to work, I gave up and used an other host MCU with an additional UART to communicate with the ESP8266. I also don't know if this still is a problem with the latest git. (maybe this weekend I have time to build a test circuit and figure out if i can get it to work, but I can't promise)

aj-ptw, 2017-10-12:
@devyte yes I got this working really really well. 

Checkout openbci/openbci_wifi for spi slave and OpenBCI/openbci_wifi_master for spi master library

devyte, 2017-10-12:
@aj-ptw was there some issue in the spi code, or did it work as-is?

FredrikJ1, 2018-03-14:
I have the same problem.
When I use two esp8266 they fail in the same way as described above:
The slave receives OK, the master gets garbled data. It seams to be one bit off (first bit missing). Problem remains even if different modes where used.
Same esp8266 as slave connecting to an Nano works OK with the exact same source code (INO). 
My conclusion is that there are something with the esp8266 as master.
I plan to redo the test with different modes in slave and master to see if its a timing issue.