esp8266/Arduino #2541: websockets not releasing memory after disconnect

themindfactory, 2016-09-22:
websockets not releasing memory after disconnect from the far end.

when I get a disconnect event I do a .disconnect(number) I do get back around 100 bytes of the 200, if I do not do a .disconnect(number) I lose about 200 bytes per disconnect from the far end.

there is nothing in my websocketevent function that would affect the heap.

#### Hardware
Hardware:			ESP-12
Core Version:      	2.3.0

### Description

websockets not releasing memory after disconnect

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		4MB/1MB
CPU Frequency:		80Mhz
Flash Mode:			dio
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		 nodemcu

### Sketch

```cpp

void webSocketEvent(uint8_t num, WStype_t type, uint8_t * payload, size_t lenght) {
  Serial.printf("WS#%d\n", num);
  webSocketNum = num;
  switch (type) {
    case WStype_DISCONNECTED:
      Serial.println(F("Websockets Disconnected!"));
      webSocket.disconnect(num);
      break;

    case WStype_CONNECTED: 
        IPAddress ip = webSocket.remoteIP(num);
        Serial.printf("Connected from %d.%d.%d.%d\n", ip[0], ip[1], ip[2], ip[3]);
        Serial.printf("Payload1 %s\n", payload);
        Serial.printf("Length %d\n", lenght);
       break;

    case WStype_TEXT:
      Serial.print(F("Payload2 length="));
      Serial.println(lenght);

      // possible long string
      serialSendString((const char*)payload);
  }
}
```

### Debug Messages

```
messages here
```

themindfactory, 2016-09-27:
This is a fairly serious issue, I have thoroughly gone over my code and can not see that it is the reason, can someone play with websocket example and make a break connections to test, I use the chrome extension to test, "Simple Websocket Client" so that would be a good thing to try in chasing this bug.

RichardS

devyte, 2017-10-12:
@themindfactory please try:
- latest git
- PR #3362 (lwip2)
- PR #3215 (sdk 2.1.0)

devyte, 2018-02-28:
Closing due to lack of feedback and age.

closed 2018-02-28 by devyte