esp8266/Arduino #2542: ESP-12E Node MCU will resets automatically at every time

BharathTatikonda, 2016-09-22:
### Basic Infos

#### Hardware
Hardware:			?ESP-12?
Core Version:      	?2.1.0-rc2?
hi, i have started ESP12-E to work with the GSM module, in this when i was running the it should be reset automatically when some messages will receives,... then it reset and restarted from first so the function is not completed,,....so, please suggest me how to reset the watchdog timer,..... when i run the program the error will be like this///.........

Soft WDT reset

ctx: cont 
sp: 3ffefc90 end: 3ffeff00 offset: 01b0

>>>stack>>>
3ffefe50:  00000001 00000001 00000000 402030ea  
3ffefe60:  00000106 00000000 000005dc 00001388  
3ffefe70:  00000005 3ffe87c0 00000004 40202fe8  
3ffefe80:  3ffe8517 3ffeed24 3ffeee1c 00000000  
3ffefe90:  00000005 000005dc 3ffe8ba2 4020313e  
3ffefea0:  4010564a feefeffe 3ffeee1c 3ffeeed0  
3ffefeb0:  3fffdad0 3ffe8ba2 00000000 40202b77  
3ffefec0:  3fffdad0 3ffeebe0 3ffeee1c 40201fd4  
3ffefed0:  feefeffe 00000000 3ffeeec8 4020213d  
3ffefee0:  feefeffe feefeffe feefeffe 4020388c  
3ffefef0:  feefeffe feefeffe 3ffeeee0 40100718  
<<<stack<<<

   so, please suggest me how to reset the watchdog timer in the esp12-e module

### Settings in IDE

Module:  			?Generic ESP8266 Module?
Flash Size: 		?4MB/1MB?
CPU Frequency:		?80Mhz?
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		?OTA / SERIAL?
Reset Method:		?ck / nodemcu?

### Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}
```

### Debug Messages

```
messages here
```

supersjimmie, 2016-09-22:
You are using some unknown library called "GSM_Shield_Esp.h".
It looks like the problem is in there. Did you ask the maker of that library why it doesn't work?

BharathTatikonda, 2016-09-22:
yahh, but it was created based on the requirement. normal gsm library is not working for esp-12e,so i have removed some part(is listening and listen functions) in the library..... and rearranged it

supersjimmie, 2016-09-22:
Can you then trace where in your code it fails?
Just put some "Serial.println("here"); at several places, and replace "here" with something that points to where in the code it is. Then you may see at what point the crash occurs...

Otherwise you can do a stack decode to discover what happens in your code at the stack that is displayed.
[ESP Exception decoder](https://github.com/me-no-dev/EspExceptionDecoder)

BharathTatikonda, 2016-09-22:
Thank u for ur suggestion, i have checked thet one, the program will fails on when the message is reading (the program will stuck on the  " messageIndex = gsm.isSMSunread(); " in this funtion, ) and at some times it stucts at   "gsm.deleteSMS(messageIndex);" in this function....... 

And i want to try with the stack decode....

BharathTatikonda, 2016-09-22:
there no result.............. so could you tell how to disabled the watchdog timer in esp12-e

supersjimmie, 2016-09-22:
Then you should investigate what happens inside ```gsm.isSMSunread();```, which is in the library (we don't know).
You cannot disable the wdt, but in most cases the wdt is triggered because of a "bad" function.
Check that ```gsm.isSMSunread();```, perhaps there is some for/while/... loop in it that takes too long. If so add the ```yield()``` command inside that loop.

We cannot help if we don't know the library.

BharathTatikonda, 2016-09-22:
ok.. i will send the gsm.isSMSunread(); function for this one from the library,........  could u check this one.......
bool GPRS::readSMS(int messageIndex, char *message, int length, char *phone, char *datetime)  
{
  /* Response is like:
  AT+CMGR=2
  
  +CMGR: "REC READ","XXXXXXXXXXX","","14/10/09,17:30:17+08"
  SMS text here
  
  So we need (more or lees), 80 chars plus expected message length in buffer. CAUTION FREE MEMORY
  */

    int i = 0;
    char gprsBuffer[80 + length];
    //char cmd[16];
	char num[4];
    char *p,*p2,*s;
    
    sim900_check_with_cmd("AT+CMGF=1\r\n","OK\r\n",CMD);
    delay(1000);
	//sprintf(cmd,"AT+CMGR=%d\r\n",messageIndex);
    //sim900_send_cmd(cmd);
	sim900_send_cmd("AT+CMGR=");
	itoa(messageIndex, num, 10);
	sim900_send_cmd(num);
	sim900_send_cmd("\r\n");
    sim900_clean_buffer(gprsBuffer,sizeof(gprsBuffer));
    sim900_read_buffer(gprsBuffer,sizeof(gprsBuffer));
      
    if(NULL != ( s = strstr(gprsBuffer,"+CMGR:"))){
        // Extract phone number string
        p = strstr(s,",");
        p2 = p + 2; //We are in the first phone number character
        p = strstr((char *)(p2), "\"");
        if (NULL != p) {
            i = 0;
            while (p2 < p) {
                phone[i++] = *(p2++);
            }
            phone[i] = '\0';            
        }
        // Extract date time string
        p = strstr((char *)(p2),",");
        p2 = p + 1; 
        p = strstr((char *)(p2), ","); 
        p2 = p + 2; //We are in the first date time character
        p = strstr((char *)(p2), "\"");
        if (NULL != p) {
            i = 0;
            while (p2 < p) {
                datetime[i++] = *(p2++);
            }
            datetime[i] = '\0';
        }        
        if(NULL != ( s = strstr(s,"\r\n"))){
            i = 0;
            p = s + 2;
            while((*p != '\r')&&(i < length-1)) {
                message[i++] = *(p++);
            }
            message[i] = '\0';
        }
        return true;
    }
    return false;    
}

bool GPRS::readSMS(int messageIndex, char *message,int length)
{
    int i = 0;
    char gprsBuffer[100];
    //char cmd[16];
	char num[4];
    char *p,*s;
    
    sim900_check_with_cmd("AT+CMGF=1\r\n","OK\r\n",CMD);
    delay(1000);
	sim900_send_cmd("AT+CMGR=");
	itoa(messageIndex, num, 10);
	sim900_send_cmd(num);
	sim900_send_cmd("\r\n");
//  sprintf(cmd,"AT+CMGR=%d\r\n",messageIndex);
//    sim900_send_cmd(cmd);
    sim900_clean_buffer(gprsBuffer,sizeof(gprsBuffer));
    sim900_read_buffer(gprsBuffer,sizeof(gprsBuffer),DEFAULT_TIMEOUT);
    if(NULL != ( s = strstr(gprsBuffer,"+CMGR:"))){
        if(NULL != ( s = strstr(s,"\r\n"))){
            p = s + 2;
            while((*p != '\r')&&(i < length-1)) {
                message[i++] = *(p++);
            }
            message[i] = '\0';
            return true;
        }
    }
    return false;   
}

BharathTatikonda, 2016-09-22:
but how can we control the with the reset function, until the process would be completed,....... same program will works on the arduino's modules. But i cant identify the problem in that.. i have stucked in this small logic upto one week... but there is no use,.

supersjimmie, 2016-09-22:
I see several (4?) ```while``` loops. Just try to add ```yield();``` within each ```while``` loop.
Arduino doesn't have a WDT, ESP has it and it cannot be disabled.
Adding yield() just prevents the WDT to kick in (in most cases).

Also, other functions like ```sim900_read_buffer()```, and ```sim900_send_cmd()``` and perhaps others are called from here.
You should also check those. (and others, called from there)

BharathTatikonda, 2016-09-22:
gsm.isSMSunread(); this function is same as in the arduino sim900 library......... could u also check with gprs sim900a libraries..................... in that libraries i have deleted only two functions i.e,.
1.void GPRS::listen(void)
  {
	gprsSerial.listen();
}
2.bool GPRS::isListening(void)
   {
	return gprsSerial.isListening();
   }

BharathTatikonda, 2016-09-22:
ok i will check with the yield(); at the loops,........

BharathTatikonda, 2016-09-22:
yes, thank you so much for your suggestion, i will got better result while using yield(); this function for only some loops only,.... i can search with other functions also,...... thank you somuch ,... if there is any queries i will contact with u,,,.....

supersjimmie, 2016-09-22:
You are welcome.  :)
I know, from my own experience, this is a problem many people have when using arduino code on esp8266. That's why it is also easy to solve, if you know where to search.
(you can close this issue with the button below if it is solved for you)

BharathTatikonda, 2016-09-22:
ok done

BharathTatikonda, 2016-09-27:
hi, could u tell me where the yield() will be presented in the libraries.....
and also the function of the yield();

closed 2016-09-29 by BharathTatikonda

gocamiot, 2017-08-31:
Same thing with esp-12f. Yield() solved it. long while loops are problematic.  Thank you.

devyte, 2017-08-31:
That is expected. The ESP requires free time off your code to handle the
wifi stack. There are watchdog timers in place to assure that. If you do an
long loop in your code, a wdt will trigger and reboot.
Calling yield() is meant to help with long operations. It gives a bit of
time to the wifi stack to handle critical operations. It can only be called
from within the loop, or functions called from the loop. It can't be called
from async callbacks.