esp8266/Arduino #2543: Support for Timezones expressed in seconds

stelgenhof, 2016-09-22:
Currently the `sntp_set_timezone` function accepts an integer with limits set between -11 and 13 (UTC hour offsets). This limits the use for timezones that have not an offset of an exact hour. See for example [UTC−09:30](https://en.wikipedia.org/wiki/UTC%E2%88%9209:30). 

There are a few more such timezones (See: https://en.wikipedia.org/wiki/Time_zone#List_of_UTC_offsets)

Additionally there is also a UTC offset UTC−12:00 and UTC+14:00. 

It would be good if the code could be updated to reflect the correct UTC offsets. In that regards it might be better to express the function argument in seconds (rather than hours). This would support for timezones like UTC-09:30.

PRO2XY, 2017-05-09:
As of the present core, the timezone is expressed in seconds in the call to configTime(....). However after being passed to the underlying function in [time.c](https://github.com/esp8266/Arduino/blob/2126146e20042878026f03a19107555f32e3431c/cores/esp8266/time.c#L64)  it is divided by 3600, when calling the sntp function [sntp_set_timezone(sint8 timezone)](https://github.com/esp8266/Arduino/blob/2126146e20042878026f03a19107555f32e3431c/cores/esp8266/time.c#L74), where it gets rounded back to a whole number. 

I tried to change the datatype on the relevant functions in sntp.c/.h and time.c/.h hoping that it might be a simple fix for timezones that are not a full hour difference, but it didn't seem to work. 
Also (somewhat related) changing daylightOffset_sec doesn't seem to be a workaround when trying to add that half an hour either.

devyte, 2017-10-12:
@stelgenhof @PRO2XY have you tried TimeLib, and NtpClientLib? They may be easier to modify.

stelgenhof, 2017-10-12:
@devyte Could you perhaps send a link for these two libraries you are referring to? There are many with the same name :). 

NTPClientLib has timezone support but with a similar limitation (time difference is an uint8_t) and TimeLib has no timezone support (This is if these are the ones you are referring to).

That's why I created my own NTPClient with full timezone support: https://github.com/stelgenhof/NTPClient. It only lacks DST support at this moment.

Nonetheless, I would be nice if it could be corrected in this SDK.

devyte, 2019-02-27:
Closing as duplicate of #5823 .

closed 2019-02-27 by devyte