esp8266/Arduino #2544: NodeMCU and Adafruit Flora Colour Sensor

LedAl, 2016-09-22:
Hello,

I'm working on creating a multi-purpose sensor on an AI-Thinker NodeMCU 1.0 (ESP8266).

I've been given a FLORA RGB Colour Sensor (https://learn.adafruit.com/adafruit-color-sensors/overview) in order to detect lux and RGB colour values, but when checking out the TCS34725 library examples they ask me to connect the SDA and SCL to analogue 4 and 5 (obviously the program has been designed for an Adruino Uno and not a NodeMCU ESP8266).

I'm wondering if any of you guys can help me connect the sensor in such a way that I can draw data from it? I think it communicates using I2C, but I'm not sure how to set up the general ports to I2C (the examples on the NodeMCU don't really help).

Thanks in Advance.

supersjimmie, 2016-09-22:
On a NodeMCU:
SDA = D3, gpio0
SCL = D4, gpio2
If you like you can use other pins. 
You need to call:
```Wire.begin(<sda-pin>, <sdc-pin>);```
So default you need:
```Wire.begin(D3, D4);```  or ```Wire.begin(0,2);```

devyte, 2017-10-12:
Off-topic, closing per #3655 .

closed 2017-10-12 by devyte