esp8266/Arduino #2545: WiFi events are not executed

gmag11, 2016-09-23:
### Basic Infos

#### Hardware
Hardware:			ESP-12
Core Version:      	Last commit as 2nd sept 2016

### Description

Specific WiFi events seem not to be executed, only onEvent handler does run. I've debugged code and found that if I change [this line in ESP8266WiFiGeneric.cpp](https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/src/ESP8266WiFiGeneric.cpp#L213) from
`WiFiEventHandler &handler = *it;` to `WiFiEventHandler handler = *it;` events handlers trigger correctly.

I've tried commit that first implemented specific event wifi handlers and it behaves the same. I've seen it working before so I do not know if I am doing something wrong. My code is on my [NTPClient repository](https://github.com/gmag11/NtpClient/tree/develop/examples/NTPClientESP8266)

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		4MB/1MB
CPU Frequency:		80Mhz
Flash Mode:			qio
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		nodemcu

### Sketch

```cpp

#include <TimeLib.h>
#include "WifiConfig.h"
#include <NtpClientLib.h>
#include <ESP8266WiFi.h>

#ifndef WIFI_CONFIG_H
#define YOUR_WIFI_SSID "YOUR_WIFI_SSID"
#define YOUR_WIFI_PASSWD "YOUR_WIFI_PASSWD"
#endif // !WIFI_CONFIG_H

void setup()
{
	Serial.begin(115200);
	WiFi.mode(WIFI_STA);
	WiFi.begin(YOUR_WIFI_SSID, YOUR_WIFI_PASSWD);

	NTP.onNTPSyncEvent([](NTPSyncEvent_t ntpEvent) {
		if (ntpEvent) {
			Serial.print("Time Sync error: ");
			if (ntpEvent == noResponse)
				Serial.println("NTP server not reachable");
			else if (ntpEvent == invalidAddress)
				Serial.println("Invalid NTP server address");
		}
		else {
			Serial.print("Got NTP time: ");
			Serial.println(NTP.getTimeDateString(NTP.getLastNTPSync()));
		}
	});
	WiFi.onEvent([](WiFiEvent_t e) {
		Serial.printf("Event wifi -----> %d\n", e);
	});
	WiFi.onStationModeGotIP([](WiFiEventStationModeGotIP ipInfo) { // As soon WiFi is connected, start NTP Client
		Serial.printf("Got IP: %s\r\n", WiFi.localIP().toString().c_str());
		NTP.begin("pool.ntp.org", 1, true);
		NTP.setInterval(63);
	});

}

void loop()
{
	static int i = 0;
	static int last = 0;

	if ((millis() - last) > 5100) {
		//Serial.println(millis() - last);
		last = millis();
		Serial.print(i); Serial.print(" ");
		Serial.print(NTP.getTimeDateString()); Serial.print(" ");
		Serial.print(NTP.isSummerTime() ? "Summer Time. " : "Winter Time. ");
		Serial.print("WiFi is ");
		Serial.print(WiFi.isConnected() ? "connected" : "not connected"); Serial.print(". ");
		Serial.print("Uptime: ");
		Serial.print(NTP.getUptimeString()); Serial.print(" since ");
		Serial.println(NTP.getTimeDateString(NTP.getFirstSync()).c_str());

		i++;
	}
	delay(0);
}
```

### Debug Messages

```
Opening port
Port open
0 Time not set Winter Time. WiFi is not connected. Uptime: 0 days 00:00:05 since Time not set
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 2
cnt 

connected with Virus_Detected!!!, channel 11
dhcp client start...
wifi evt: 0
Event wifi -----> 0
ip:192.168.5.17,mask:255.255.255.0,gw:192.168.5.1
wifi evt: 3
Event wifi -----> 3
1 Time not set Winter Time. WiFi is connected. Uptime: 0 days 00:00:10 since Time not set
pm open,type:2 0
2 Time not set Winter Time. WiFi is connected. Uptime: 0 days 00:00:15 since Time not set
3 Time not set Winter Time. WiFi is connected. Uptime: 0 days 00:00:20 since Time not set
4 Time not set Winter Time. WiFi is connected. Uptime: 0 days 00:00:25 since Time not set
5 Time not set Winter Time. WiFi is connected. Uptime: 0 days 00:00:30 since Time not set
6 Time not set Winter Time. WiFi is connected. Uptime: 0 days 00:00:35 since Time not set
```

### Debug Messages after making suggested change
In [this line in ESP8266WiFiGeneric.cpp](https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/src/ESP8266WiFiGeneric.cpp#L213)
```
Opening port
Port open
0 Time not set Winter Time. WiFi is not connected. Uptime: 0 days 00:00:05 since Time not set
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 2
cnt 

connected with Virus_Detected!!!, channel 11
dhcp client start...
wifi evt: 0
Event wifi -----> 0
ip:192.168.5.17,mask:255.255.255.0,gw:192.168.5.1
wifi evt: 3
Event wifi -----> 3
Got IP: 192.168.5.17
please start sntp first !
Time Sync error: NTP server not reachable
Fri Sep 23 16:18:07 2016

Fri Sep 23 16:18:08 2016

1 Time not set Winter Time. WiFi is connected. Uptime: 0 days 00:00:10 since Time not set
pm open,type:2 0
2 Got NTP time: 17:18:14 23/09/2016
17:18:14 23/09/2016 Summer Time. WiFi is connected. Uptime: 0 days 00:00:15 since 17:18:14 23/09/2016
3 17:18:19 23/09/2016 Summer Time. WiFi is connected. Uptime: 0 days 00:00:20 since 17:18:14 23/09/2016
```

igrr, 2016-09-23:
You aren't saving the handle obtained from WiFi.onStationModeGotIP, so the event handler is automatically unsubscribed. You need to hold the handle in order for the handler to be called.

gmag11, 2016-09-23:
Yes, it works so.
They must be global variables then. Is there any other way? `WiFiEventHandler e1, e2;`

igrr, 2016-09-23:
Not necessarily global. They can be members of an object which is receiving the event callback. That's the point — lifetime of the callback is bound to the lifetime of the object. Once the object is deleted, event handler is automatically unsubscribed.
If the life time of the handler is same as the lifetime of the application, then the object will be either global, or static at function scope.

gmag11, 2016-09-23:
Now I understand. Thank you!

closed 2016-09-23 by gmag11

dachshund-digital, 2019-03-07:
This bug maybe back again or some variant of it... onStationModeConnected and onStationModeGotIP event handlers never fire.  Used the NTPClientLib example for ESP8266 (see git for code https://github.com/gmag11/NtpClient/tree/master/examples/NTPClientESP8266) and only event handler that consistently fires is the onStationModeDisconnected handler and it fires over and over when a valid active WiFi connection already exists.  Keep getting reason code 201 when status is checked, which seems completely wrong when active Wifi connect present.

smartynov, 2019-11-15:
I second this, I'm getting a strange behaviour of `onStationModeConnected` on NodeMCU too – can't get it working, it never fires.

wladwnt, 2019-12-31:
Cant get working onStationModeGotIP, it never fires, onStationModeDisconnected works great.

michaelkipper, 2021-04-12:
I'm seeing a reliable firing of `onStationModeGotIP` but sometimes missing `onStationModeConnected`.

What it should look like:
```
Starting up...
Done setup.
onStationModeConnected: SSID 'Kipper2' channel 6
onStationModeGotIP: 10.0.0.155
HTTP request: /metrics
```

What it sometimes looks like:
```
Starting up...
Done setup.
onStationModeGotIP: 10.0.0.155
HTTP request: /metrics
```

d-a-v, 2021-04-12:
If onStationModeGotIP is reliable, can't it be assumed that when an IP is got, a connection must have previously been enabled ?

michaelkipper, 2021-04-12:
> If onStationModeGotIP is reliable, can't it be assumed that when an IP is got, a connection must have previously been enabled ?

Is that true for the _device_ or the _session_?

And furthermore, I'm not confident that if a connection is disconnected and re-established, that a new IP will have been got. Overall, I'd like to see more documentation into the lifecycle of a WiFi connection which will hopefully illuminate why I'm seeing what I'm seeing.

devyte, 2021-04-15:
This is a closed issue. 
Having said that, which core version? I seem to remember that the wifi connect event doesn't fire when an early connection is established during boot (persistent true and all that). There was a recent PR merged that changes that, so early connections are disabled by default. Assuming that's the cause, latest git shouldn't show what you observe.

HubKing, 2021-09-14:
> 
> 
> You aren't saving the handle obtained from WiFi.onStationModeGotIP, so the event handler is automatically unsubscribed. You need to hold the handle in order for the handler to be called.

Damn, I have wasted tens of minutes trying to find out the reason. I have never seen  a requirement of saving the result of an callback subscription function as a global variable in other languages, so I never thought of that could be the reason before reading this comment.