esp8266/Arduino #2549: Can't compile if using std::vector

Geend, 2016-09-25:
### Description
I can't compile my sketch if I'm using std::vector anywhere in it.
The error message below refers to the following lines in the Arduino.h:
```
#ifndef _GLIBCXX_VECTOR
// arduino is not compatible with std::vector
#define min(a,b) ((a)<(b)?(a):(b))
#define max(a,b) ((a)>(b)?(a):(b))
#endif
```
If I delete these lines it compiles just fine, but I haven't verified if it actually works at runtime.

So. Is this intended? Is the esp8266 version of the arduino framework not capable of running std::vector? Or am I just doing something wrong?

### Debug Messages
```
In file included from /home/torben/.platformio/packages/framework-arduinoespressif/libraries/ESP8266WiFi/src/WiFiClient.h:25:0,
from src/RestHttpServer/RestHttpServer.h:8,
from src/RestHttpServer/RestHttpServer.cpp:5:
/home/torben/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/4.8.2/bits/stl_vector.h: In member function 'std::vector<_Tp, _Alloc>::size_type std::vector<_Tp, _Alloc>::_M_check_len(std::vector<_Tp, _Alloc>::size_type, const char*) const':
/home/torben/.platformio/packages/framework-arduinoespressif/cores/esp8266/Arduino.h:253:18: error: expected unqualified-id before '(' token
#define max(a,b) ((a)>(b)?(a):(b))
^
/home/torben/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/4.8.2/bits/stl_bvector.h: In member function 'std::vector<bool, _Alloc>::size_type std::vector<bool, _Alloc>::_M_check_len(std::vector<bool, _Alloc>::size_type, const char*) const':
/home/torben/.platformio/packages/framework-arduinoespressif/cores/esp8266/Arduino.h:253:18: error: expected unqualified-id before '(' token
#define max(a,b) ((a)>(b)?(a):(b))
^
```

closed 2016-10-08 by Geend

gojimmypi, 2016-12-07:
did you find a solution for this?

mkarklins, 2016-12-14:
In this include order both libraries work:
```
#include <vector>
#include <WiFiClient.h>
```

Geend, 2016-12-19:
If I remeber correctly including Arduino.h in every class I'm using vectors in fixed it.