esp8266/Arduino #2551: I2C: Missing delay in twi_write_start

vlast3k, 2016-09-26:
### Basic Infos

#### Hardware
Hardware:			ESP-12F (80Mhz)
Core Version:      	2.3.0

### Description
In the last weeks i had extreme trouble in establishing connection with a CO2 Sensor (CDM7160) over I2C. Tried various kinds of pullup resistors, finally settled on 2k. And with lot's of error recovery code i was able to read from the sensor most of the time. Then i connected a Dust Sensor over I2C (PM2005), and then the whole thing became very unstable. The most time Wire.endTransmission() was throwing 4 the first time, and then it was fine, but in some cases it wasn't able to start communication at all.
Strangely enough Wire.status() was returning I2C_OK right before this call.
Finally i saw that 4 is returned when: 
  if(!twi_write_start()) return 4;//line busy

And what i did was to add an additional delay in twi_write_start
```
static bool twi_write_start(void) {
  SCL_HIGH();
  SDA_HIGH();
---->>>>  twi_delay(twi_dcount);
  if (SDA_READ() == 0) return false;
  twi_delay(twi_dcount);
  SDA_LOW();
  twi_delay(twi_dcount);
  return true;
}
```

after i added this, now it works w/o any trouble. So i was wondering Is it correct to expect that the rise time for SDA will be just a cycle ? Especially if i had called before Wire.status, which ends with calling  if(!twi_write_start()) , which puts SDA LOW at the end. But perhaps for consistency it is good to wait a bit after both lines are put high, to make a reading if the line rose.

I am by far not too knowledgeable in I2C, and maybe my hw design (with both sensors connected to 10 cm wires) is not good enough, but with this fix,the results are quite stable right now

themindfactory, 2016-09-26:
I would say this is specific to you, from my experience with I2C devices,
if they are either designed with pure logic they are fast to reply to state
changes, if they are badly designed or controlled via a software I2C like a
slave device in a microcontroller, depending on design they can be slow and
require delays.

RichardS

vlast3k, 2016-09-26:
i understand, indeed this one seems to be particularly bad designed and
picky :(

closed 2016-09-27 by vlast3k

EBZ-Krisemendt, 2017-02-28:
i know this issue is closed, but i'm struggling with the same sensor. 
Do you got any example code on how you recieving co2 readings?
i currently starting to first:

// init sensor continuous operating mode
Wire.beginTransmission(0xD2); // Binary slave adress 1101001 + R/W = 1 (write)
Wire.write(0x01); // register: operating mode
Wire.write(0x06); // continuous operation
Wire.endTransmission

when i then try to read:
// according to pdf-docu i need to first write then read
Wire.beginTransmission(0xD2); // as before
Wire.write(0x03); // starting point for reading
Wire.endTransmission();

// now recieve data
Wire.requestFrom(0xD3, 4); // Binary slave address 1101001 + R/W = 0 (read)
// handle response

but unfortunatly i don't even get any response. even the init gives a NACK, while documentation says it should give ACK after each byte-block. any help by you is highly appreciated :)