esp8266/Arduino #2552: TCP connection always Accidental disconnection

kingxf1995, 2016-09-27:
#### Hardware
Hardware:			ESP-12E
Core Version:      	2.3.0

### Description
I found esp826612e network may have some problem, so I write a simple code to test it.
In short, ESP826612e act as station join my router wifi, and act as a tcp server, accept only one connection, my computer act as a TCP client. Client send a byte then server recived it and send back a same byte. In loop, sometime ESP8266 think TCP connection is break, computer found tcp connection is break after a short time

### Sketch

```cpp

#include <ESP8266WiFi.h>

WiFiServer server(23333);
WiFiClient client;
bool connected = false;

const char *ssid = "xxx";
const char *password = "yyy";

void setup() {
  Serial.begin(115200);
  Serial.println();
  server.begin();
  WiFi.mode( WIFI_STA );
  WiFi.begin ( ssid, password );
	// Wait for connection
	while ( WiFi.status() != WL_CONNECTED ) {
		delay(500);
    Serial.print('.');
	}
	Serial.print("\nIPAddress: ");
	Serial.println ( WiFi.localIP() );
}

void loop() {
  if (client.connected()) {
    if (!connected) {
      connected = true;
      Serial.println("\ntcp connected");
    }
    int ch;
    if ((ch=client.read()) != -1) {
      Serial.printf("%c", ch);
      client.write((char)ch);
    }
  }
  else {
    if (connected) {
      connected = false;
      Serial.println("\ntcp disconnected");
    }
    client = server.available();
    client.setNoDelay(true);
  }
}
```

### test script

```python
import socket  
import time

host    = '192.168.31.215'
port    = 23333
addr    = (host, port)

client=socket.socket(socket.AF_INET, socket.SOCK_STREAM)
client.connect(addr)
client.setsockopt(socket.IPPROTO_TCP, socket.TCP_NODELAY, 1)
last = time.time()

while True:
  client.send('x')
  data=client.recv(1)
  if data == 'x':
    cur = time.time()
    print '%4d ms' % ((cur - last) * 1000)
    time.sleep(0.05)
    
    last = time.time()
```

kentaylor, 2016-12-09:
@kingxf1995, if the WiFi connection is unreliable, for example due to radio interference, then the behaviour you report is exactly what you would want, that is, both ends are aware there is no longer a TCP connection.

However, there is [circumstances](https://github.com/esp8266/Arduino/issues/2750) where the ESP thinks it is connected so that client.connected() will report true when it is not connected and your sketch would get stuck in an endless loop.

mtnbrit, 2016-12-09:
Is it true that the device still responds to ping when its stuck in this state? Ive seen this, if so.

kentaylor, 2016-12-09:
When @mtnbrit  says

> stuck in this state

which state is meant? 

If the query relates to client.connected() reporting connection state incorrectly please ask in https://github.com/esp8266/Arduino/issues/2750

mtnbrit, 2016-12-09:
sorry yes I mean when its stuck in the loop forever, as you mention in that thread, ill ask over there...

kingxf1995, 2016-12-11:
Thank you @kentaylor  @mtnbrit , I have solved this problem. In fact I used a bad router which led to packet loss seriously.

devyte, 2017-10-12:
Per previous comment, closing as resolved.

closed 2017-10-12 by devyte