esp8266/Arduino #2553: SPIFFS File implementation of seek is non-standard for Arduino

Makuna, 2016-09-27:
### Basic Infos
SPIFFS File implementation of seek is non-standard for Arduino

### Description
All implementations of "seek" for Arduino File objects only take one param and assume seek from the front of the file.
The SPIFFs only implements one "seek" that requires a second param that defines the SeekMode.  This makes it difficult to write code/libraries that will work across SD file objects and SPIFFs file objects.

The solution is change the SeekMode to have a default value of SeekSet or add another seek that doesn't take the second parameter.

FS.h
```
bool seek(uint32_t pos, SeekMode mode = SeekSet);
```
or
```
bool seek(uint32_t pos)
{
    seek(pos, SeekSet);
}
```

Makuna, 2016-09-27:
ALSO, the read method was changed.  While agree with returning size_t, the change of the first param from void* to uint8_t* could be problematic; but is easily worked around by always casting to uint8_t* when passing anything that is not already a collection of 8bit types (like a structure).

Torxgewinde, 2017-03-17:
In addition I had to remove the "const" keyword as follows to use the GetPixelColor function:
`typename T_COLOR_FEATURE::ColorObject GetPixelColor(int16_t x, int16_t y)`
otherwise the compilation failed with a message similar "to discards qualifiers [-fpermissive]".

devyte, 2017-10-12:
Referenced PR is already merged, closing.

closed 2017-10-12 by devyte