esp8266/Arduino #2556: Always getting HTTPC_ERROR_CONNECTION_REFUSED for HTTPS GET

J6B, 2016-09-27:
### Basic Infos
My sketch do a GET using HTTPClient.
I was in release 2.1.0 and I just updated to 2.3.0 release
My sketch was able to send GET for HTTP and HTTPS URI using 2.1.0 release.
Since I updated to 2.3.0, HTTP works fine.
But the issue is while I'm trying to send same GET request using HTTPS.

I added fingerprint of my server certificate to my "begin" call (don't know why but it became mandatory in 2.3.0 but it was not in 2.1.0) and I get a true result.
But, then, when a call GET function to do the request, it always return -1 which mean HTTPC_ERROR_CONNECTION_REFUSED

#### Hardware
Hardware:			ESP-01
Core Version:      	2.3.0

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		1MB
CPU Frequency:		80Mhz
Flash Mode:			dio
Flash Frequency:	40Mhz
Upload Using:		Both OTA and SERIAL
Reset Method:		ck

### Sketch

```cpp

#include <ESP8266WiFi.h>
#include <ESP8266HTTPClient.h>

void setup() {
 //init Wifi
}

void loop() {

String completeURI = "https://192.168.1.123/plugins/teleinfo/core/php/jeeTeleinfo.php?api=YQFMNcxGJ52XSrHYcF3U&ADCO=050522038502&HCHP=072686874&PAPP=00730";

  //send HTTP request
  HTTPClient http;
  int beginResult = http.begin(completeURI,"‎83 0d 3c 85 19 e5 29 ca da 38 b9 9c f2 66 bb 59 cd 00 9c aa");
  int requestResult = http.GET();
  Serial.print(beginResult);
  Serial.println(requestResult);
  http.end();

}

mahmed8003, 2016-09-29:
I am also having the same issue, But I am making POST request to http server.
Here is debug log:
connected with VZO, channel 1
dhcp client start...
wifi evt: 0
chg_B1:-40
ip:192.168.1.4,mask:255.255.255.0,gw:192.168.1.1
wifi evt: 3
• WiFi connected
[HTTP-Client][begin] host: 192.168.1.3 port: 3005 uri: /api/device/register
[hostByName] Host: 192.168.1.3 is a IP!
[HTTP-Client] failed connect to 192.168.1.3:3005
[HTTP-Client][returnError] error(-1): connection refused
Serious Error response
Error code
-1
[HTTP-Client][end] tcp is closed
:ref 1
:rcl
:abort
pm open,type:2 0

thezenox, 2016-12-01:
I have the same issue with https POST, same URL with http works fine
http.begin("https://upload.inseryoururlhere/script.php");
[...] setting header and request
int httpCode = http.POST(request); 
always returns -1 (HTTPC_ERROR_CONNECTION_REFUSED)

From ESP8266HTTPClient.cpp
```
int HTTPClient::sendRequest(const char * type, uint8_t * payload, size_t size)
{
    // connect to server
    if(!connect()) {
        return returnError(HTTPC_ERROR_CONNECTION_REFUSED);
    }
```
where is the connect() from ? cant find it

J6B, 2016-12-02:
Hi,

You need to provide Certificate Thumbprint for HTTPS request  : 
`http.begin("https://192.168.1.12/test.html", "7a 9c f4 db 40 d3 62 5a 6e 21 bc 5c cc 66 c8 3e a1 45 59 38"); //HTTPS`

JohnAntonusMaximus, 2017-01-09:
I provided the Certificate Thumbprint and I'm still getting a "connection failed" refusal from the server. The SSL certificate is working fine too, successfully sent and received data both from curl and Postman. Any ideas on this?

JohnAntonusMaximus, 2017-01-09:
--FIXED--

:-)

Hey guys, so here's what I did that worked for me, you're most likely having the same issue. The example sketch is wrong, but there's more to it.

--------------------------------------------------------------------------------
**First make sure your SSL Certificate is set up correct using CURL:**
curl -kvI https://www.example.com
_It should print SSL certificate verify ok. If it prints something like common name: www.example.com (does not match 'www.somedomain.com') then something is not configured correctly._
---------------------------------------------------------------------------------

**First open your terminal and insert the following command:**
openssl s_client -connect www.example.com:443

**Copy the output from the terminal connection from '-----BEGIN CERTIFICATE-----' to '-----END CERTIFICATE-----' INCLUDING those two lines.**

**Create a .pem file with the cert sent from the server**
echo "[ctrl+v]" >> cert.pem

**Enter the following command to generate the correct SHA1 Fingerprint**
openssl x509 -noout -in cert.pem -fingerprint -sha1

**Copy the fingerprint and paste it into your sketch where the thumbprint should go AS IS, do not remove the colons or capital letters**
http.begin("https://www.example.com/test","B5:68:8F:AE:29:68:4E:44:59:24:E5:1A:18:F1:8A:6A:EA:60:40:4E");

J6B, 2017-01-10:
Thanks for your help,

Sadly, I still have the same issue.

First I did what you described
Checked with **CURL**, I got : ` SSL certificate verify result: self signed certificate (18), continuing anyway.`

Then I enabled debug and got this : 

> [HTTP-Client][end] tcp is closed
> [HTTP-Client][begin] url: https://192.168.1.5
> [HTTP-Client][begin] host: 192.168.1.5 port: 443 url: 
> [HTTP-Client][begin] httpsFingerprint: â€Ž6F:E6:2C:15:4C:F0:88:2A:AE:56:C0:4A:19:B4:27:63:AB:68:D8:CC
> please start sntp first !
> State:	sending Client Hello (1)
> State:	receiving Server Hello (2)
> State:	receiving Certificate (11)
> State:	receiving Server Hello Done (14)
> State:	sending Client Key Exchange (16)
> State:	sending Finished (16)
> State:	receiving Finished (16)
> [HTTP-Client] connected to 192.168.1.5:443
> [HTTP-Client] transport level verify failed
> [HTTP-Client][returnError] error(-1): connection refused
> beginResult1
> requestResult-1
>

Do you have any other idea (The CN of the certificate is 192.168.1.5)

JohnAntonusMaximus, 2017-01-10:
Hey bud, the URL you put in is a local IP address for a localhost on your network, so I can't CURL it, your certificate needs to be on a production server. Push it to Heroku, you'll need to pay $7 to upgrade to a  "hobby" dyno for the SSL certificate, but this method will work 100% on that. That's how I figured this out.

mahmed8003, 2017-01-10:
My issue get fixed when I call Http methods outside from following function
```
onWifiGotIp(const WiFiEventStationModeGotIP& event) {
    ///making http request from this methods was not working for me
}
```
My issue was not related to HTTPS by the way.

J6B, 2017-01-10:
Does someone can do debug and post it there?
Isn't it Strange that i find in debug this : 

> httpsFingerprint: **â€Ž**6F:E6:2C...

instead of : 

> httpsFingerprint: 6F:E6:2C...

davisonja, 2017-01-10:
Yes. You may have an 'invisible character' in your https fingerprint in
your sketch.
I removed mine by editing the sketch, putting the cursor by the (in your
case) first '6' and pressing backspace until I removed the opening
string-quote. (then putting the string quote back :)
I had copy-and-pasted my fingerprint out of Chrome, which turns out was a
bad move (since I got the spare bytes, which were non-printing and thus
invisible in the Arduino IDE.).

J,

J6B, 2017-01-10:
Bravo,

It finally works!!
It was special character invisible in IDE (davisonja was right).

Once I removed it, It worked fine : 

> [HTTP-Client][begin] url: https://192.168.1.5/core/api/jeeApi.php?apikey=02Zwiei25HWYb7w6WG1T5GL42wqjSYi2Oj6V7B7BwDBjLg7J&type=virtual&id=53&value=1
> [HTTP-Client][begin] host: 192.168.1.5 port: 443 url: /core/api/jeeApi.php?apikey=02Zwiei25HWYb7w6WG1T5GL42wqjSYi2Oj6V7B7BwDBjLg7J&type=virtual&id=53&value=1
> [HTTP-Client][begin] httpsFingerprint: 9e 40 9b 61 ee 19 34 84 05 50 f8 90 04 6a 46 2e 72 77 e7 d5
> please start sntp first !
> State:	sending Client Hello (1)
> State:	receiving Server Hello (2)
> State:	receiving Certificate (11)
> State:	receiving Server Hello Done (14)
> State:	sending Client Key Exchange (16)
> State:	sending Finished (16)
> State:	receiving Finished (16)
> [HTTP-Client] connected to 192.168.1.5:443
> [HTTP-Client][handleHeaderResponse] RX: 'HTTP/1.1 200 OK'
> [HTTP-Client][handleHeaderResponse] RX: 'Date: Tue, 10 Jan 2017 21:53:11 GMT'
> [HTTP-Client][handleHeaderResponse] RX: 'Server: Apache'
> [HTTP-Client][handleHeaderResponse] RX: 'Set-Cookie: PHPSESSID=5g24n8sf27s6q5vht6he94am36; path=/'
> [HTTP-Client][handleHeaderResponse] RX: 'Expires: Thu, 19 Nov 1981 08:52:00 GMT'
> [HTTP-Client][handleHeaderResponse] RX: 'Cache-Control: no-store, no-cache, must-revalidate, post-check=0, pre-check=0'
> [HTTP-Client][handleHeaderResponse] RX: 'Pragma: no-cache'
> [HTTP-Client][handleHeaderResponse] RX: 'Content-Length: 0'
> [HTTP-Client][handleHeaderResponse] RX: 'Connection: close'
> [HTTP-Client][handleHeaderResponse] RX: 'Content-Type: text/html; charset=UTF-8'
> [HTTP-Client][handleHeaderResponse] RX: ''
> [HTTP-Client][handleHeaderResponse] code: 200
> beginResult1
> requestResult200
> [HTTP-Client][end] tcp is closed

Thanks a lot to all of you

closed 2017-01-10 by J6B

landsteph, 2017-03-26:
I was facing the same issue, and I finally found the same solution. It was special character invisible in IDE.
But I found also why : copy/past fingerprint from Windows Certificate dialog box !
some characters are existing before the first hex, these chars are unwanted !

dapug, 2017-03-28:
Any other ideas?  None of these suggestions worked for me.  I have tried the thumbprint both with spaces and with colon ":" between the characters.  I have tried caps and lowercase on the letters. I have verified no hidden characters in the code.  

This is not a complex thing, simply a method call with parameters of URL and Thumbprint strings... yet all I get is "Connection Refused".  I hate to open a new issue, sucks we can't re-open existing ones.

landsteph, 2017-03-28:
follow the Verify function, it need two things to get it validated : 
1) the right fingerprint (delete your entire line in arduino and copy past the fingerprint from notepad.exe. personnaly, I spent hours thinking all good, but using the arrow keys, I saw the cursor not moving, that was due to invible chars. Thank you Microsoft for these invisible chars !)
2) look at WiFiClientSecure::verify(...), at the end, Domain need to be validated too (vs CN or SAN)

Did you tried to activate the "debug level" > "all" ?
it will activate DEBUGV macros, and you should see more info, like "pos:%d len:%d invalid hex sequence: %c%c" which told me that there was invalid chars.

I have an Huzzah esp8266, but the generic ESP8266 is working too, and allow full debug.

dapug, 2017-03-29:
I'm using **ESP8266HTTPClient** and I don't see any examples where others have made an explicit call to Verify as a two-step process of some kind (see original post above in fact, where is Verify?). 

I enabled debug, and see this:
**Invalid X509 ASN.1 file (X509 not ok)**

landsteph, 2017-03-29:
When you call `begin(url, fingerprint)` it imply to connect to the Host using SSL/TLS connection :

If you look inside the begin(host, fp) function, you will see `_transportTraits = TransportTraitsPtr(new TLSTraits(httpsFingerprint));` and if look inside the constructors of TLSTraits, you can see it is based on `WiFiClientSecure`. and the constructor call the `WiFiClientSecure::verfiy` function !

concerning the debug output you mentionned : 
1) is the CN or ASN matching the hostname you provided in url ?
2) did you tried to validate the X509 certs you made ? `openssl x509 -in certificate.crt -text -noout`

if you enable the full debug, what is the output ?
you should see, at least, one of these lignes :
```pos: ..... len: ... fingerprint too short
pos: ..... len: ... invalid hex sequence:
fingerprint doesn't match
domain name: ...
SAN ... : '...', no match
CN ... : '...', no match```

and i finaly, are you able to connect to the server using https and a web browser ?

good luke

dapug, 2017-03-29:
I didn't generate a cert, I am trying to connect to [data.sparkfun.com](https://data.sparkfun.com/streams/)
SHA1 fingerprint:  "29 9F 93 FA B0 89 37 C4 BC 76 D7 EB DE 13"

yes indeed, the CN of my request matches this cert, of course.  https://data.sparkfun.com/streams/...

yes of course, https://data.sparkfun.com is live and running, accessible from the browser

landsteph, 2017-03-30:
Hi !
I have tried this :

```
#define SSID_STR "wifi_ssid"
#define PASS_STR "password"

#include <ESP8266WiFi.h>
#include <ESP8266HTTPClient.h>

/*
  Debug Port = Serial
  Debug Level = All
*/

#ifdef DEBUG_ESP_PORT
#define DEBUG_MSG(...) DEBUG_ESP_PORT.printf( __VA_ARGS__ )
#else
#define DEBUG_MSG(...)
#endif

void setup() {
#ifdef DEBUG_ESP_PORT
  Serial.begin(115200);
  delay(1000);
#endif
  DEBUG_MSG("\n[SETUP]");
  WiFi.mode(WIFI_STA);
}

void loop() {
  DEBUG_MSG("[LOOP]");

  WiFi.begin(SSID_STR, PASS_STR);
  while (WiFi.status() != WL_CONNECTED) {
    DEBUG_WIFI(".");
    delay(100);
  }

  HTTPClient Http;
  String url = "https://data.sparkfun.com/streams/";
  String fp = "29 9f 93 fa b0 89 37 c4 bc 76 d7 eb de 13 6e ec 82 a4 4d 68";
  Http.begin(url,fp);
  DEBUG_HTTPCLIENT("Http.GET() returned : %s", Http.errorToString(Http.GET()).c_str());
  WiFi.disconnect();
  WiFi.mode(WIFI_OFF);
  ESP.deepSleep(60 * 1000 * 1000); // 60s
}
```

and get :
```
[SETUP][LOOP]f r0, ......................scandone
state: 0 -> 2 (b0)
.state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 3
cnt

connected with wifi_ssid, channel 4
dhcp client start...
wifi evt: 0
.......ip:###.###.###.###,mask:255.255.255.0,gw:###.###.###.###
wifi evt: 3
[HTTP-Client][begin] url: https://data.sparkfun.com/streams/
[HTTP-Client][begin] host: data.sparkfun.com port: 443 url: /streams/
[HTTP-Client][begin] httpsFingerprint: 29 9f 93 fa b0 89 37 c4 bc 76 d7 eb de 13 6e ec 82 a4 4d 68
[hostByName] request IP for: data.sparkfun.com
[hostByName] Host: data.sparkfun.com IP: 54.86.132.254
:ref 1
ssl/tls1.c:545 malloc 6864, left 28640
please start sntp first !
State:  sending Client Hello (1)
:wr
:sent 80
:rn 7
:ww
:rd 5, 7, 0
:rdi 7, 5
:rd 2, 7, 5
:rdi 2, 2
:c0 2, 7
Alert: invalid version
Error: SSL error 70
:wr
:rcl
:abort
:ww
Alert: unexpected message
Alert: close notify
[HTTP-Client] failed connect to data.sparkfun.com:443
[HTTP-Client][returnError] error(-1): connection refused
Http.GET() returned : connection refusedstate: 5 -> 0 (0)
rm 0
del if0

```

The key is "Error: SSL error 70" which mean my own ESP8266 is not able to connect to this server as it is not supporting TLS1.2. In order to be sure, I have tried the following :
1) create  a very basic https server using openssl s_server -accept 8443 -www 
2) try it using a webbrowser, it will show you what type of tls/ssl encryption you can do on your pc
3) use your esp8266 to connect to your "openssl s_server" and .getString() will show you what tls/ssl encryption it can handle.
for me it returned :
```
s_server -cert /etc/ssl/private/testing.crt -www -accept 8443
Secure Renegotiation IS NOT supported
Ciphers supported in s_server binary
TLSv1/SSLv3:ECDHE-RSA-AES256-GCM-SHA384TLSv1/SSLv3:ECDHE-ECDSA-AES256-GCM-SHA384
TLSv1/SSLv3:ECDHE-RSA-AES256-SHA384  TLSv1/SSLv3:ECDHE-ECDSA-AES256-SHA384
TLSv1/SSLv3:ECDHE-RSA-AES256-SHA     TLSv1/SSLv3:ECDHE-ECDSA-AES256-SHA
TLSv1/SSLv3:SRP-DSS-AES-256-CBC-SHA  TLSv1/SSLv3:SRP-RSA-AES-256-CBC-SHA
TLSv1/SSLv3:SRP-AES-256-CBC-SHA      TLSv1/SSLv3:DHE-DSS-AES256-GCM-SHA384
TLSv1/SSLv3:DHE-RSA-AES256-GCM-SHA384TLSv1/SSLv3:DHE-RSA-AES256-SHA256
TLSv1/SSLv3:DHE-DSS-AES256-SHA256    TLSv1/SSLv3:DHE-RSA-AES256-SHA
TLSv1/SSLv3:DHE-DSS-AES256-SHA       TLSv1/SSLv3:DHE-RSA-CAMELLIA256-SHA
TLSv1/SSLv3:DHE-DSS-CAMELLIA256-SHA  TLSv1/SSLv3:ECDH-RSA-AES256-GCM-SHA384
TLSv1/SSLv3:ECDH-ECDSA-AES256-GCM-SHA384TLSv1/SSLv3:ECDH-RSA-AES256-SHA384
TLSv1/SSLv3:ECDH-ECDSA-AES256-SHA384 TLSv1/SSLv3:ECDH-RSA-AES256-SHA
TLSv1/SSLv3:ECDH-ECDSA-AES256-SHA    TLSv1/SSLv3:AES256-GCM-SHA384
TLSv1/SSLv3:AES256-SHA256            TLSv1/SSLv3:AES256-SHA
TLSv1/SSLv3:CAMELLIA256-SHA          TLSv1/SSLv3:PSK-AES256-CBC-SHA
TLSv1/SSLv3:ECDHE-RSA-AES128-GCM-SHA256TLSv1/SSLv3:ECDHE-ECDSA-AES128-GCM-SHA256
TLSv1/SSLv3:ECDHE-RSA-AES128-SHA256  TLSv1/SSLv3:ECDHE-ECDSA-AES128-SHA256
TLSv1/SSLv3:ECDHE-RSA-AES128-SHA     TLSv1/SSLv3:ECDHE-ECDSA-AES128-SHA
TLSv1/SSLv3:SRP-DSS-AES-128-CBC-SHA  TLSv1/SSLv3:SRP-RSA-AES-128-CBC-SHA
TLSv1/SSLv3:SRP-AES-128-CBC-SHA      TLSv1/SSLv3:DHE-DSS-AES128-GCM-SHA256
TLSv1/SSLv3:DHE-RSA-AES128-GCM-SHA256TLSv1/SSLv3:DHE-RSA-AES128-SHA256
TLSv1/SSLv3:DHE-DSS-AES128-SHA256    TLSv1/SSLv3:DHE-RSA-AES128-SHA
TLSv1/SSLv3:DHE-DSS-AES128-SHA       TLSv1/SSLv3:DHE-RSA-SEED-SHA
TLSv1/SSLv3:DHE-DSS-SEED-SHA         TLSv1/SSLv3:DHE-RSA-CAMELLIA128-SHA
TLSv1/SSLv3:DHE-DSS-CAMELLIA128-SHA  TLSv1/SSLv3:ECDH-RSA-AES128-GCM-SHA256
TLSv1/SSLv3:ECDH-ECDSA-AES128-GCM-SHA256TLSv1/SSLv3:ECDH-RSA-AES128-SHA256
TLSv1/SSLv3:ECDH-ECDSA-AES128-SHA256 TLSv1/SSLv3:ECDH-RSA-AES128-SHA
TLSv1/SSLv3:ECDH-ECDSA-AES128-SHA    TLSv1/SSLv3:AES128-GCM-SHA256
TLSv1/SSLv3:AES128-SHA256            TLSv1/SSLv3:AES128-SHA
TLSv1/SSLv3:SEED-SHA                 TLSv1/SSLv3:CAMELLIA128-SHA
TLSv1/SSLv3:PSK-AES128-CBC-SHA       TLSv1/SSLv3:ECDHE-RSA-RC4-SHA
TLSv1/SSLv3:ECDHE-ECDSA-RC4-SHA      TLSv1/SSLv3:ECDH-RSA-RC4-SHA
TLSv1/SSLv3:ECDH-ECDSA-RC4-SHA       TLSv1/SSLv3:RC4-SHA
TLSv1/SSLv3:RC4-MD5                  TLSv1/SSLv3:PSK-RC4-SHA
TLSv1/SSLv3:ECDHE-RSA-DES-CBC3-SHA   TLSv1/SSLv3:ECDHE-ECDSA-DES-CBC3-SHA
TLSv1/SSLv3:SRP-DSS-3DES-EDE-CBC-SHA TLSv1/SSLv3:SRP-RSA-3DES-EDE-CBC-SHA
TLSv1/SSLv3:SRP-3DES-EDE-CBC-SHA     TLSv1/SSLv3:EDH-RSA-DES-CBC3-SHA
TLSv1/SSLv3:EDH-DSS-DES-CBC3-SHA     TLSv1/SSLv3:ECDH-RSA-DES-CBC3-SHA
TLSv1/SSLv3:ECDH-ECDSA-DES-CBC3-SHA  TLSv1/SSLv3:DES-CBC3-SHA
TLSv1/SSLv3:PSK-3DES-EDE-CBC-SHA

```

Tips: create your own ssl using :
```
sudo apt-get install openssl ssl-cert
sudo make-ssl-cert /usr/share/ssl-cert/ssleay.cnf /etc/ssl/private/testing.crt
```

dapug, 2017-03-30:
Thank you, this is great info, super helpful. 

I saw posts that TLS1.2 is supported on ESP8266, but I had to do more digging and apparently that came AFTER the most recent "release" (v2.3.0 which I have). So, um, yeah... the BoardManager is woefully behind in regards to the ESP8266 Arduino SDK.

pieman64, 2017-03-30:
@dapug The ESP god, @igrr  has a 79 page PDF for **v2.4.0** at https://media.readthedocs.org/pdf/arduino-esp8266/docs_to_readthedocs/arduino-esp8266.pdf
The document is dated 20 Feb 2017 (about a month ago). Page 28 states TLS 1.2 is **not** supported.
I have connections to sites using TLS 1.2 and using v2.3.0 so I would say the docs are outdated / wrong.

landsteph, 2017-03-30:
https://github.com/esp8266/Arduino/releases > latest release : 2.3.0

be carreful, sites using tls1.2 also support lower security. 'data.sparkfun.com' don't.
in order to test that way too, you can try the folling command :
`nmap --script ssl-enum-ciphers -p 443 data.sparkfun.com`

```
Starting Nmap 6.47 ( http://nmap.org ) at 2017-03-30 23:35 CEST
Stats: 0:00:01 elapsed; 0 hosts completed (0 up), 1 undergoing Ping Scan
Ping Scan Timing: About 100.00% done; ETC: 23:35 (0:00:00 remaining)
Nmap scan report for data.sparkfun.com (54.86.132.254)
Host is up (0.11s latency).
rDNS record for 54.86.132.254: ec2-54-86-132-254.compute-1.amazonaws.com
PORT    STATE SERVICE
443/tcp open  https
| ssl-enum-ciphers:
|   SSLv3: No supported ciphers found
|   TLSv1.2:
|     ciphers:
|       TLS_ECDHE_RSA_WITH_AES_128_CBC_SHA - strong
|       TLS_ECDHE_RSA_WITH_AES_128_CBC_SHA256 - strong
|       TLS_ECDHE_RSA_WITH_AES_128_GCM_SHA256 - strong
|       TLS_ECDHE_RSA_WITH_AES_256_CBC_SHA - strong
|       TLS_ECDHE_RSA_WITH_AES_256_CBC_SHA384 - strong
|       TLS_ECDHE_RSA_WITH_AES_256_GCM_SHA384 - strong
|       TLS_RSA_WITH_AES_128_CBC_SHA - strong
|       TLS_RSA_WITH_AES_128_CBC_SHA256 - strong
|       TLS_RSA_WITH_AES_128_GCM_SHA256 - strong
|       TLS_RSA_WITH_AES_256_CBC_SHA - strong
|       TLS_RSA_WITH_AES_256_CBC_SHA256 - strong
|       TLS_RSA_WITH_AES_256_GCM_SHA384 - strong
|     compressors:
|       NULL
|_  least strength: strong

Nmap done: 1 IP address (1 host up) scanned in 6.45 seconds

```

becareful, perhaps you think you are TLS1.2 because your browser said TLS1.2 but...
example : aws.amazon.com
`nmap --script ssl-enum-ciphers -p 443 aws.amazon.com`

```
Starting Nmap 6.47 ( http://nmap.org ) at 2017-03-30 23:38 CEST
Nmap scan report for aws.amazon.com (54.239.31.69)
Host is up (0.11s latency).
PORT    STATE SERVICE
443/tcp open  https
| ssl-enum-ciphers:
|   SSLv3: No supported ciphers found
|   TLSv1.0:
|     ciphers:
|       TLS_ECDHE_RSA_WITH_AES_128_CBC_SHA - strong
|       TLS_RSA_WITH_3DES_EDE_CBC_SHA - strong
|       TLS_RSA_WITH_AES_128_CBC_SHA - strong
|     compressors:
|       NULL
|   TLSv1.1:
|     ciphers:
|       TLS_ECDHE_RSA_WITH_AES_128_CBC_SHA - strong
|       TLS_RSA_WITH_3DES_EDE_CBC_SHA - strong
|       TLS_RSA_WITH_AES_128_CBC_SHA - strong
|     compressors:
|       NULL
|   TLSv1.2:
|     ciphers:
|       TLS_ECDHE_RSA_WITH_AES_128_CBC_SHA - strong
|       TLS_ECDHE_RSA_WITH_AES_128_CBC_SHA256 - strong
|       TLS_ECDHE_RSA_WITH_AES_128_GCM_SHA256 - strong
|       TLS_RSA_WITH_3DES_EDE_CBC_SHA - strong
|       TLS_RSA_WITH_AES_128_CBC_SHA - strong
|       TLS_RSA_WITH_AES_128_CBC_SHA256 - strong
|       TLS_RSA_WITH_AES_128_GCM_SHA256 - strong
|     compressors:
|       NULL
|_  least strength: strong

Nmap done: 1 IP address (1 host up) scanned in 5.36 seconds

```

anyway...

good luke with your iot projects !

karenerobinson, 2018-08-08:
@dapug I'm curious if you solved this.

sourav2342, 2023-03-03:
hello, i was facing the same issue HTTPC_ERROR_CONNECTION_REFUSED  (-1) but i couldn't figure out what is the issue
and i new to this field can some one explain where i am going wrong this is my esp32 code

#include <WiFi.h>
#include <HTTPClient.h>

const char* ssid = "ssdsg";  // Enter SSID here
const char* password = "skdjfhs23";//.. password

//Your Domain name with URL path or IP address with path
const char* serverName = "http://216.22.45.3:3000/post_data";

unsigned long lastTime = 0;
unsigned long timerDelay = 2000;

void setup() {
  Serial.begin(115200);

  WiFi.begin(ssid, password);
  Serial.println("Connecting");
  while(WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.print("Connected to WiFi network with IP Address: ");
  Serial.println(WiFi.localIP());
 
  Serial.println("Timer set to 5 seconds (timerDelay variable), it will take 5 seconds before publishing the first reading.");
}

void loop() {
  //Send an HTTP POST 
  if ((millis() - lastTime) > timerDelay) {
    //Check WiFi connection status
    if(WiFi.status()== WL_CONNECTED){
      WiFiClient client;
      HTTPClient http;
 
      http.begin(client, serverName);
  
      http.addHeader("Content-Type", "application/x-www-form-urlencoded");
      // Data to send with HTTP POST
      String httpRequestData = "10.14";           
     
      int httpResponseCode = http.POST(httpRequestData);
      
      
     
      Serial.print("HTTP Response code: ");
      Serial.println(httpResponseCode);
        
      // Free resources
      http.end();
    }
    else {
      Serial.println("WiFi Disconnected");
    }
    lastTime = millis();
  }
}

i have deployed a node.js code in render which has a post request route and it has ssl certificate which plotes the 10.14 value on graph but it shows '-1' responsecode meaning not connected. can someone help me with this issue.

sourav2342, 2023-03-03:
hello, i was facing the same issue HTTPC_ERROR_CONNECTION_REFUSED  (-1) but i couldn't figure out 
and i new to this field can some one explain where i am going wrong, this is my esp32 code

#include <WiFi.h>
#include <HTTPClient.h>

const char* ssid = "ssdsg";  // Enter SSID here
const char* password = "skdjfhs23";//.. password

//Your Domain name with URL path or IP address with path
const char* serverName = "http://216.22.45.3:3000/post_data";

unsigned long lastTime = 0;
unsigned long timerDelay = 2000;

void setup() {
  Serial.begin(115200);

  WiFi.begin(ssid, password);
  Serial.println("Connecting");
  while(WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.print("Connected to WiFi network with IP Address: ");
  Serial.println(WiFi.localIP());
 
  Serial.println("Timer set to 5 seconds (timerDelay variable), it will take 5 seconds before publishing the first reading.");
}

void loop() {
  //Send an HTTP POST 
  if ((millis() - lastTime) > timerDelay) {
    //Check WiFi connection status
    if(WiFi.status()== WL_CONNECTED){
      WiFiClient client;
      HTTPClient http;
 
      http.begin(client, serverName);
  
      http.addHeader("Content-Type", "application/x-www-form-urlencoded");
      // Data to send with HTTP POST
      String httpRequestData = "10.14";           
     
      int httpResponseCode = http.POST(httpRequestData);
      
      
     
      Serial.print("HTTP Response code: ");
      Serial.println(httpResponseCode);
        
      // Free resources
      http.end();
    }
    else {
      Serial.println("WiFi Disconnected");
    }
    lastTime = millis();
  }
}

i have deployed a node.js code in render which has a post request route and it has ssl certificate which plotes the 10.14 value on graph but it shows '-1' responsecode meaning not connected.

sourav2342, 2023-03-03:
hello, i was facing the same issue HTTPC_ERROR_CONNECTION_REFUSED  (-1) but i couldn't figure out 
and i new to this field can some one explain where i am going wrong this is my esp32 code

#include <WiFi.h>
#include <HTTPClient.h>

const char* ssid = "ssdsg";  // Enter SSID here
const char* password = "skdjfhs23";//.. password

//Your Domain name with URL path or IP address with path
const char* serverName = "http://216.22.45.3:3000/post_data";

unsigned long lastTime = 0;
unsigned long timerDelay = 2000;

void setup() {
  Serial.begin(115200);

  WiFi.begin(ssid, password);
  Serial.println("Connecting");
  while(WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.print("Connected to WiFi network with IP Address: ");
  Serial.println(WiFi.localIP());
 
  Serial.println("Timer set to 5 seconds (timerDelay variable), it will take 5 seconds before publishing the first reading.");
}

void loop() {
  //Send an HTTP POST 
  if ((millis() - lastTime) > timerDelay) {
    //Check WiFi connection status
    if(WiFi.status()== WL_CONNECTED){
      WiFiClient client;
      HTTPClient http;
 
      http.begin(client, serverName);
  
      http.addHeader("Content-Type", "application/x-www-form-urlencoded");
      // Data to send with HTTP POST
      String httpRequestData = "10.14";           
     
      int httpResponseCode = http.POST(httpRequestData);
      
      
     
      Serial.print("HTTP Response code: ");
      Serial.println(httpResponseCode);
        
      // Free resources
      http.end();
    }
    else {
      Serial.println("WiFi Disconnected");
    }
    lastTime = millis();
  }
}

i have deployed a node.js code in render which has a post request route and it has ssl certificate which plotes the 10.14 value on graph but it shows '-1' responsecode meaning not connected. can someone help me with this issue.

d-a-v, 2023-03-03:
@sourav2342 this issue is closed.
This repository is about esp8266.
Please head to esp32 arduino repository and open a new issue there.