esp8266/Arduino #2558: ESP-12E Node MCU will resets automatically at every time and stack overflow

BharathTatikonda, 2016-09-29:
Basic Infos
Hardware

Hardware: ?ESP-12?
Core Version: ?2.1.0-rc2?
hi, i have started ESP12-E to work with the GSM module, in this when i was running the it should be reset automatically when some messages will receives,... then it reset and restarted from first so the function is not completed,,....so, please suggest me how to disable the watchdog timer,..... when i run the program the error will be like this///.........

Soft WDT reset

ctx: cont
sp: 3ffefc90 end: 3ffeff00 offset: 01b0

            stack>>>
            3ffefe50: 00000001 00000001 00000000 402030ea

            3ffefe60: 00000106 00000000 000005dc 00001388

            3ffefe70: 00000005 3ffe87c0 00000004 40202fe8

            3ffefe80: 3ffe8517 3ffeed24 3ffeee1c 00000000

            3ffefe90: 00000005 000005dc 3ffe8ba2 4020313e

            3ffefea0: 4010564a feefeffe 3ffeee1c 3ffeeed0

            3ffefeb0: 3fffdad0 3ffe8ba2 00000000 40202b77

            3ffefec0: 3fffdad0 3ffeebe0 3ffeee1c 40201fd4

            3ffefed0: feefeffe 00000000 3ffeeec8 4020213d

            3ffefee0: feefeffe feefeffe feefeffe 4020388c

            3ffefef0: feefeffe feefeffe 3ffeeee0 40100718

            <<<stack<<<

so, please suggest me how to disable the watchdog timer and controlling of stack overflow in the esp12-e module
Settings in IDE

Module: ?Generic ESP8266 Module?
Flash Size: ?4MB/1MB?
CPU Frequency: ?80Mhz?
Flash Mode: ?qio?
Flash Frequency: ?40Mhz?
Upload Using: ?OTA / SERIAL?
Reset Method: ?ck / nodemcu?

reaper7, 2016-09-29:
You are using any gms library or Your own functions for communication with gsm module?
Probably some functions waiting too long for response from gsm module and this is a reason of WDT reset.
You must search for "while loop" functions inside Your code(or gsm library) and insert yield(); inside.

BharathTatikonda, 2016-09-29:
yahh, i am using gsm libraries and i have also used the yield(); in while loops, it will ok means some what reduces the reset functioning but at some times the esp will going to reset,.... i want to modify that purely.....................

BharathTatikonda, 2016-09-29:
please tell me how to manage this reset one and stackover flow completely,..............

reaper7, 2016-09-29:
Please install ESPExceptionDecoder
howto:
https://github.com/me-no-dev/EspExceptionDecoder

BharathTatikonda, 2016-09-29:
ok i will try and then contect with u

BharathTatikonda, 2016-09-29:
after installing, i want to include in my program or i want to run in my controller or just save in to my library section...........

WereCatf, 2016-09-29:
You can't disable the watchdog completely, you have to fix the code to call yield().

BharathTatikonda, 2016-09-29:
ooh what is the main function of yield():  and where i should use the yield(); funtions

WereCatf, 2016-09-29:
yield() does some background-stuff that you don't have to know the specifics of, but calling it periodically keeps the watchdog-timer happy. If you have long while-loops or for-loops or other sections of code that take a long time to run then those sections will have to also call yield() every now and then in order not to trigger watchdog.

BharathTatikonda, 2016-09-29:
ok what about in the if-else statements..............

BharathTatikonda, 2016-09-29:
but probably in my code sometimes getting the complete actions but in sometimes the esp will be going to reset,..... i want to maintain the complete actions at everytime,,...how it is possible?,,,,..

WereCatf, 2016-09-29:
Like I said, use yield() in any sections of code that may take a long time to run. It's up to you to know what sections take a long time to run and which ones don't.

BharathTatikonda, 2016-09-30:
Ok how much time the watchdog timer will going to be reset the
microcontroller............

WereCatf, 2016-09-30:
See https://github.com/esp8266/Arduino/pull/2533/files

BharathTatikonda, 2016-10-19:
Hey esp8266/Arduino,

Discover high quality jobs and connect with recruiters on LinkedIn.

- THATIKONDA

Accept invite: https://www.linkedin.com/comm/start/accept-invitation?sharedKey=GGKgEMX6&invitationId=6194371956964421632&trk=eml-invite_guest-invite_guest-203-india_a&trkEmail=eml-invite_guest-invite_guest-203-india_a-null-0%7Eiuggdgu8%7Erk

You received an invitation to connect. LinkedIn will use your email address to make suggestions to our members in features like People You May Know. Unsubscribe here: https://www.linkedin.com/e/v2?e=0-iuggdgu8-rk&t=lun&midToken=AQFmrFVhGEd36Q&ek=invite_guest&loid=AQF7fIUYHX2odQAAAVfbUlaXYXPnB0tGgzLWnA73K3tPYP0sXAhlhHbwfff2UK0QL_IWCivXuz-bd7GwkhbaBeYIdYKbGgVMMcB7k9OVUQyHmtjnPUrh5ONuBZUXegLvXFIMS21noPNQ9oZOYMpdF4gUsBo1kfT6YMMAtUhuMowb7Za8gDGwhQSiQW7EvWF3hKjz&eid=0-iuggdgu8-rk

This email was sent to reply@reply.github.com.

If you need assistance or have questions, please contact LinkedIn Customer Service: https://www.linkedin.com/e/v2?e=0-iuggdgu8-rk&a=customerServiceUrl&ek=invite_guest

&copy; 2016 LinkedIn Corporation, 2029 Stierlin Court, Mountain View CA 94043. LinkedIn and the LinkedIn logo are registered trademarks of LinkedIn.

WereCatf, 2016-10-19:
DO NOT SPAM HERE.

BharathTatikonda, 2016-10-25:
THATIKONDA Bharath would like to connect on LinkedIn. How would you like to respond?

Accept: https://www.linkedin.com/comm/start/accept-invitation?sharedKey=GGKgEMX6&invitationId=6194371956964421632&trk=eml-first_guest_reminder_01-hero-121-accept_text&trkEmail=eml-first_guest_reminder_01-hero-121-accept_text-null-punv94%7Eiuozox2l%7Ebb

View THATIKONDA Bharath&#39;s profile: https://www.linkedin.com/comm/start/accept-invitation?sharedKey=GGKgEMX6&invitationId=6194371956964421632&trk=eml-first_guest_reminder_01-hero-3-profile_text&trkEmail=eml-first_guest_reminder_01-hero-3-profile_text-null-punv94%7Eiuozox2l%7Ebb

You received an invitation to connect. LinkedIn will use your email address to make suggestions to our members in features like People You May Know. Unsubscribe here: https://www.linkedin.com/e/v2?e=punv94-iuozox2l-bb&t=lun&midToken=AQFmrFVhGEd36Q&ek=first_guest_reminder_01&li=123&m=unsub&ts=HTML&eid=punv94-iuozox2l-bb&loid=AQE70XdBRxNmEgAAAVf7Fbnm41UvMMziUIrcoyTEieyuJEcZalGP3LjWoMsrtlLKv6hsbfdkIyW6kiab6xecE3tDUdXYy7OLLrno_0Z_Q9m3ODIf2r48OAxZ8Seko3GGPExy67hKbbBSmznPWt9XueB7pYjtzI8oTTuch52ximtY9BSR2X6k87Q1vBzj0vXbf0oU

This email was sent to reply@reply.github.com.

If you need assistance or have questions, please contact LinkedIn Customer Service: https://www.linkedin.com/e/v2?e=punv94-iuozox2l-bb&a=customerServiceUrl&ek=first_guest_reminder_01

&copy; 2016 LinkedIn Corporation, 2029 Stierlin Court, Mountain View CA 94043. LinkedIn and the LinkedIn logo are registered trademarks of LinkedIn.

BharathTatikonda, 2016-11-01:
Hi ,

THATIKONDA Bharath would like to connect on LinkedIn. How would you like to respond?

Accept: https://www.linkedin.com/comm/start/accept-invitation?sharedKey=GGKgEMX6&invitationId=6194371956964421632&trk=eml-second_guest_reminder_01-hero-120-accept_text&trkEmail=eml-second_guest_reminder_01-hero-120-accept_text-null-punv94%7Eiuz99qw8%7Ef

View THATIKONDA Bharath&#39;s profile: https://www.linkedin.com/comm/start/accept-invitation?sharedKey=GGKgEMX6&invitationId=6194371956964421632&trk=eml-second_guest_reminder_01-hero-3-profile_text&trkEmail=eml-second_guest_reminder_01-hero-3-profile_text-null-punv94%7Eiuz99qw8%7Ef

You received an invitation to connect. LinkedIn will use your email address to make suggestions to our members in features like People You May Know. Unsubscribe here: https://www.linkedin.com/e/v2?e=punv94-iuz99qw8-f&t=lun&midToken=AQFmrFVhGEd36Q&ek=second_guest_reminder_01&li=122&m=unsub&ts=HTML&eid=punv94-iuz99qw8-f&loid=AQGWtjw_1aI75QAAAVggBtp0z2rv_LoMzlUYbk3p84VjjYLvSFT16FDeLbutoc0XurZhjhCiXqTqQ2931YTnmZs4uwHylNnMusVgQ_Phh4TXW4cBTb_osSuty90miMocp-Te-4lm1S5rTfUf9_M76QVJDHrxbXxaOyMasPQ-GT4TchuPCZH9Sh3Xln5G8v6BBn0O

This email was sent to reply@reply.github.com.

If you need assistance or have questions, please contact LinkedIn Customer Service: https://www.linkedin.com/e/v2?e=punv94-iuz99qw8-f&a=customerServiceUrl&ek=second_guest_reminder_01

&copy; 2016 LinkedIn Corporation, 2029 Stierlin Court, Mountain View CA 94043. LinkedIn and the LinkedIn logo are registered trademarks of LinkedIn.

satech-domo, 2017-09-04:
One suggestion to get info about your ESP when itgoing wrong 
set up the terminal speed at 74880 and restart ESP. 
You may get different messages to help your diagnostic.

devyte, 2017-10-12:
Closing per #3655 .

closed 2017-10-12 by devyte

sunil11101, 2019-04-12:
it can be a problem of power connect battery which can supply sufficient amount of power with no voltage sage