esp8266/Arduino #2559: How do I finish using and dispose of mdns after I used it to resolve addresses

lorneb, 2016-09-29:
Question Only:

I'm using mDNS to resolve sope ip addresses and ports for two services, once that done it no longer need it.  What would be the best practice for disposing of it?

Thanks

WereCatf, 2016-09-29:
I dunno if it would be the best practice, but you could call its destructor as `MDNS.~MDNSResponder()` to clean it up, or disable the global MDNSResponder-instance "MDNS" altogether and allocate/deallocate it yourself when needed.

devyte, 2017-10-12:
@lorneb previous comment is correct: disable the global object, and instantiate your own, either dynamically or scoped.
In any case, this is not the right place to ask such questions. Closing per #3655 .

closed 2017-10-12 by devyte