esp8266/Arduino #2560: SPIFFS crashes after adding unrelated code when saving configuration

helpme1, 2016-09-29:
Module:  			NodeMCU
Flash Size: 		4MB
CPU Frequency:		80Mhz
Flash Mode:			dio?
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		 nodemcu

I am encountering a strange behavior. I am using ESP8266 arduino SPIFFS to store configuration settings. Here is the relevant part of mycode; 

```
void loop()
{
      handleUartRxOk();
}

void handleUartRxOk() {
   String cmd;
   
   char charBuff[3200];
   char char_print[50];
   static bool terminatorReceived = false;

   char incomingChar = 0;   // for incoming serial data

   if (Serial.available()) {
      incomingChar = Serial.read();
      saveChar(incomingChar);

      if (incomingChar == '\r') {
         terminatorReceived = true;
      }

      if (terminatorReceived) {
         buffer[buffer_index - 1] = '\0';
         cmd = String(buffer);
         if (cmd == "XXX") {
            ConfigSettings.ssid = "SSID_XX";
            ConfigSettings.password = "PASSWORD_XX";
            saveConfig();
         } 
         
         buffer_index = 0;
         terminatorReceived = false;
      }
   }
} 
```

In the above code, the UART will receive a command "XXX\r", then run saveConfig() which will save the config parameters ssid and parameters into SPIFSS. This code works perfectly fine until I add more code which is totally unrelated.

This is how the new code looks like. 

```
void handleUartRxOk() {
   String cmd;
   
   char charBuff[3200];
   char char_print[50];
   static bool terminatorReceived = false;

   char incomingChar = 0;   // for incoming serial data

   if (Serial.available()) {
      incomingChar = Serial.read();
      saveChar(incomingChar);

      if (incomingChar == '\r') {
         terminatorReceived = true;
      }

      if (terminatorReceived) {
         buffer[buffer_index - 1] = '\0';
         cmd = String(buffer);
         if (cmd == "XXX") {
            ConfigSettings.ssid = "SSID_XX";
            ConfigSettings.password = "PASSWORD_XX";
            saveConfig();
         } 

         //Why does adding this else statement cause saveConfig() to crash?
         else {
               strcat(charBuff, cmd.c_str());
            }   
         
         buffer_index = 0;
         terminatorReceived = false;
      }
   }
} 
```

After adding the additional else clause, sending "XXX\r" to the UARt and causing saveConfig() will cause an exception error. This is puzzling since the new code did not even get to run.

The exception error is as below;

_Exception (3):

epc1=0x401002f0 epc2=0x00000000 epc3=0x00000000 excvaddr=0x400072f6 depc=0x00000
000

ctx: sys 

sp: 3fff06b0 end: 3fffffb0 offset: 01a0

>>>stack>>>

3fff0850: 00000000 00000016 3fff2878 40202319 
3fff0860: 3fff2f34 07050000 00000000 00000000 
3fff0870: 3fff2874 3ffe8aa8 3fff08e0 401004d8 
3fff0880: 3fff1890 00000000 00000000 4020a228 
3fff0890: 3fff2874 3ffe8aa8 3fff08e0 4020ab35 
3fff08a0: 00000000 00000000 3ffe8aa3 3fff0738 
3fff08b0: 00000016 00000005 00000001 40209c81 
3fff08c0: 3ffe90e4 3fff0904 3fff07a4 3fff0738 
3fff08d0: 3ffe8aa8 3fff1a5c 3fff0b60 402099f8 
3fff08e0: feefeffe feefeffe 00000002 00000005 
3fff08f0: 3fff07a4 3fff0904 3fff0b60 402074f8 
3fff0900: 3ffe8cc0 3fff0900 3fff090c 3fff092c 
3fff0910: 3ffe8a6c 00000002 3fff091c 2d525744 
3fff0920: 2d313239 46313439 feefef00 3fff0948 
3fff0930: 3ffe8a74 00000002 3fff093c 474a3473 
3fff0940: 38664d63 40100a00 3fff096c 3ffe8a80 
3fff0950: 00000002 3fff0958 74736574 2e383631 
3fff0960: 6f6f6f6d 6d6f632e 00000000 00000000 
3fff0970: 3ffe8a8c 00000002 3fff097c 0066666f 
3fff0980: 40101be1 3ffed400 3ffedaf0 40100ece 
3fff0990: fffffff2 00b86301 3ffedde0 40101d70 
3fff09a0: 3ffea1f0 00000000 00000000 3ffedaf0 
3fff09b0: 3ffedaf0 40100a62 00000001 00000000 
3fff09c0: 00000000 4010082f 00000000 3ffedac0 
3fff09d0: 00000001 00000000 00000002 40100ece 
3fff09e0: 60000000 0000000d 3ffedac0 40100ece 
3fff09f0: 00000000 00000000 0000001f 00000022 
3fff0a00: 3fffc200 40107838 3fffc250 4000050c 
3fff0a10: 40000f83 00000030 00000012 ffffffff 
3fff0a20: 401004ba 00000023 00000000 00000019 
3fff0a30: 000002a0 000000a0 00000000 00000000 
3fff0a40: 00000ca0 00000002 00002ac6 00000016 
3fff0a50: 00000003 000000b0 3fff0780 00000030 
3fff0a60: 00000000 00000000 0000001f 40104cad 
3fff0a70: 4000050c 00000010 3fff0780 00000030 
3fff0a80: 40000f83 00000030 00000012 ffffffff_

lorneb, 2016-09-29:
Do you really need this buffer to be this big and not global?
char charBuff[3200];

sticilface, 2016-09-29:
I second that.  That will be on the stack which is only 4K.  You are better using new and delete to manage anything over 100 bytes TBH.

helpme1, 2016-09-30:
Thanks for the tips. Can I confirm with you that the maximum stack size is 4K? If yes, where is this defined? Can it be changed?

devyte, 2017-10-12:
@helpme1 previous comments are coorect: the issue is most likely a stack blowup due to that huge buffer. Yes, the stack size can be increased, but I don't recommend it, because it reduces your available heap. Instead, declare the buffer in global scope, create it dynamically, or use a managed object like std::vector, std::string, or String.
Closing as user error.

closed 2017-10-12 by devyte