esp8266/Arduino #2561: Flashing binary file with SPIFFS

helpme1, 2016-09-29:
I am using ESP8266 Arduino with SPIFFS to store configuration settings. 

This command is used for flashing the binary firmware.bin into ESP8266 for Arduino.
esptool.exe -cd nodemcu -cb 115200 -cp COM3 -ca 0x00000 -cf firmware.bin 

This command is used for flashing the binary firmware.nin into ESPP8266 using the SMING framework.

esptool.exe -p COM3 -b 115200 write_flash -ff 40m -fm dio -fs 4m 0x00000 0x00000.bin 0x09000 0x09000.bin 0x44000 spiff_rom.bin

There are some puzzling aspects in the command used for Arduino flash download.

- Why isn't there a separate bin file for SPIFFS for Arduino? I noticed the tool "mkspiffs.exe" in esp8266\tools\mkspiffs folder is not used during compilation of the bin file. Why is it not used?

- How does Arduino esptool know which address to use for allocating to SPIFFS file system address?

I would like to understand the process of programming binary file as I encountered strange behavior with SPIFFS. 

Can someone help? Thank you very much.

lorneb, 2016-09-29:
Hi @helpme1 I use this project to upload files to the SPIFFS area. 

https://github.com/esp8266/arduino-esp8266fs-plugin
In the Arduino IDE under tools - Flash Size you can see the allocation between code and SPIFFS e.g 4M (3M SPIFFS)

Hope that helps

WereCatf, 2016-09-29:
@helpme1 The Arduino IDE doesn't flash SPIFFS unless you specifically tell it to flash SPIFFS. This is because, well, it doesn't make sense to always wipe SPIFFS's contents clean, like e.g. people may have configuration-settings and stuff there and they only need to update the sketch, not the files on SPIFFS. Like @lorneb there said, there is a separate command for flashing SPIFFS.

Also, in the IDE you select how to divide the Flash between code-area and SPIFFS, like e.g. on Nodemcu (4MB Flash) you get the options to choose between 1MB code-area/3MB SPIFFS, or 3MB code-area/1MB SPIFFS. That's how it knows the address.

helpme1, 2016-09-30:
Thanks. Clearer now.

closed 2016-09-30 by helpme1