esp8266/Arduino #2562: NodeMCU 12E keeps trying to reconnect instead of running arduino code

Imara90, 2016-09-29:
### Basic Infos

#### Hardware
Hardware:			ESP-12E module
Core Version:      	NodeMCU 1.0

### Description
I used the connectWifi() function to connect to my wifi network on my phone once to test whether it was working in the Arduino Sketch. However, when I wanted to take it out of the code again, to continue the code without WiFi, the module keeps trying to reconnect to my phone and does not run any of the arduino code anymore. I left out most of the code because it's irrelevant to the problem and quite long.

### Settings in IDE

Module:  			Node MCU ESP-12E module
Flash Size: 		4MB
CPU Frequency:	80Mhz
Flash Frequency:	80Mhz
Upload Using:		SERIAL 115200
Reset Method:		nodemcu button

### Sketch

void setup() {
  // put your setup code here, to run once:
  Serial.begin(9600);
  Serial.setDebugOutput(true);
  Serial.println(F("Welkom bij de buitenbloem"));

  // set up input
  pinMode(KNOP, INPUT);
  pinMode(PIR, INPUT);
  pinMode(BUSY, INPUT);

  /*
    // calibrate the pir
    digitalWrite(PIR, LOW);
    //give the sensor some time to calibrate
    Serial.print("calibrating sensor ");
    for (int i = 0; i < calibrationTime; i++) {
    Serial.print(".");
    delay(1000);
    }
    Serial.println(" done");
    Serial.println("SENSOR ACTIVE");
    delay(50);
  */

  // set up output
  pinMode(KNOPLED, OUTPUT);
  pixels.begin(); // This initializes the NeoPixel library.
  setPixels(pixelsOff);

  // set up mp3 player
  mySerial.begin(9600);
  mp3_set_serial (mySerial);  //set Serial for DFPlayer-mini mp3 module
  mp3_set_volume (LVOL);
  delay(1000);  //wait 1ms for mp3 module to set volume

  // connect to the wifi
  connectWifi();
  analogWriteRange(255);
}

void connectWifi()
{
  // if it still connected to another network, disconnect
  if (WiFi.status() == WL_CONNECTION_LOST || WiFi.status() == WL_CONNECTED){
    WiFi.disconnect();
    while(WiFi.status() != WL_DISCONNECTED){
      delay(1000);
      Serial.println("disconnect");
    }
  }
  
  Serial.print("Connecting to " + *MY_SSID);
  WiFi.begin(MY_SSID, MY_PWD);
  while (WiFi.status() != WL_CONNECTED) {
    delay(1000);
    Serial.print(".");
  }
  setPixels(pixels.Color(0,255 , 0));
  delay(500);
  setPixels(pixelsOff);
  delay(500);
  
  Serial.println("");
  Serial.println("Connected");
  Serial.println("");
}//end connect

### Debug Messages

I enabled the serial debug output and it says:

I%ÿ ÿ5BþÿWelkom bij de buitenbloem
teerdscandone
f 0, ..scandone
no Xperia Z3 Compact_8af8 found, reconnect after 1s
reconnect
f -180, ..scandone
no Xperia Z3 Compact_8af8 found, reconnect after 1s
reconnect
f r0, ..scandone
no Xperia Z3 Compact_8af8 found, reconnect after 1s
reconnect

messages here
Also if anyone know why the serial monitor prints the initial scribblings, would be greatly appreciated. 

These are the pins I use:
// pin definities
// Serial uses UART0, WHICH IS MAPED TO GPIO1 (TX) and GPIO3 (RX)
#define KNOP D3    // GPIO
#define KNOPLED D4  // GPIO
#define PIR D0      // GPIO
#define NEO D6      // GPIO04
#define BUSY D5     // GPIO05 TODO weghalen?

SoftwareSerial mySerial(D1, D2); // RX, TX

krzychb, 2016-09-29:
Hi @Imara90,

This is expected behavior, check description of [station mode](https://github.com/esp8266/Arduino/tree/master/doc/esp8266wifi#station).

To stop it you can use e.g. [`disconnect()`](https://github.com/esp8266/Arduino/blob/master/doc/esp8266wifi/station-class.md#disconnect) or [`WiFi.mode(WIFI_OFF)`](https://github.com/esp8266/Arduino/blob/master/doc/esp8266wifi/generic-class.md#mode).

One time use of the above functions will be enough. 
New settings will be save to flash and module will not be able to reconnect. 

Yet another option is to wipe flash.

Imara90, 2016-10-04:
Hi @krzychb, 

Thanks, this worked like a charm. Sorry for the easy question, but there's just so much info to process when starting with the ESP ;).
I used the WiFi.disconnect() and the code runs the normal sketch for debugging. However I will re-enable the autoconnect when I use the code for the actual application. 

- Imara

closed 2016-10-04 by Imara90