esp8266/Arduino #2564: IDE and setup for freeRTOS based code in ESP12E

sarwadenj, 2016-09-29:
Can anybody tell me the steps to compile RTOS sdk based program and how to upload the program to the ESP12E ? I am very new and familiar with ESP8266 programming in arduino IDE only.
Also somewhere i seen that Arduino IDE is also used to write, compile the freeRTOS based code for ESP12E.

Thanks in advance.

sarwadenj, 2016-09-30:
Hi,
Anyone have any IDEA about this. Please give me suggestions.

Thanks in advance.

rahuldeo2047, 2017-05-31:
Can you please link to the 'somewhere' ?

> Also somewhere i seen that Arduino IDE is also used to write, compile the freeRTOS based code for ESP12E.

devyte, 2017-10-12:
@sarwadenj wrong repo, this is the NON-OS firmware (i.e.: no FreeRTOS).
Closing.

closed 2017-10-12 by devyte