esp8266/Arduino #2565: hash function suggestion:  add BLAKE2s, decommission MD5

acollaborator, 2016-09-30:
Feature request:   Implement BLAKE2s hash functions in esp8266 Arduino core and then quickly transition away from MD5 in order to improve OTA Update's resistance to compromise.  Exploits are reported to exist that create a new file with the same MD5 hash as a target file.

overview of BLAKE2 versus MD5
https://leastauthority.com/blog/BLAKE2-harder-better-faster-stronger-than-MD5.html

code and crypto function benchmark results for Arduino and ARM
https://rweather.github.io/arduinolibs/crypto.html

WereCatf, 2016-09-30:
MD5 is only used to verify the OTA-update didn't get corrupted in-flight, it's not a security-feature. If you want security you shouldn't be using Arduino OTA anyways as the traffic is all unencrypted.

devyte, 2017-10-12:
@acollaborator I agree with @WereCatf . If security is a concern, you shouldn't be using ArduinoOTA. If you're using httpClient, you can use secure sockets. If you're using the webserver, you can set up a secure server (if a limited one).
Although interesting, I don't see a need for this.
Closing.

closed 2017-10-12 by devyte