esp8266/Arduino #2566: error: pll_cal exceeds 2ms!!!

mkeyno, 2016-09-30:
---------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Hardware:			 ESP-12F
Core Version:      	last update 2.2

### Description

just after soldering the fresh ESP-12F on the board and upload the helloworld example of library , its throw 
`error: pll_cal exceeds 2ms!!!`
in every couple of second without any crash and exception, instead no connection lunched whereas serial print shows its work and gain the ip address , I've  checked couple of lib example as well and also regard to this[ link](http://bbs.espressif.com/viewtopic.php?t=2114) put the `wifi_set_sleep_type(NONE_SLEEP_T); ` inside setup but no success
is it possible the module had broken ?

### Settings in IDE

Module:  			?Generic ESP8266 Module?
Flash Size: 		?2MB/1MB?
CPU Frequency:		?80Mhz?
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		?OTA / SERIAL?
Reset Method:		?ck / nodemcu?

error: pll_cal exceeds 2ms!!!

middelink, 2016-10-01:
This comes directly out of the SDK libphy.a, so I don't think it is an Arduino library issue. You might want to try and wipe the entire flash area, in case there is some residual settings left. Not knowing your environment I don't know how you want to do that though.

Good luck.

mkeyno, 2016-10-02:
yes I did both of those example and I thinks its wifi bricked  but its other function work well and can use for no wifi MCU chip

arsalanhassanawan, 2016-10-03:
Hi

I had the same issue and I was able to resolve it by replacing the libphy.a file (as brought into attention by @middelink) in the SDK that comes with the arduino libraries for ESP8266.
As of now, the ESP8266 arduino libraries come with SDK version 1.5.3_16_04_18. I downloaded the SDK version 2.0.0_16_08_10 from the Espressif website from here: [http://bbs.espressif.com/viewtopic.php?f=46&t=2451](url). I replaced the old file "C:\Users\<user>\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\tools\sdk\lib\libphy.a" with the new one "ESP8266_NONOS_SDK_V2.0.0_16_08_10.zip\ESP8266_NONOS_SDK\lib\libphy.a". I compiled and uploaded my arduino code again and the issue was resolved! :)
Hope it helps!

mkeyno, 2016-10-03:
@middelink  , @arsalanhassanawan , I change my usb-serial cable and its work fine now , this is so odd , is not ? 
also for some sketch always throw exception and serial garbage  whereas with fix usb-serial module work fine
how come serial hardware links can affect on sketch ?

middelink, 2016-10-03:
Well, actually, if you take the power from USB as well, it might be that the other cable simply has too much voltage drop @300mA peak. If you have a good scope, you can verify. Or slap a big ass capacitor on your circuit :)

3mp3ri0r, 2017-01-12:
Hi all. To get away "error: pll_cal exceeds 2ms!!!", just let esp8266 power pin got 3.3v power before ch_phd pin. The "error: pll_cal exceeds 2ms!!!" won't bother you anymore. Hahaha.

McMajan, 2017-03-03:
@arsalanhassanawan: thank you!!!! After 2 day of headache you solution works!!!!

hongo2, 2017-09-16:
@3mperior you are the man bro.

devyte, 2017-09-16:
All, the latest git is already on sdk 2.0.0, so it should be using the libphy referenced here. Please give that a try.
You should not be just replacing the libphy in a prior version of the core. The sdk 2.0.0 requires certain wifi radio calibration calls, which I believe don't exist in prior versions of the core.

@hongo2 please edit your post and add markup, the schematic isn't clear  as it is.

hongo2, 2017-09-16:
hi @devyte check this out:

esp8266 nodemcu error: pll_cal exceeds 2ms!!! schematics solution
 
[image]

devyte, 2017-09-16:
I see that you're supplying 5V to the Vin, and then also using a resistor
dividor connected on the 3.3V pin. Am I reading that right?
I'm not an expert on wiring the nodemcu, but if I am reading it right, I
don't think you're supposed to. Supplying 5V to Vin powers the on board
regulator, which in turn gives you the 3.3V on the pin. If so, adding the
resistor divider only loads the regulator output, which worsens the voltage
for the ESP itself. That can mess with the wifi radio.

Anyways, please report here the behavior with latest git to see if the
pll_cal error is gone.

hongo2, 2017-09-16:
@devyte 
 yes you are reading right, a cuple of 100k resistors will not load anything.

after I added that divider the pll_cal error is gone. tried 1000 times,

before that, the nodemcu module gave me 100% of times pll_cal error when unplug usb.
after this hardware, mod the error just disappear.

please try this wiring and let me know the results, you will not burn anything.

devyte, 2017-10-12:
@hongo2 was this resolved with latest git?

hongo2, 2017-10-12:
yes it was, but another error came up when connect more than 4 clients, i
get an error saying max_connection it does not happen with the release

devyte, 2017-10-12:
@hongo2 from [here](https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/src/ESP8266WiFiAP.h):
`        bool softAP(const char* ssid, const char* passphrase = NULL, int channel = 1, int ssid_hidden = 0, int max_connection = 4);
`

hongo2, 2017-10-13:
thanks @devyte 

I am setting up AP like this...,
works great on release, but on latest git gives max_connection error, what should I do to use latest git on both devices AP and Client ?

struct softap_config config;
wifi_softap_get_config(&config);   // Get config first.
config.max_connection = 32;        // how many stations can connect to ESP8266 softAP at most.
wifi_softap_set_config(&config);   // Set ESP8266 softap config

WiFi.mode(WIFI_AP_STA);
WiFi.softAP(apName,apPassword);

devyte, 2017-10-13:
@hongo2 except for very specific cases, you should NOT be mixing SDK and Arduino calls. In this case, this:
```
struct softap_config config;
wifi_softap_get_config(&config); // Get config first.
config.max_connection = 32; // how many stations can connect to ESP8266 softAP at most.
wifi_softap_set_config(&config); // Set ESP8266 softap config
```

is getting overriden by this:
`WiFi.softAP(apName,apPassword);
`
I showed you the signature fo the softAP() method for a reason. Just do this:
`WiFi.softAP(apName, apPassword, 1, 0, 8);
`
that will setup the AP on channel 1, not hidden, and allow 8 clients. It's not a good idea to go above 8 due to mem constraints.

I think the original issue has been resolved, and we've strayed far beyond into code support, so I'm closing now.

closed 2017-10-13 by devyte

lattice0, 2024-01-03:
got it on esp32c6, no relation to arduino, but wifi/zigbee stopped working

DanielTerletzkiy, 2024-03-24:
also got this on a new DFROBOT Firebeetle 2 esp32c6, there is not a lot of info for this message...

lattice0, 2024-03-24:
@DanielTerletzkiy I got it on 2 other brand new esp32c6, the. The third one worked. When it gets this error it never recovers. Looks like something burned, I don't know.

DanielTerletzkiy, 2024-03-24:
@lattice0 uff this is not nice... i discovered that software reset with abort or ESP.restart does not work, _but_ using the hardware reset button instantly fixes this. I am thinking of attaching a cable to one of the gpios and the rst pin to reset the board like a hardware reset...
Also so far it only happens when using my battery connected to it, which currently sits at "45%" usable charge, i will run this test again while connected to a power source tonight and report back if it still gets stuck in this loop, if that fails, i will consider the hardware reset.

I also created a dfrobot forum post here: https://www.dfrobot.com/forum/topic/334203