esp8266/Arduino #2567: New signatures to ESP8266WebServer::sendContent

wellrats, 2016-10-01:
Hi all,
Would it be possible in next version of this lib insert another signatures for
ESP8266WebServer::sendContent as follow

```
void ESP8266WebServer::sendContent(const char* content, size_t len);
void ESP8266WebServer::sendContent(const char* content)
```

Useful when parsing a ```char*``` buffer like a HTML template from a file, and sending little pieces of this buffer to client after parsed. Call ``` <server_instance>.client().write( ... )``` directly is not an option, because final length of parsed buffer is unknown and encoding is ```chunked```

Thanks in advance

The implementation.

```
void ESP8266WebServer::sendContent(const char* content, size_t len) {
  const char * footer = "\r\n";
  if(_chunked) {
    char * chunkSize = (char *)malloc(11);
    if(chunkSize){
      sprintf(chunkSize, "%x%s", len, footer);
      _currentClient.write(chunkSize, strlen(chunkSize));
      free(chunkSize);
    }
  }
  _currentClient.write(content, len);
  if(_chunked){
    _currentClient.write(footer, 2);
  }
}

void ESP8266WebServer::sendContent(const char* content) {
  sendContent(content, strlen(content));
}

void ESP8266WebServer::sendContent(const String& content) {
  sendContent(content.c_str(), content.length());
}
```

luc-github, 2016-10-01:
Actually you do not need to specify size when sending if you use `setContentLength(CONTENT_LENGTH_UNKNOWN);` you just need to finish your sending by `sendContent("");`

you can see how it can be used with template on my project : https://github.com/luc-github/ESP3D/blob/master/esp3d/webinterface.cpp#L214-L394

one send part is here FYR: https://github.com/luc-github/ESP3D/blob/master/esp3d/webinterface.cpp#L379-L392

wellrats, 2016-10-01:
I understand that.

I just think it's not necessary duplicate my big ```char *``` buffer, turning it into a ```String``` to call ```sendContent```.

luc-github, 2016-10-01:
Sorry I misunderstood the question - but I would say why duplicate ? why not use `String` only instead of big `char *` buffer, this is what I do reading template files and processing them

wellrats, 2016-10-01:
OK. no problem.
The point is, more signatures means more flexibility.
The code posted above is working to ```String``` and ```char*```

Just a suggestion.

devyte, 2017-10-12:
@wellrats could you please make a PR with your suggestion?

PR #6341 (2019-07-25): Add plain char* signatures WebServer::sendContent
Fixes #2567

Allow the web server to send plain C strings instead of requring they be
encapsulated inside a String class object.  Saves memory vs. having to
convert C strings to Strings (i.e. duplication of data), overloads on
the efficient sendContent_P(char*) methods.
commit aa0bc337 (2019-07-25): Add plain char* signatures WebServer::sendContent (#6341)

Fixes #2567

Allow the web server to send plain C strings instead of requring they be
encapsulated inside a String class object.  Saves memory vs. having to
convert C strings to Strings (i.e. duplication of data), overloads on
the efficient sendContent_P(char*) methods.
closed 2019-07-25 by d-a-v