esp8266/Arduino #2568: ESP8266HTTPUpdateServer - cannot update - exception

exander77, 2016-10-02:
``
Update: HAHttpRelay.ino.bin
sleep disable
..............................................................................................................................Update Success: 257168
Rebooting...

Exception (2):
epc1=0x3ffe8c17 epc2=0x00000000 epc3=0x00000000 excvaddr=0x3ffe8c14 depc=0x00000000

ctx: cont 
sp: 3fff0b50 end: 3fff0fb0 offset: 01a0

>>>stack>>>
3fff0cf0:  00001328 00000000 3fff0df0 00000030  
3fff0d00:  0a2e2e2e 00000000 00000001 3fff0df0  
3fff0d10:  00000010 00000007 0000001c 401006d8  
3fff0d20:  3fff0d60 3fff0d30 00000008 4020851a  
3fff0d30:  3ffefe9c 000002e5 000002e5 4010020c  
3fff0d40:  00000000 00000004 3fff0da0 3fff0db1  
3fff0d50:  40107080 00000000 3fff2719 00000010  
3fff0d60:  00000025 00000000 000000a5 4021c9a5  
3fff0d70:  40203cb9 00000000 00000000 3fff0d00  
3fff0d80:  3ffefe9c 000001af 000001af 4010020c  
3fff0d90:  3fff0db0 3fff0da0 00000004 401008c8  
3fff0da0:  00000000 3ffe8d0d 3fff2719 00000000  
3fff0db0:  000000a5 ff000000 00000314 4010020c  
3fff0dc0:  00000010 3ffe84b7 00000000 4010031d  
3fff0dd0:  3fff0e40 000016e8 3ffefe9c 40100426  
3fff0de0:  3ffe87c3 00000025 3fff26f4 40100537  
3fff0df0:  401070ad 00000000 00000010 401006c2  
3fff0e00:  0000003a 3fff0e6c 3fff0e60 4020851a  
3fff0e10:  3ffe87c3 00000000 3fff0e60 40208572  
3fff0e20:  3fff0e40 3fff0e30 3fff0e60 402085a2  
3fff0e30:  3ffe87c3 3fff272c 3fff0e60 402085f0  
3fff0e40:  3ffe87c3 a5a50010 0000000c 40206f40  
3fff0e50:  3ffe87c3 3fff272c 3fff26f4 40204791  
3fff0e60:  00000000 00000000 00000000 3fff28cc  
3fff0e70:  0000003f 00000034 00000354 40204d94  
3fff0e80:  00000000 3ffe84d8 3ffe84b7 401006d8  
3fff0e90:  0000001c 3ffef448 3fff0ef0 40204d94  
3fff0ea0:  00000002 00000001 3fff26a4 40203061  
3fff0eb0:  3ffefe9c 00000314 00000314 4010020c  
3fff0ec0:  00000002 00000001 3fff28a4 401008c8  
3fff0ed0:  00000000 00000001 3fff265c 40204d8a  
3fff0ee0:  3fff265c 3ffef484 3fff265c 40204dc6  
3fff0ef0:  00000000 00000000 00000000 402086c8  
3fff0f00:  3fff265c 3ffef484 3ffef448 40204e59  
3fff0f10:  3fff1d84 0000000f 00000007 40203bcc  
3fff0f20:  00000001 0003ed64 3ffefd14 00000000  
3fff0f30:  00000002 0000000c 000003e8 40206951  
3fff0f40:  00000000 00000000 3ffef448 3ffeff80  
3fff0f50:  00000001 3ffef468 3ffef448 402050df  
3fff0f60:  3ffe8658 00000000 000003e8 3ffeff90  
3fff0f70:  00000000 3fff2934 00000001 402010a3  
3fff0f80:  3fffdad0 00000000 3ffeff79 40202bc4  
3fff0f90:  3fffdad0 00000000 3ffeff79 40207054  
3fff0fa0:  feefeffe feefeffe 3ffeff90 40100958  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)
`

### Basic Infos

#### Hardware
Hardware:			?ESP-1?
Core Version:      	?2.1.0?

### Description

Problem description

### Settings in IDE

Module:  			?Generic ESP8266 Module?
Flash Size: 		?1MB/512k?
CPU Frequency:		?80Mhz?
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		?SERIAL?
Reset Method:		?ck?

closed 2016-10-02 by exander77

pieman64, 2016-10-02:
Notice the issue is closed @exander77, did you sort out the problem?

How big is your sketch?

exander77, 2016-10-02:
Quite small, I found out that the new sketch really gets flash, even though the exception. Also my curl hangs for some reason, maybe the connection is not correctly closed before restart.