esp8266/Arduino #2570: ESP8266WebServer Exception(3)

RcrdBrt, 2016-10-02:
Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.md

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.md

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Hardware:			ESP-12E
Core Version:      	2.3.0

### Description

Problem description

### Settings in IDE

Module:  			NodeMCU 1.0 (ESP-12-E)
Flash Size: 		4MB
CPU Frequency:		80MHz
Upload Using:		Arduino IDE

### Sketch

```
/* DHTServer - ESP8266 Webserver with a DHT sensor as an input

   Based on ESP8266Webserver, DHTexample, and BlinkWithoutDelay (thank you)

   Version 1.0  5/3/2014  Version 1.0   Mike Barela for Adafruit Industries
*/
#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>
#include <DHT.h>
#define DHTTYPE DHT22
#define DHTPIN  5
#define red 4

const char* ssid     = "XXXXXXX";
const char* password = "XXXXXXXXXX";

ESP8266WebServer server(3004);

// Initialize DHT sensor
// NOTE: For working with a faster than ATmega328p 16 MHz Arduino chip, like an ESP8266,
// you need to increase the threshold for cycle counts considered a 1 or 0.
// You can do this by passing a 3rd parameter for this threshold.  It's a bit
// of fiddling to find the right value, but in general the faster the CPU the
// higher the value.  The default for a 16mhz AVR is a value of 6.  For an
// Arduino Due that runs at 84mhz a value of 30 works.
// This is for the ESP8266 processor on ESP-01
DHT dht(DHTPIN, DHTTYPE, 26); // 11 works fine for ESP8266-01

float humidity, temp_f;  // Values read from sensor
// Generally, you should use "unsigned long" for variables that hold time
unsigned long previousMillis = 0;        // will store last temp was read
const long interval = 2000;              // interval at which to read sensor

void handle_root() {

  getWeather();
  server.send(200, "application/json", "{\"temperature\": \"" + String(temp_f) + "\", " + "\"humidity\": \"" + String(humidity) + "\"}");
  delay(2000);
}

void setup(void)
{
  // You can open the Arduino IDE Serial Monitor window to see what the code is doing
  Serial.begin(115200);
  dht.begin();           // initialize temperature sensor

  //Initialize static IP for server purposes
  IPAddress ip(192, 168, 1, 153);
  IPAddress gateway(192, 168, 1, 1);
  IPAddress subnet(255, 255, 255, 0);
  WiFi.config(ip, gateway, subnet);
  Serial.print("Setting static IP to: ");
  Serial.println(ip);

  // Connect to WiFi network
  WiFi.begin(ssid, password);
  Serial.print("\n\r \n\rWorking to connect");

  // Wait for connection
  while (WiFi.status() != WL_CONNECTED) {
    digitalWrite(red, HIGH);
    delay(250);
    Serial.print(".");
    digitalWrite(red, LOW);
    delay(250);
  }
  Serial.println("");
  Serial.println("DHT Weather Reading Server");
  Serial.print("Connected to ");
  Serial.println(ssid);
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());
  WiFi.persistent(false); //workaround to buffer overflow of the wifi lib

  // Routers expressjs alike
  /*
    server.on("/", []() {
    server.send(200, "text/plain", "Welcome to the esp8266 website! Please request \"/get_weather\"");
    }); */
  server.on("/get_weather", handle_root);

  server.begin();
  Serial.println("HTTP server started");
}

void loop(void)
{
  server.handleClient();
}

void getWeather() {
  // Wait at least 2 seconds seconds between measurements.
  // if the difference between the current time and last time you read
  // the sensor is bigger than the interval you set, read the sensor
  // Works better than delay for things happening elsewhere also
  unsigned long currentMillis = millis();

  if (currentMillis - previousMillis >= interval) {
    // save the last time you read the sensor
    previousMillis = currentMillis;

    // Reading temperature for humidity takes about 250 milliseconds!
    // Sensor readings may also be up to 2 seconds 'old' (it's a very slow sensor)
    humidity = dht.readHumidity();          // Read humidity (percent)
    temp_f = dht.readTemperature(false);     // Read temperature as Fahrenheit
    // Check if any reads failed and exit early (to try again).
    if (isnan(humidity) || isnan(temp_f)) {
      Serial.println("Failed to read from DHT sensor!");
      return;
    }
  }
}
```

### Debug Messages

```

Exception (3):
epc1=0x4000bf64 epc2=0x00000000 epc3=0x00000000 excvaddr=0x40056fcb depc=0x00000000

ctx: cont 
sp: 3ffefee0 end: 3fff01e0 offset: 01a0

>>>stack>>>
3fff0080:  41c4cccd 41c4cccd 00000000 00000000  
3fff0090:  40056fcb 44000000 00000001 40203c80  
3fff00a0:  00000001 3ffee798 3ffef180 4020587d  
3fff00b0:  00000010 00000001 3fff153c 40202383  
3fff00c0:  00000010 3fff0120 3fff0120 40205aac  
3fff00d0:  3fff1d2c 0000000c 3fff0120 40205afb  
3fff00e0:  3ffef160 0000018a 0000018a 4010020c  
3fff00f0:  00000001 00000001 3fff153c 4020642a  
3fff0100:  00000000 3fffdad0 3fff153c 40203c76  
3fff0110:  3fff153c 3ffee7f0 3fff153c 40203cb2  
3fff0120:  00000000 00000000 00000000 40205c5c  
3fff0130:  3fff153c 3ffee7f0 3ffee7b0 40203d45  
3fff0140:  3fff1d2c 0000000f 0000000c 40202ccc  
3fff0150:  3ffee7f0 3ffee7b0 3ffef1c0 00000001  
3fff0160:  00000001 40203288 0000000f 3ffef1c0  
3fff0170:  00000000 00000000 3ffee7b0 3ffef1ac  
3fff0180:  00000001 3ffee7d4 3ffee7b0 40203f27  
3fff0190:  3ffe8550 00000000 000003e8 0101a8c0  
3fff01a0:  00000000 3fff19f4 feefeffe feefeffe  
3fff01b0:  3fffdad0 00000000 3ffef1a4 402022cc  
3fff01c0:  3fffdad0 00000000 3ffef1a4 40206258  
3fff01d0:  feefeffe feefeffe 3ffef1c0 40100718  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,6)

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset

```

I tried to make a simple webserver for displaying DHT22 data using the [Adafruit DHT library](https://learn.adafruit.com/esp8266-temperature-slash-humidity-webserver/code) but I'm having some strange behavior and crashes. After a certain amount of requests, the esp8266 board just hangs outputting that stack trace through the serial port. I searched the web for it and I found that time ago [someone opened an issue](https://github.com/esp8266/Arduino/issues/1997) similar to this one I have but the fix should have already been merged from the release 2.3.0-rc1.
I have even tried to put `WiFi.persistent(false);` in the code with no success.

slepax, 2016-10-28:
Hi

Just wondering if you found a solution to this problem? I have a similar issue.

Tried both using a cap on the rail, and WiFi.setPersistent(false).

Still fails.

RcrdBrt, 2016-10-28:
@slepax It's been 48 hours now with the esp8266 up and running without crashes. I added a [yield()](https://www.arduino.cc/en/Reference/SchedulerYield) call in tne webserver callback and another one after starting it in the loop {} (I don't know if the webserver is blocking though, I didn't test if the instructions after it in the loop get executed).

It helped a bit but in order to solve it further, I added 100nF ceramic capacitors ("104" label) in all the powerlines of the board. I read that esp8266 has serious power spikes due to the WiFi being used or not and capacitors (here used as smoothing capacitors) help a lot with that.
I added a 100K pull-up resistor in GPIO4 as in the pic because I read it somewhere that it might help for stability, too.

So my suggestion is: try adding small smoothing capacitors everywhere (small ones release the load faster afaik) + a GPIO4 pull-up resistor. If it doesn't help add some yield() calls in your code.

[image]

devyte, 2017-10-12:
@RcrdBrt that huge delay can cause a blow up in the wifi stack.
In any case, closin as resolved per last comment.

closed 2017-10-12 by devyte