esp8266/Arduino #2572: ADC values problem

youdemo, 2016-10-04:
Hello,
I only have an ESP12F which I am looking to use as a thermometer. The wiring is essentially
+3.3v ------ 10k resistor ------ (A)------ NTC 100k thermistor ------- GND
With point (A) connected to the ADC pin.

When  reading the ADC value using system_adc_read() I get values bellow:
ADC Values         Temperature(℃)
940                             30
910                             34

I want to get the temperature values from 10 to 150. How can I calc the relationship between the ADC values and temperature?
thanks
Lee

drmpf, 2016-10-04:
try this page http://www.petervis.com/electronics%20guides/calculators/thermistor/thermistor.html

youdemo, 2016-10-04:
TKS drmpf,
I calc the ADC value and the actual test output value is not the same.
Temp----calc ADC value---output value
29℃-------915.66-------------940
34℃-------883.76-------------912

How can I fix it?
thanks
Lee

duncan-a, 2016-10-04:
Probably down to the tolerance of your 10K fixed resistor - try finding a
high precision version - and the actual value of your 3.3V supply.

jourjine, 2016-10-04:
Max rawADc is 1024 
max voltage is 1.06

better way
+3.3v ------ NTC 100k thermistor  ------ (A)------ 326k resistor ------- GND
and use https://en.wikipedia.org/wiki/Steinhart%E2%80%93Hart_equation
if you dont have tech docs for NTC i may manual calibrate it

and the next better way use shunt resistor (linearize thermistors)
              |------ 25 k resistor ------|
+3.3v ------ NTC 100k thermistor  ------ (A)------ 600 resistor ------- GND

http://www.edn.com/design/sensors/4429105/Linearize-thermistors-with-new-formula

read about adc on esp
http://www.esp8266-projects.com/2015/03/internal-adc-esp8266.html
about noise
http://www.elcojacobs.com/eleminating-noise-from-sensor-readings-on-arduino-with-digital-filtering/

PS
i use esp and ntc100k on rapidoven

youdemo, 2016-10-05:
Tks All,
@jourjine   I calc the ADC value on your post article and make an Excel table list it.But I calc the ADC value and the actual test output value is not the same. Can you post your calc code now?

youdemo, 2016-10-05:
@jourjine 
ps: I only use ESP-12F board and ntc100k,not used Arduino.

jourjine, 2016-10-05:
its old code, may have little mistakes, because now i use Steinhart–Hart equation

urgent!!! -> rawADC value /= mV

1) take rawADC (code from esp8266-projects.com)
int readADC_dcm() { // increase resolution from 10 to 12 bits
```
        int ad2 = 0;
        int i=0;
        while (i < 16) {
              ad2 += analogRead(THERMISTORPIN);
              i++;
         }
         ad2 = ad2 >> 2;
         return ad2;
    }
```

2) tempTable (manual calibrated, code from reprap)
first version - mV, temp

```
const short temptable_11[][2] PROGMEM = {
{ 27, 125 },
{ 31, 120 },
{ 41, 110 },
{ 55, 100 },
{ 76, 90 },
{ 90, 85 },
{ 106, 80 },
{ 150, 70 },
{ 214, 60 },
{ 214, 60 },
{ 308, 50 },
{ 447, 40 },
{ 647, 30 },
{ 775, 25 },
{ 924, 20 },
{ 1285, 10 },
{ 2000, 0 },
};
```
in second version i use rawADC, temp and skip conversion rawADC to mVolts
but now i use Steihart.

3) convert rawADc to mVolts
```
float vOut = 1.06; // max input voltage (resistor devider must be correct values)
ADCresolution = 4096; // 1024 for 10 bit
rawADC = readADC_dcm();
mVolts = rawADC / ADCresolution * Vout; // convert raw to mv
```

4) table LookUpc(code from reprap)
```
float volts2temp(float raw) {
float celsius = 0;
uint8_t i;

for (i = 1; i < BEDTEMPTABLE_LEN; i++) {
if (PGM_RD_W(temptable_11[i][0]) > raw)
{
celsius = PGM_RD_W(temptable_11[i - 1][1]) + (raw - PGM_RD_W(temptable_11[i - 1][0])) * (float)(PGM_RD_W(temptable_11[i][1]) - PGM_RD_W(temptable_11[i - 1][1])) / (float)(PGM_RD_W(temptable_11[i][0]) - PGM_RD_W(temptable_11[i - 1][0]));
break;
}
}
// Overflow: Set to last value in the table
if (i == BEDTEMPTABLE_LEN) celsius = PGM_RD_W(temptable_11[i - 1][1]);

return celsius;
}
```

and current little code:

```
average is rawADC value
void doControl() {
          mVolts = average / 1024.0 * Vout;  // convert rawADC to mVolts
          float NTCres = log((Vin * ResistorR2 / mVolts) - ResistorR2); // convert mVolts to resistance
          Input = (1.0 / (shC1 + (shC2 * NTCres) + (shC3 * pow(NTCres, 3)))) - 273.15;  // compute Temp
          if (pidEnable) {
            myPID.Compute();
           }
         }

```

jourjine, 2016-10-05:
void loop() {

  currentTimeTemp = millis();
    if (currentTimeTemp >= (loopTimeTemp + 500)) {               
      readTemperature();
    loopTimeTemp = currentTimeTemp;
    }

  currentTime = millis(); 
    if (currentTime >= (loopTime + onesecond)) {            
      mainCookLoop();
    loopTime = currentTime;                          
    }

  Blynk.run();

  ArduinoOTA.handle();

  if (pidEnable) {
  currentTimePID = millis();                           
  if (currentTimePID >= (loopTimePID + 125)) {     
          if (cookMode == STOPMODE || cookMode == STANDBYMODE) digitalWrite(RELAY, LOW); 
              else DriveOutput();
      loopTimePID = currentTimePID;                    
      } // end timer run
  } // end pid enable
} // end loop

youdemo, 2016-10-05:
@jourjine undefined reference to `analogRead' in 1)

yagizhanugur, 2016-10-13:
http://esp8266tutorials.blogspot.com.tr/2016/09/esp8266-ntc-temperature-thermistor.html

Try that page, it will helps.

devyte, 2017-10-12:
Off-topic, closing per #3655

closed 2017-10-12 by devyte