esp8266/Arduino #2574: Error in compiling

martignonia, 2016-10-04:
Hello everyone,

I'm trying using a brand new ESPDUINO.

When I compile Arduino Uno code that is normally work with Arduino I have a lot of compiling errors:

```
Arduino: 1.6.12 (Mac OS X), Board: "ESPDuino (ESP-13 Module), 80 MHz, 115200, 4M (3M SPIFFS)"

/Users/i054435/Documents/Arduino/libraries/VirtualWire/VirtualWire.cpp: In function 'void vw_setup(uint16_t)':
/Users/i054435/Documents/Arduino/libraries/VirtualWire/VirtualWire.cpp:390:5: error: 'TCCR1A' was not declared in this scope
     TCCR1A = 0; // Output Compare pins disconnected
     ^
/Users/i054435/Documents/Arduino/libraries/VirtualWire/VirtualWire.cpp:391:5: error: 'TCCR1B' was not declared in this scope
     TCCR1B = _BV(WGM12); // Turn on CTC mode
     ^
In file included from /Users/i054435/Documents/Arduino/libraries/VirtualWire/VirtualWire.h:146:0,
                 from /Users/i054435/Documents/Arduino/libraries/VirtualWire/VirtualWire.cpp:31:
/Users/i054435/Documents/Arduino/libraries/VirtualWire/VirtualWire.cpp:391:18: error: 'WGM12' was not declared in this scope
     TCCR1B = _BV(WGM12); // Turn on CTC mode
                  ^
/Users/i054435/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/Arduino.h:184:25: note: in definition of macro '_BV'
 #define _BV(b) (1UL << (b))
                         ^
/Users/i054435/Documents/Arduino/libraries/VirtualWire/VirtualWire.cpp:398:5: error: 'OCR1A' was not declared in this scope
     OCR1A = nticks;
     ^
/Users/i054435/Documents/Arduino/libraries/VirtualWire/VirtualWire.cpp:405:5: error: 'TIMSK' was not declared in this scope
     TIMSK |= _BV(OCIE1A);
     ^
In file included from /Users/i054435/Documents/Arduino/libraries/VirtualWire/VirtualWire.h:146:0,
                 from /Users/i054435/Documents/Arduino/libraries/VirtualWire/VirtualWire.cpp:31:
/Users/i054435/Documents/Arduino/libraries/VirtualWire/VirtualWire.cpp:405:18: error: 'OCIE1A' was not declared in this scope
     TIMSK |= _BV(OCIE1A);
                  ^
/Users/i054435/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/Arduino.h:184:25: note: in definition of macro '_BV'
 #define _BV(b) (1UL << (b))
                         ^
/Users/i054435/Documents/Arduino/libraries/VirtualWire/VirtualWire.cpp: At global scope:
/Users/i054435/Documents/Arduino/libraries/VirtualWire/VirtualWire.cpp:588:7: error: expected constructor, destructor, or type conversion before '(' token
 SIGNAL(TIMER1_COMPA_vect)
       ^
/Users/i054435/Documents/Arduino/libraries/VirtualWire/VirtualWire.cpp:668:1: error: expected '}' at end of input
 }
 ^
exit status 1
Error compiling for board ESPDuino (ESP-13 Module).

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
```

Is there something that I'm doing wrong?

thank you
Andrea

krzychb, 2016-10-04:
Hi @martignonia,

I believe the VirtualWire library you use is not compatible with ESP8266.
TCCR1A and TCCR1B are AVR Timer/Counter Control Registers 
ESP8266 is not an AVR / has different architecture.

You need to look for another library that is compatible with ESP8266.

martignonia, 2016-10-04:
Hi @krzychb also if I'm using other library I have same problem. Seems that any external library is not compatible with this board.

krzychb, 2016-10-04:
> Seems that any external library is not compatible with this board.

Regarding library compatibility please check:
* https://github.com/esp8266/Arduino/blob/master/doc/faq/a03-library-does-not-work.md
* https://github.com/esp8266/Arduino/blob/master/doc/libraries.md#other-libraries-not-included-with-the-ide

closed 2017-10-12 by devyte