esp8266/Arduino #2578: softAP Subnet Mask unable to be set for DHCP

mtongnz, 2016-10-05:
When running in softAP mode, there is no way to set what subnet mask is sent to DHCP clients.  WiFi.softAPConfig allows you to set the subnet for the ESP8266 but this isn't passed to the DHCP server - it always assigns 255.255.255.0

It appears to me that this is set in [Arduino/tools/sdk/lwip/src/app/dhcpserver.c on lines 135-150](https://github.com/esp8266/Arduino/blob/18f66e99693020a758b8f5ac92e042cbdecb7760/tools/sdk/lwip/src/app/dhcpserver.c#L135).  The subnet mask is hard coded in.

Is it possible to link [WiFi.softAPConfig](https://github.com/esp8266/Arduino/commit/4cc06c2100b1cab366dbe8f8aec31c1de37d0903) to set these values or provide another function to achieve this?

There seem to be [a few other issues logged](https://github.com/esp8266/Arduino/issues/1756) which incorrectly thing the subnet isn't set at all.  The subnet appears to be set correctly on the ESP8266, just not given out by the DHCP server.

Thanks

devyte, 2017-10-11:
@mtongnz how does PR #3637 work for you?

mtongnz, 2017-10-12:
This issue was resolved with #2648.  Sorry I forgot to close this :)

closed 2017-10-12 by mtongnz