esp8266/Arduino #2579: i2c mpu6050 + hmc5883l

andreyos, 2016-10-05:
I have connected mpu6050 + hmc5883l to I2c (pin D1 -> 4GPIO, D2 ->5GPIO) as default) and 
sometimes ESP can found the sensors and sometimes cannot found their on start. Other behavior ESP lose sensors in process.. Usually, any reset (button or power off) does not help.

I tried to use:
- pull-up resistors for SDA SCL 4.7k, 10k, without - no result.
- phy. wires 50cm, 5 cm - no result.
- new hardware esp (nodemcu), new mpu6050 and new hmc5883l - no result.
- pin 0, pin 2 - no result.
When some one sensor(mpu or hmc) is disconnected then second sensor is found by ESP board (100%). If connect back first sensor then some time (1min - some hours) the sensors can be visible for board but ESP losing the sensors again.

Example today, I used phy. wires 50cm (esp did not find the sensors), I cut the wires to 5cm - ESP found the sensors and worked correctly 1-2 hours but after that lost sensors and did not find them. Any reset did not help.

it was reproduced on Core Version 2.2.0 and 2.3.0

https://goo.gl/photos/TXRHtZroF83YKc1m9

**MPU-6050**
vcc - 3.3v from board
gnd - gnd from board
scl - D1 from board
sda - D2 from board
ad0 - gnd (address 0x68) from board

**HMC5883l**
vcc - 3.3v from board
gnd - gnd from board
scl - D1 from board
sda - D2 from board

Hardware:			lolin nodemcu v3 esp8266mod
Core Version:      	2.3.0

Module:  			NodeMcu 1.0 ESP12E
Flash Size: 		4MB
CPU Frequency:		80Mhz
Upload Using:		usb
Reset Method:		nodemcu

#include <Arduino.h>
#include <Wire.h>
void setup()
{
	Serial.begin(115200);
	Wire.begin();
        delay(1000);

		//------------------------------------------------
		byte start_address = 0;
		byte end_address = 127;
		byte rc;
		Serial.println("\nI2C Scanner");

		Serial.print("Scanning I2C bus from ");
		Serial.print(start_address, DEC);  Serial.print(" to ");  Serial.print(end_address, DEC);
		Serial.println("...");
		for (byte addr = start_address; addr <= end_address; addr++) {
                        delay(100);
			Wire.beginTransmission(addr);
			rc = Wire.endTransmission();
			if (addr<16) Serial.print("0");
			Serial.print(addr, HEX);
			if (rc == 0) {
				Serial.print(" found!");
			}
			else {
				Serial.print(" "); Serial.print(rc); Serial.print("     ");
			}
			Serial.print((addr % 8) == 7 ? "\n" : " ");
		}
		Serial.println("\n-------------------------------\nPossible devices:");

### Debug Messages
I2C Scanner
Scanning I2C bus from 0 to 127...
00 2      01 2      02 2      03 2      04 2      05 2      06 2      07 2     
08 2      09 2      0A 2      0B 2      0C 2      0D 2      0E 2      0F 2     
10 2      11 2      12 2      13 2      14 2      15 2      16 2      17 2     
18 2      19 2      1A 2      1B 2      1C 2      1D 2      **1E 2**      1F 2     
20 2      21 2      22 2      23 2      24 2      25 2      26 2      27 2     
28 2      29 2      2A 2      2B 2      2C 2      2D 2      2E 2      2F 2     
30 2      31 2      32 2      33 2      34 2      35 2      36 2      37 2     
38 2      39 2      3A 2      3B 2      3C 2      3D 2      3E 2      3F 2     
40 2      41 2      42 2      43 2      44 2      45 2      46 2      47 2     
48 2      49 2      4A 2      4B 2      4C 2      4D 2      4E 2      4F 2     
50 2      51 2      52 2      53 2      54 2      55 2      56 2      57 2     
58 2      59 2      5A 2      5B 2      5C 2      5D 2      5E 2      5F 2     
60 2      61 2      62 2      63 2      64 2      65 2      66 2      67 2     
**68 2**      69 2      6A 2      6B 2      6C 2      6D 2      6E 2      6F 2     
70 2      71 2      72 2      73 2      74 2      75 2      76 2      77 2     
78 2      79 2      7A 2      7B 2      7C 2      7D 2      7E 2      7F 2 
```
xx - address, y - result
all address have error 2, but **mpu6050** and **hmc5883l** are CONNECTED.
```

do you have ideas what is it?

WereCatf, 2016-10-06:
It looks like both of those modules may already have pull-ups on them, have you checked that and what value they are using?

andreyos, 2016-10-06:
I did not find document where is specified the resistors. I did manual measurement:
 |- -  - - -  - - - MPU-6050 - - - HMC5883L

VCC - SDA:    - -     **660 kOm**   - -   **4.7 kOm**
VCC - SCL	:     - -     **660 kOm**   - -   **4.7 kOm**
GND - SDA:   - -      **90 kOm**    - -  **>2000 kOm**
GNG - SCL:    - -      **90 kOm**    - -  **>2000 kOm**
VCC - GND:   - -      **782 kOm**  - -  **>2000 kOm**
SDA - SCL:     - -      **4.3kOm**    - -  **9.4 kOm**

* MPU-6050:  AD0 is not connected. The board contains LED- maybe it can explain VCC - GND: 782 kOm

Also, the sensors connected to one wires of I2C and some values of the resistors can/must be affected.

andreyos, 2016-10-06:
I have suspicion that I2C driver of ESP is poor  hard implementation. It can explain unstable work.

andreyos, 2016-10-06:
As I see, MPU-6050 has pull-down and HMC5883L has pull-ups by default ???

|- - - - - - - - MPU-6050 - - HMC5883L
VCC - SDA: - - 660 kOm - - 4.7 kOm
VCC - SCL : - - 660 kOm - - 4.7 kOm
GND - SDA: - - 90 kOm - - >2000 kOm
GNG - SCL: - - 90 kOm - - >2000 kOm

WereCatf, 2016-10-06:
Are you absolutely sure you're not screwing up your measurements? Because 660k pull-ups on I2C-lines would be *way* too weak and out of spec.

andreyos, 2016-10-06:
double checked - 660k.When mpu 6050 is alone on i2c - works fine (and not matter is or not pull up resistors SDA SCL). HMC5883L works fine too when the sensor is alone on i2c. BUT when both sensors are connected - they are NOT WORK (with 2.2k, 4.7k, 20k and without pull up resist. SDA SCL)!!!   WHY???

pasko-zh, 2016-10-07:
According to the photo you are using two breakout boards, can you tell which one? Your MPU-6050 looks similar to the [sparkfun version](https://www.sparkfun.com/products/11028), which has built 10K pull ups, see [schematics](https://cdn.sparkfun.com/datasheets/Sensors/IMU/Triple_Axis_Accelerometer-Gyro_Breakout_-_MPU-6050_v12.pdf) of the breakout. 
i.e., first you should make sure whether the breakout boards contain pull ups or not. I guess both do -- because this is the "normal" case.

Concerning, "sometimes working, sometimes not": There was a very similar case with a HTU21D breakout board, see this [issue ](https://github.com/pasko-zh/brzo_i2c/issues/5) for brzo_i2c library. When I did the analysis, I found out that breakout board has some issues due to the level shifting, see [here](https://github.com/pasko-zh/brzo_i2c/tree/master/examples/HTU21D).

WereCatf, 2016-10-08:
@pasko-zh Stronger pull-ups should fix the horribly slow rise-time. Andreyos does mention he tried with 4.7k resistors, but maybe he should try with 1.5k resistors (don't go lower than that, tho) just to see if the problem stems from that. If stronger pull-ups don't fix it then the issue lies elsewhere.

pasko-zh, 2016-10-08:
@WereCatf I know :smile: I do always calculate the pull up value, especially for [fast mode plus SCL speeds](https://github.com/pasko-zh/brzo_i2c/wiki#scl-speed-in-fast-mode-plus). 

However, I **explicitly** followed the "how to wire the breakout board", i.e. without ("additional") pullups, because this is what many people just do. As those people do not calculate anything before wiring/assembly in hardware, instead they just follow what is written online—which is fine, I am not saying this is bad, don't get me wrong here.

Concerning the HTU21D, of course raise times are better with low values (strong pull ups). However there is still the same behaviour, i.e. on startup and after every other temp reading, the sensor doesn't work. That's why noted this in the [readme](https://github.com/pasko-zh/brzo_i2c/tree/master/examples/HTU21D). And this behaviour is very similar to what he describes, even if compeletey different sensor chips are used, however the breakout boards most probably do have a similar circuit...

andreyos, 2016-10-17:
Fixed. Added RC filter between nodemcu and sensors: Electrolytic capacitor 2000mkF 10v + resistor 10kOm.

reason: nodemcu and any other usb power have step-down converter (Switched-mode power supply) - it creates noise in i2c bus. The noise can be detected by oscilloscope.  

Thanks.

Juppit, 2016-10-18:
Nodemcu V0.9 has this SPX3819. So a small capacitor of 10nf on the bypass pin should do it,
see the documentation on https://github.com/nodemcu/nodemcu-devkit/blob/master/Documents/SPX3819M5-L-3-3.pdf page 7.

devyte, 2017-10-12:
Closing as resolved.

closed 2017-10-12 by devyte