esp8266/Arduino #2581: WEMOS D1 mini Pro - 16M supporting

4refr0nt, 2016-10-06:
#### Hardware
Hardware:			WEMOS D1 mini Pro - 16M

#### What's new with D1 mini Pro

- 16M bytes (128M bit) Flash
- External antenna connector
- Built-in ceramic antenna
- New CP2104 USB-TO-UART IC
- Same size as D1 mini, but more light

[View D1 mini Pro on WeMos Aliexpress store](http://ali.ski/gsnRGA)

WereCatf, 2016-10-06:
This is not an issue and this is not the place to advertise your affiliate-links.

TotallyInformation, 2016-10-29:
Hi, I've got 2 of these now and would like to understand how to update the boards.txt to make use of the 16M memory?

Not as simple as I'd hoped. I'm guessing that these are a copy of the D1 Mini but with the larger FLASH chips attached.

WereCatf, 2016-10-29:
@TotallyInformation It'd be best to make a proper variant out of it. I can take a stab at it in a few hours.

TotallyInformation, 2016-10-29:
Fab, thanks WereCatf.

WereCatf, 2016-10-29:
@TotallyInformation Okay, copy https://github.com/WereCatf/Arduino/blob/WemosD1MiniPro/boards.txt over your existing one and that should be it. I do not have one of these boards, so I can't actually verify that it works, so it would be nice if you'd chime back here if it works as it should.

pieman64, 2016-10-30:
@WereCatf I have some of the Pro's and did some manual adjustments to boards.txt. Am I right in thinking we still can't access the memory from 4M to 16M?

WereCatf, 2016-10-30:
@pieman64 I don't know, I ain't got any 16MB-boards. If you've got the pros then you could just try.

wfraga18, 2016-10-30:
why WEMOS never answer the forum questions ?? launch a product, but don't have support !

HOW TO USE 16MB ???

WereCatf, 2016-10-30:
@wfraga18 Ask them, not us. We don't work for Wemos. As for using it: how about you actually read the replies here, instead of raging? I provided a new boards.txt above you can try.

TotallyInformation, 2016-10-30:
Thanks @WereCatf, I'm at a conference for a few days, I'll try after that. Good work.

4refr0nt, 2016-10-30:
@WereCatf, I'm try flash one of examples to `Wemos D1 mini Pro` with new [`boards.txt`](https://github.com/WereCatf/Arduino/blob/WemosD1MiniPro/boards.txt) without any issues (spiffs not tested)

martinayotte, 2016-11-09:
I've received my Mini Pro today. The sketch is running fine with 16MB settings, but SPIFFS and OTA becomes broken. It is not broken if I switch back to 4MB settings.
So, for now, I think the additional free Flash is only usable using direct calls to Esp.flashWrite() and Esp.flashRead(), although I didn't try it yet ...

martinayotte, 2016-11-11:
Using the hack details here http://www.packom.org/esp8266/16mb/flash/eeprom/2016/10/14/esp8266-16mbyte-flash_handling.html, I was able to get SPIFFS working on 16MB, but there still issues : SDK still try to access WiFi configs like 4MB, so I had to move SPIFFS_START after the first 4MB, with 12MB remaining. Also, I still face issue with OTA, when e-boot is call, I get "need boot 1.4+", and hangs forever until doing a erase_flash with esptool.py.

pbecchi, 2017-04-11:
Any update on over 4 mB support?
I read that should be included in next Arduino Code release 2.4!
It is already supported by the GitHub version?

martinayotte, 2017-04-11:
Take a look at https://github.com/esp8266/Arduino/pull/2351#issuecomment-290830716
Of course, you need to tweak Arduino code to get those hacks merged manually.

pbecchi, 2017-04-12:
Thanks!
I understand that your mod and packing.org one are not merged yet.
Since I have some Esp8266 100 with 8Mb loosing 3 out of 8 isn't the best solution.
I was thinking to add a second Spiffs with SpiffsStart after 4Mb.Do you think it is feasible?

martinayotte, 2017-04-12:
Their a lot of work to be done to have 2 SPIFFS at the same time ...
But you can still use the unused part of the Flash as a raw flash storage, or you can reduce the 4MB part to 2MB and have a 6MB SPIFFS.

pbecchi, 2017-04-13:
This may be the best solution ....Some other value set to reduce 4Mb to 2?

martinayotte, 2017-04-13:
Yes, the SPIFFS location is defined in LD scripts, and you choose which one from Arduino Menu.

capedra, 2017-04-23:
@martinayotte I can't make SPIFFS from FS.h to work with the 12Mb of flash.

martinayotte, 2017-04-23:
In my setup, I moved the start above the 4M boundary, because ROM still access WiFi configs at the end of this first 4M.
`PROVIDE ( _SPIFFS_start = 0x40600000 );`
@capedra , did you modified the EspClass in Esp.cpp with the hack mentioned in https://github.com/platformio/platform-espressif8266/issues/8#issuecomment-293421839 ?

capedra, 2017-04-24:
@martinayotte Yes, but these changes do not work with the functions from "FS.h" file.

jonhp, 2017-05-05:
I have read all of the posts on this issue. I think most of what is here is beyond the "usual" Arduino user. So, at the risk of collecting some developer ire, I would gently ask if added D1 mini Pro support is a goal and on what time-frame that might happen? 
BTW I replaced the boards.txt with the one that has D1 mini Pro definitions. I still get Arduino IDE messages about the maximum size is 1MB.

devyte, 2017-05-05:
Not sure if relevant, but SDK 2.1 is out, changelog says it now supports up
to 16M.

igrr, 2017-05-08:
SDK update to 2.1 will be tracked here: https://github.com/esp8266/Arduino/issues/3215

igrr, 2017-05-23:
#3278 has preliminary support for 8MB and 16MB flash size

capedra, 2017-06-29:
@igrr any progress?

igrr, 2017-06-29:
You can try the branch in #3278. As far as having SPIFFS on a 16 MB flash goes, it works.

pieman64, 2017-06-29:
@igrr so 16MB SPIFFS support is still not available in the master branch, right?
Any idea when it might be?

capedra, 2017-07-04:
@pieman64 
@igrr 
@projectgus 
@Spritetm 

Ivan, I've been using SDK 2.4.0-RC1 because you said that the new fixes were more important than this 16 megabytes upgrade. I don't even know how to install the branch #3278 on Windows. Also, does this branch include the new changes of 2.4.0-RC1 ?

When will all the checkpoints become checked there on that branch? Are you going to finish that or is your main focus on ESP-32 from now on?

Maybe SDK 2.4.1 would sound better than 2.4.0-RC2?

Best Regards.

igrr, 2017-07-04:
You can follow instructions for "Using git version" given in project's README.md to install git version of ESP8266 Arduino core. Then do `git checkout pull/3278/head` to switch to the PR branch. Make sure you uninstall any versions of the core installed using boards manager, before you attempt installing git version.

Regarding next steps, the plan is to fix pending issues with 2.4.0 release first. I can work on an update to the new non-OS SDK after that. Out of the checklist, all points except for "esptool-ck: support uploading to 8/16 MB flash chips" are fairly straightforward. I can't give any ETA unfortunately, as i have very little spare time right now.

capedra, 2017-07-08:
@igrr 
Hello, I can't mount the filesystem using the function SPIFFS.begin() from FS.h.

igrr, 2017-07-08:
What debug output do you get if you set "Debug level: Core" and add `Serial.setDebugOutput(true);` after `Serial.begin`?

capedra, 2017-07-08:
```
SPIFFSImpl: allocating 512+180+1400=2092 bytes
SPIFFSImpl: mounting fs @100000, size=efb000, block=2000, page=100
SPIFFSImpl: mount rc=-1
```

capedra, 2017-07-10:
@igrr Ivan, what could possibly be wrong?

igrr, 2017-07-10:
Looks like flash size is still set to 4 MB in the binary image header.
Can you try replacing your esptool-ck with a version from this branch?
https://github.com/igrr/esptool-ck/tree/feature/8m_16m

capedra, 2017-07-10:
@igrr Hello again, Ivan. Thank you a lot for helping me solve that. How can I generate the esptool.exe from this branch? I tried importing the files to a new project on Visual Studio but it didn't work at all. Please help newbies like me. :)

Best Regards!

igrr, 2017-07-10:
Normally it is build in MinGW, you can find the steps in appveyor.yml file in esptool-ck repository root.

There is also an automatically built version available here: https://ci.appveyor.com/project/igrr/esptool-ck/build/0.0.113/artifacts

capedra, 2017-07-10:
@igrr All I get now is the following output from the SPIFFS sketch:
```
SPIFFSImpl: allocating 512+180+1400=2092 bytes
SPIFFSImpl: mounting fs @100000, size=efb000, block=2000, page=100
SPIFFSImpl: mount rc=-1
```
I guess "SPIFFSImpl: mount rc" should return 0 (SPIFFS_OK) instead of -1?

pieman64, 2017-07-10:
@capedra so did you just replace the esptool.exe with the one provided by @igrr and then run your "12 minute" sketch to get 16MB SPIFFS (or more precisely 1MB + 15MB SPIFFS)?
Are you using the master branch, 2.4.0-RC1 or the old 2.3.0?

capedra, 2017-07-10:
@pieman64
Exactly, I replaced the esptool.exe that the Arduino IDE was using with that one included on the ZIP file of Ivan.

It's important to note that it took 12 minutes because it was my first run after moving from 4 MB. 

So, it seems to me that the mount part became also the format part after the -1 response.

About the version that I'm using, it's still 2.4.0-RC1. I did want to keep that version because of the corrections but I had to manually implement the changes of the branch 3278 on RC1.

Good luck for everyone. I know it may sound off topic but now I can't understand why ESP-32 has only 4 MB of flash, @igrr. I heard some rumors that it will support 64 MB of flash soon? Great job!

igrr, 2017-07-10:
>  know it may sound off topic but now I can't understand why ESP-32 has only 4 MB of flash

Just like the ESP8266, ESP32 does not *have* flash by itself. Board/module designers can use any flash size between 1MB and 64MB. Up to 16MB of flash can be memory-mapped. ESP-WROOM-32 modules can be ordered with flash sizes other than 4MB, subject to MOQ.

capedra, 2017-07-10:
@igrr
I look for the day that ESP will run Linux and has enough storage and RAM memory to smoothly work with LAMP.

TotallyInformation, 2017-07-10:
Off topic now by a long way but if you want an ESP with enough grunt to run Linux, you really should be looking at a Raspberry Pi Zero or one of the similar devices. The ESP devices are microprocessors not general purpose computers.

pieman64, 2017-07-10:
@capedra thanks for the details. I'm using the master branch as I wanted all the latest fixes. Could you please provide details of how you "patched" your 2.4.0-RC1 with the 3278 branch on your Windows system?

@igrr you wrote _"Then do git checkout pull/3278/head to switch to the PR branch. Make sure you uninstall any versions of the core installed using boards manager, before you attempt installing git version."_ Could you please expand on the uninstall process?

capedra, 2017-07-10:
@TotallyInformation
Besides the fact that microprocessors don't work well with sensors, there's also the problem that customers can buy only one Raspberry Pi Zero from each store, so, if I need ten Raspberry Pi Zero units, I will have to pay about USD 25-35 per unit on China (not UK because of the problem that I mentioned up above). So, I think that we still need a board that we can call "the killer" in terms of dimensions and also logistics.

@pieman64
Assuming that you or someone else already installed RC1, you just need to implement Ivan's 43 latest files' changes of the branch 3278 directly on the RC1 files. If you're using Windows, I suggest you to use a software called Everything, from Voidtools. Besides the fact that it's an adventure to change each file hand by hand, it's also an interesting way to learn how his outstanding LOC (Lines of code) work very nicely.

About the uninstall that he was talking about, I uninstalled 2.3.0 using the Arduino IDE board manager. It helps you to uninstall the SDK version that you're currently using and then you can choose another version on that same combobox.

duncan-a, 2017-07-17:
> There are other similar units that are available now & the limitation on
> PiZero numbers should hopefully be temporary.
>
> The point I was making is that there is a difference between an SBC and a
> microprocessor. As you say they are good for different things. Linux is an
> OS for computers not for microprocessors.
>
> The ESP units are microprocessors. Microprocessors are good at realtime
> processing and need an OS that supports that, Linux is not designed for
> realtime systems.

Pi Zero in quantities greater than one has been promised ever since it
first appeared on the cover of MagPi magazine - it seems highly unlikely
that it will ever change as there is little profit margin on the boards
themselves but quite a bit on the shipping.

Back to the original topic...

capedra, 2017-07-19:
@igrr 
Ivan, does the fact that I changed Flash speed to 80 MHz instead of 40 MHz and also Flash mode to QIO cause flash data corruption? Because I was overwriting two files at the flash these days when it suddenly stopped rewriting anymore. Also, whenever I deleted the files, nothing happened. The function "SPIFFS.remove" had no effect toward the files. According to WinSCP, it couldn't create/open file to rewrite to disk.

Unfortunately, I mistakenly formatted the SPIFFS and everything got back to normal. Sorry for that.
Right now, should I test it with Flash speed of 40 MHz and/or DIO Flash mode? I think I'm going to set it to DIO and 80 MHz of Flash speed. What do you think?

Best Regards!

capedra, 2017-07-20:
@igrr Could you please help?

igrr, 2017-07-20:
Sorry, i don't have any boards which would have >4MB of QIO flash to test this. If you are seeing data corruption at 80MHz, try dropping the frequency to 40MHz and see if that resolves the problem.

capedra, 2017-07-28:
@igrr Ivan, this is the third time that I face problems with the flash of Wemos D1 mini Pro during my durability tests.

capedra, 2017-07-28:
I will post a spreadsheet on here soon with the results of my tests.

capedra, 2017-07-29:
@igrr I'm starting to think that these flash memories don't pass on durability tests.

capedra, 2017-07-29:
I'm currently testing.

capedra, 2017-07-30:
@igrr 

I'm sorry but 16 MB (actually 14900866 bytes, according to FSInfo) hasn't been working well. I've been getting corrupted files on flash, the fragmentation is increasing all the time ( Richard also proved that on http://www.esp8266.com/viewtopic.php?f=32&t=12052 ). At the beginning I was thinking that it was the library that I was using (https://github.com/nailbuster/esp8266FTPServer), but now, I'm sure that the problem is really on these flash modules. Reading time decreases a lot after uploading new files, Writing time also decreases a lot after writing new files, as you can see at the spreadsheet on the end of this post.

This is the final test that I did in case anyone wants to know:
I formatted SPIFFS after not being able to delete/write new files on it;
after that, I uploaded a 10 MB MP3 file to the flash module. Works very well when I listened to it on the Web Server that I put on Wemos. After that, I uploaded a new song, and the system considered the transfer successful. So, when I listened to it on the Web Server, the music was corrupted, because there were so many glitches on it that the ESP even stopped sending the file. Finally, to prove that it's really a flash memory corruption on Wemos D1 Mini Pro, I downloaded this same file and proved that I can't trust SPI NOR Flash memories anymore because the same problem happened.

This way, I can't commercialize my products with these flash modules because my clients aren't going to complain that their devices stopped upgrading someday! These flash modules aren't reliable at all. Also, I can't add an SD card to my project because I don't have 4 digital pins available and I won't change my one year and a half project.

Oh, and FSInfo details below just in case anyone get interested:
```
Total Bytes:14900866
Used Bytes:14174974
Block Size:8192
Page Size:256
Max Open Files:5
Max Path Length:32
```

I have been testing night and day (even on weekends) the SPIFFS. I'm sorry but it's the truth about it. I'm going to test on 4MB flash modules now but it seems to me that Richard already tested that. I even made a spreadsheet detailing everything that happens on Wemos. You can take a look below:

https://docs.google.com/spreadsheets/d/1PzlkPWUjimDEFwIp7aisl51l3x82thJiUp7Gj_lKtLo/edit?usp=sharing

capedra, 2017-07-30:
Apparently, the same thing has been going on with 4 MB modules:
https://docs.google.com/spreadsheets/d/1EZMl0QQpJEAXiOtZD6HPva6eVbaHfxS47wsQSCsu3EU/edit?usp=sharing

jonhp, 2017-07-30:
Can you load a flash up to 14.1/14.9 and have reliable results?

https://docs.google.com/spreadsheets/d/1PzlkPWUjimDEFwIp7aisl51l3x82thJiUp7Gj_lKtLo/edit?usp=sharing

martinayotte, 2017-07-30:
Then, it is probably a SPIFFS issue ...
Did you try to write your music files as plain Raw data and keep track of the begin sectors and lengths where files are located ?
If no corruption there, it will prove that it is a software issue.

devyte, 2017-07-30:
Spiffs is not without bugs. I believe the max theoretical file size is
16MB, but I haven't seen anyone even approach such a large file.
I think the suggestion of trying the raw flash api as a test has merit.
However, don't consider it as a final solution, because you wouldn't be
doing wear leveling.
If Spiffs does have an issue, there are debug messages that can be turned
on, and an issue can be opened on pellepl's repo.

capedra, 2017-07-31:
@jonhp Yes, I guess so. The 10 MB worked fine just after I formatted the SPIFFS.

@martinayotte I didn't try to do that but I really think that it's a software issue because I tested it on a new Wemos D1 Mini Pro from their official store. I don't think that I already achieved the well-known 10000 limit of flash. Maybe you could try to do that because SPIFFS isn't reliable anyway. 

@devyte I've seen lots of bugs on SPIFFS lately but then again, none is trying to solve that, so, I'm moving on and I'll be using SD cards instead of SPIFFS from now on.

Thank you everyone for your support.

[60 more replies]

closed 2019-08-29 by d-a-v