esp8266/Arduino #2582: WiFi.mode(WIFI_AP_STA)

castro2487, 2016-10-06:
Hi to all.

    i'm doing a little project and i have a doubt if anyone can help: ¿it's possible to scan and connect Wifi network without disconnect a network that i create?

    because i have some development to check current in my house and i want to make it portable, the first time that i start the board the ESP8266 12 create and Access Points, ask me some that and the Wifi network of my house because i will create a server and access via internet.

battuashwik, 2016-10-06:
Yep. This is possible. You can also host a access point and connect to another at the same time. (implement a wifi mesh). The above mode is for that purpose.

devyte, 2017-10-03:
Closing as off-topic, see #3655 .

closed 2017-10-03 by devyte