esp8266/Arduino #2589: Can't detect client disconnection

NdK73, 2016-10-08:
### Basic Infos
Once I accept a client connection via WiFiServer.available() I'm unable to detect when the client disconnects uncleanly (for example because of a reset).

#### Hardware
Hardware:			ESP-01
Core Version:      	2.3.0-rc2

### Description

See "basic infos".
WiFiClient.status seems to keep returning ESTABLISHED (I tried waiting for about 2h, so it's not a timeout issue) even if the connection is dead. Not even sending something (that obviously won't get ACKed) seems to change the client status.

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		4MB/1MB
CPU Frequency:		80Mhz
Flash Mode:			DIO
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		ck

MoegerleStephan, 2017-01-10:
I've the same issue here.
In my case there're several ESP-12 boards that should establishing a connection with the master board to exchange data. After a short exchanging time they should disconnect from the master that some other boards get the change for exchanging data with the master.

The problem now is that the master AP didn't recognize when ESP clients are leaving the AP network. Therefore after some established connections with the client boards, the master stops establishing new connections with other ESP boards altought they weren't connected anymore.
I'm wondering that nobody else had an issue like that.

Where can I find a method for kicking out wifi clients in general or where can I check for connections to death clients?
Isn't there are polling method, that checks connection to each client anywhere?

Thanks in advance and best regards,
Stephan

NdK73, 2017-01-10:
Are you calling WiFi.disconnect() on the client? This way the AP should notice. Else I fear there's no way to differentiate between an idle client and a dead one unless generating some traffic (even a "ping" could work).

hieuvt6-fti, 2017-06-02:
I also have this issue.

I use a simple Android app that could connect or disconnect to my WiFiServer on ESP8266. Sometimes the ESP8266 failed to recognize the disconnection, and passed both of my connection check.

    WiFiClient client;
    ...
    if (client) {
      if (client.connected()) {
        ...
      }
    }

devyte, 2017-10-11:
@NdK73 please try PR #3362 (lwip2), and report back here.

devyte, 2018-07-13:
Closing due to lack of feedback in several months.

closed 2018-07-13 by devyte

mugi-luffy, 2018-12-07:
I'am having the exact same issue using the latest stable core version.
connected() always returns true.
I have two ESP8266: one running as server (WiFiServer) and the other one as client. After unplugging the client, the server still returning true for connected(), even after generating some traffic on the server with "client.println("message")".
Could this Issue be reopened or should I just make a new one?

d-a-v, 2018-12-07:
SDK has changed since latest 2.4.2 stable version.
Would you be able to retry with current git version (or wait a few days for the next stable release) ?
It the issue persists then please open a new issue, including serial log with debug options enabled.

C-Yasitha, 2020-02-17:
Hello! I found a new solution with the current version. You can get a response from client.write() function and that return value 0, if the client didn't receive the message. The return function has a timeout (or try many times to write) but it is working.
`int res = client.write(sbuf, len);
if(res==0){
 client.stop() // or restart your code.
}`

this worked for me
@NdK73

bee-lover, 2023-08-26:
@C-Yasitha That worked great!

This is how I implemented it for my code

```
int test = client.write(1);

      if(test == 0) {
        client.stop();
        break;
      } 
```