esp8266/Arduino #2591: Compile sketch using build.py

kommancs96, 2016-10-09:
Hello.
Can someone provide me an example on how do I compile a `.ino` sketch using the `build.py` tool?
I have a NodeMCU v1.0 board, and I wish to automate a custom compile process.
Thanks.

igrr, 2016-10-09:
```
$ python build.py -h
usage: build.py [-h] [-v] [-i IDE_PATH] [-p BUILD_PATH] [-l LIBRARY_PATH]
                [-b BOARD_NAME] [-s {512K0,512K64,1M512,4M1M,4M3M}]
                [-f {80,160}] [-m {dio,qio}] [-w {none,all,more}]
                [-o OUTPUT_BINARY] [-k] [--flash_freq {40,80}]
                [--debug_port {Serial,Serial1}] [--debug_level DEBUG_LEVEL]
                sketch_path

Sketch build helper

positional arguments:
  sketch_path           Sketch file path

optional arguments:
  -h, --help            show this help message and exit
  -v, --verbose         Enable verbose output
  -i IDE_PATH, --ide_path IDE_PATH
                        Arduino IDE path
  -p BUILD_PATH, --build_path BUILD_PATH
                        Build directory
  -l LIBRARY_PATH, --library_path LIBRARY_PATH
                        Additional library path
  -b BOARD_NAME, --board_name BOARD_NAME
                        Board name
  -s {512K0,512K64,1M512,4M1M,4M3M}, --flash_size {512K0,512K64,1M512,4M1M,4M3M}
                        Flash size
  -f {80,160}, --cpu_freq {80,160}
                        CPU frequency
  -m {dio,qio}, --flash_mode {dio,qio}
                        Flash mode
  -w {none,all,more}, --warnings {none,all,more}
                        Compilation warnings level
  -o OUTPUT_BINARY, --output_binary OUTPUT_BINARY
                        File name for output binary
  -k, --keep            Don't delete temporary build directory
  --flash_freq {40,80}  Flash frequency
  --debug_port {Serial,Serial1}
                        Debug port
  --debug_level DEBUG_LEVEL
                        Debug level
```

Please comment which of these options you need explained.

kommancs96, 2016-10-09:
1. BOARD_NAME : which board names are valid?
2. LIBRARY_PATH : what if my sketch uses more than one library?
3. --flash_mode {dio,qio} what are dio and qio?

igrr, 2016-10-09:
You can find board names in boards.txt file in the root of the repository. Library path points to a directory where libraries are located (e.g. Documents/Arduino/libraries). You can supply more than one -l argument if you need.
DIO and QIO are flash chip modes (dual IO and quad IO). Depends on your hardware, e.g. NodeMCU uses DIO, Sparkfun Thing uses QIO.

Humancell, 2016-10-10:
Not directly related, but somewhat related, what is the "tool" to use to program a board from the command-line?  Once I've built my code, is there another tool to push the code to my ESP-12F?

kommancs96, 2016-10-11:
Yes, I use this tool to upload my hex files to the board: https://github.com/themadinventor/esptool

kommancs96, 2016-10-11:
So I tried to buid my sketch using the command below:
`python $HOME/build.py WebsocketDimmer.ino -b NodeMCU\ 1.0\ \(ESP-12E\ Module\) --ide_path /Applications/Arduino.app `

This threw the following error:

```
Traceback (most recent call last):
  File "/Users/nagydaniel/build.py", line 145, in <module>
    sys.exit(main())
  File "/Users/nagydaniel/build.py", line 134, in main
    res = compile(tmp_dir, sketch_path, tools_dir, hardware_dir, ide_path, f, args)
  File "/Users/nagydaniel/build.py", line 67, in compile
    p = subprocess.Popen(cmds, stdout=f, stderr=subprocess.STDOUT)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/subprocess.py", line 710, in __init__
    errread, errwrite)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/subprocess.py", line 1335, in _execute_child
    raise child_exception
OSError: [Errno 2] No such file or directory
```

mzpx70, 2016-10-26:
Hi
I faced the same problem with the same result. From Arduino IDE it works fine, but if I try to manually build my project it fails.
I'm using it from Win10 64 bit. I guess the problem can come from directory usage. 

Was there any progress in this topic ?

Thanks!

kommancs96, 2016-10-26:
Hi
I have not made any progress since I asked the question.

devyte, 2017-10-11:
Closing, see #3655 .

closed 2017-10-11 by devyte