esp8266/Arduino #2593: WiFi.status() always not connected.

fabiosoft, 2016-10-10:
### Basic Infos

#### Hardware
Hardware:	        ESP-01
Core Version:      	2.3.0

### Description

`WiFi.status()` always not connected. _WL_CONNECTED_ is never set with version 2.3.0, instead with version 2.0.0 everything works ok.

It outputs a `not connected state` ...while surfing web!! How is it possible? 

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		512k
CPU Frequency:		80Mhz
Flash Mode:			dio
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:	ck

### Sketch

```cpp

#include <Arduino.h>

#include <ESP8266WiFi.h>

int status = WL_IDLE_STATUS;                     // the Wifi radio's status

void setup() {
  Serial.begin(57600);

  // attempt to connect to Wifi network:
  while ( status != WL_CONNECTED) { //<---- never changed! status never changes
    Serial.println("Attempting to connect to...");
    status = WiFi.begin("xxxx", "xxxx");

    // wait 10 seconds for connection:
    delay(1000);
  }

  // once you are connected :
  Serial.print("You're connected to the network");
}

void loop() {
  // check the network status connection once every 10 seconds:
  delay(1000);
  Serial.println(WiFi.status()); // <--- always NOT_CONNECTED status.
}

```

WereCatf, 2016-10-10:
No idea what's wrong, it works just fine on my ESPs.

gilbertobueno, 2016-10-10:
Please, see issue 2186. Maybe it can help you. [https://github.com/esp8266/Arduino/issues/2186](url)

KingModest, 2016-10-22:
Hi, am still having this issue, with AT commands module connects, but if I want to do this, status != WL_CONNECTED - the statement always return true

savankoradia, 2017-03-18:
Same issue here. WiFi.status() returns 255. But still able to connect using AT command.

mitsuhito, 2017-08-12:
I have same issue. my hardware is esp-wroom-02.
https://international.switch-science.com/catalog/2500/

max5555, 2017-08-21:
A workaround is to replace 
    `while ( status != WL_CONNECTED) {`
with
    `while (WiFi.localIP().toString() == "0.0.0.0") {`

devyte, 2017-10-11:
@fabiosoft is this issue still valid with latest git?

Fly4TheKing, 2018-01-01:
I just tested with the latest, and it still does the same for me. Sometimes (~30%) works, but usually returns WL_DISCONNECTED indefinitely.

muthu29, 2018-02-03:
Did anyone find the solution. I have the same problem always Wifi.status() is not connected and when I tried to print the Wifi.status() it returns value 0 for five times and a 6 once. I am wondering how this will by any chance give a String value of WL_CONNECTED :(

Fly4TheKing, 2018-02-04:
@muthu29 the workaround by @max5555 is good. To answer your question about return value, WL_CONNECTED is an enum value that equates to 3. It's defined in wl_definitions.h:

`typedef enum {
    WL_NO_SHIELD        = 255,   // for compatibility with WiFi Shield library
    WL_IDLE_STATUS      = 0,
    WL_NO_SSID_AVAIL    = 1,
    WL_SCAN_COMPLETED   = 2,
    WL_CONNECTED        = 3,
    WL_CONNECT_FAILED   = 4,
    WL_CONNECTION_LOST  = 5,
    WL_DISCONNECTED     = 6
} wl_status_t;`

devyte, 2018-07-12:
This should be addressed in latest master.
@fabianoms @Fly4TheKing @muthu29 @KingModest could someone please retest?

muthu29, 2018-07-12:
For me it was a Typo in the SSID Password. It is working when I corrected it.

devyte, 2018-07-13:
Closing due to recent lwip glue fixes and WiFi::status() testing.

closed 2018-07-13 by devyte

slow-down, 2018-11-07:
How can it be, that it's still not fixed, even with 2.4.2 ... It's been more than 3 years. I had to use the workaround to fix it..

ScientificNerd, 2019-09-29:
I tried the workaround, and it failed.  So, I printed "WiFi.localIP().toString()" to the console, and to my amazement, it did not contain "0.0.0.0", but "(IP unset)", so I replaced the "0.0.0.0" with  "(IP unset)" in the while statement, and it still didn't connect.  Then I decided to check the router connections, and found that the IP had been issued, and the ESP was connected to the router. 

At that point, I thought maybe the router was taking a bit longer to get everything done before the "while" timeout, so I added 15 seconds, and also decided to change the "while" statement to:

while (WiFi.localIP().toString() != IP) {
    WiFi.begin(ssid, password) ;
    // failed, retry
    Serial.print(".");
    delay(20000);
}
or
while (WiFi.localIP().toString() == "(IP unset)") {
    WiFi.begin(ssid, password) ;
    // failed, retry
    Serial.print(".");
    delay(20000);

Where "IP" was set as a global char* constant, as I had set the IP address for the ESP MAC in the router.  Both were tested, and work.
  
I am using a Netgear WNDR3000-2, which takes its sweet time connecting.  Since I need my ESP addresses to be fixed, I used the router to save the addresses rather than try to get the ESP's to request a fixed IP.

Now that this stumbling block is passed, I can finally move on to productive parts of my project.

d-a-v, 2019-09-29:
Please try and replace
```
while (WiFi.localIP().toString() == "(IP unset)") {
```
by
```
while (!WiFi.localIP().isSet()) {
```
Checking against a string is always worse when there are other means.

ScientificNerd, 2019-09-30:
That works very well, Thanks!  You are quite right about checking against strings, as it takes more program memory, as well as more time than checking for bool.

ebrahimkarimi, 2020-06-01:
for me, none of the above worked!!!
after digging deep i was able to debug it find a solution that may work for everyone
i just enabled debugging in Arduino IDE and used Serial to connect the esp8266-12e to my phone and debugged it using USB Serial Terminal.
i found out there was messages like `evt : <number>` after a searched i realized there is WiFi events. so i found [this](https://roei.stream/2018/08/07/esp8266-wifi-events-list-with-arduino/). i realized connect event triggers after `WiFi.begin(ssid,pwd)` but StationGotIP event wont get triggered correctly:
[image]
you can see that after connection established, event 0 triggered ([event docs](https://arduino-esp8266.readthedocs.io/en/latest/esp8266wifi/generic-examples.html)) but event 3 ([EVENT_STAMODE_GOT_IP](https://github.com/esp8266/Arduino/blob/8ee67ab2b53463466fd9f035eef2c542ad9a6775/tools/sdk/include/user_interface.h#L470)) didn't.
so i think the best and safe solution would be registering a [EVENT_STAMODE_CONNECTED](https://github.com/esp8266/Arduino/blob/8ee67ab2b53463466fd9f035eef2c542ad9a6775/tools/sdk/include/user_interface.h#L467) event.

some usable links:
[Arduino ESP8266 - wifi evt: 3 and wifi evt: 7 meaning?](https://arduino.stackexchange.com/a/52654)

mazorvanunu, 2021-04-27:
does anybody has a problem when connecting the usb the mice starts moving on its own? anybody encouterd a problem with wifi.begin crashes the system? anyone encoutered with sudden outputs that never occured without changing anything with the code\system configs??? anybody encoutered machine just crashing without output then after a few minutes an exception trace is printed? anybody encountered file system crashing (vfs_write...) when calling to wifi.begin? same with spiffs? 

WOW - i wouldve asked in the esp forum but honestly that never do any good. they never give reasonable inputs except check usb power... i have changed cables and have 3 different esp boards.

what a load of cr....!

mazorvanunu, 2021-04-27:
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:10944
load:0x40080400,len:6360
entry 0x400806b4
config content(285): 

NEVER SAW THIS BEFORE THEN SUDDENLY IT APPEARS, i mean wtf what a fucking joke. loved the way esp presents as if everything is honest and fine - WHY THE HELL DO THEY HAVE 3 DIFFERENTS KIND OF "FILE SYSTEMS" ??????????? you only need one that works.

mazorvanunu, 2021-04-27:
why would the wifi begin entry even touch the flash memory/driver??? there is no logical reason except the fact that the people "working" (those amazing brilliant engineers) over there want to fucking make everything complex OR spy on everyone or some shit like that. JUST GO FUCKING HOME. WORST CHIP I EVER CAME ACROSS, omg the lost time, WHAT A FARCE!

mazorvanunu, 2021-04-27:
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:10944
load:0x40080400,len:6360
entry 0x400806b4

look it just goes on and on, this is a different "exception trace" of some sorts.... SW_CPU_RESET

3 other kinds also appear at random (in the same line of code.)

mazorvanunu, 2021-04-27:
if you want to use a chip, do not use ESP. just dont, im telling you, dig around - you will thank me later.

d-a-v, 2021-04-27:
Thanks @mazorvanunu we got your message

Bighoneypot, 2021-04-27:
Please, try with another SSID and Password, could be a problem with your router. Use Hotspot in your smartphone and try.
Leave the hotspot screen open and watch if your esp is connected on your hotspot.

mazorvanunu, 2021-04-27:
if you want to use a chip, do not use ESP. just dont, im telling you, dig around - you will thank me later.

wasted an entire year of my life on this project which always gets stock on various "platform" problems - honestly, im pretty devastated so sure everybody there has nice little degrees - promising us things are working well but always fine prints start showing up. 

never tell you the real deal, just want you to look at the bright side like this is a yogi course and not real life.

when i see a client i immediately tell him the bad news - i want him to know and be ready - not because i'm such a good person but because to not do so will hurt us both much more! 

just look at this code - if you go into esp this is the NORM!

 if (WiFi.isConnected() || WiFi.reconnect())
{
		const char * theSoCalledIP = WiFi.localIP().toString().c_str();

		if (strcmp(theSoCalledIP, "(IP unset)") == 0 || strcmp(theSoCalledIP, "0.0.0.0") == 0)
		{
			rcount++;
			continue;
		}
}

JUST WOW - no other word to describe it. this suppose to be embedded, things working straight even if on the most basic of levels. 
if i could ask the espressif crew for a single thing it will be - look at your own documentation, on your toolchain, tell people (who are not familiar with this "platform") to have a good and listen to their insights. TAKE OWNERSHIP OF YOUR CODE AND PLATFORM!.

mazorvanunu, 2021-04-27:
also my business associates has already bought a decent numbers of this chips. what a waste. would've preferred to give this money to the needy, or even just burn it away.

mazorvanunu, 2021-04-27:
come delete my comments "contributor" or maybe tell me to check up on my power cord. that will surely make a huge difference, i feel engulfed with your support. <3

d-a-v, 2021-04-27:
@mazorvanunu
This is a closed issue. Please open a new one with your data, don't forget to fill the issue template and provide an MCVE.
I deleted offending comments.