esp8266/Arduino #2594: ESP hang by Core 2.3.0

Bajajar, 2016-10-11:
### Basic Infos

#### Hardware
Hardware:			ESP-12E
Core Version:      	2.3.0

### Description

I flashed a sketch with PubSubClient version 2.6.0 and Core version 2.3.0. The problem occurred right after ESP connected to MQTT broker. The ESP hang about 30 seconds, then it could send and receive messages. Not only version 2.3.0 but also version 2.2.0 has the same problem too. However, the problem won't happen with Core version 2.1.0. 

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		4MB/3MB
CPU Frequency:		80Mhz
Flash Mode:			dio
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		ck

bHogan2, 2016-10-13:
I too have sketches that reliably connect with the 2.1.0 core and fail to connect. or only connect a few times, with the 2.3.0 core. When a failure occurs, it happens in this loop:

WiFi.begin(ssid, password);           // connect to the network
while (WiFi.status() != WL_CONNECTED) {
    delay(500);
  }
 
If I reset my router, it proceeds from the hangup, and I am able to connect a few times. (Not sure what that points to.)

drmpf, 2016-10-13:
I had a similar problem and 'fixed' it by going back to V2.2.0 core

bHogan2, 2016-10-14:
Fixed my problem with the workaround described in Issue 2186. i.e.
CHANGED:
 WiFi.begin(ssid, password);
TO:
  if (WiFi.status() != WL_CONNECTED) {  // FIX FOR USING 1.3.0 CORE (only .begin if not connected)
    WiFi.begin(ssid, password);         // connect to the network
  }

Bajajar, 2016-10-14:
` (WiFi.status() != WL_CONNECTED) {
WiFi.waitForConnectResult();
} ` 
Unfortunately the above solution doesn't resolve my case.

kendo55, 2016-10-23:
Fixed my problem too !!!

if (WiFi.status() != WL_CONNECTED) {  // FIX FOR USING 2.3.0 CORE (only .begin if not connected)
    WiFi.begin(ssid, password);       // connect to the network
}    
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
 }

Bajajar, , try it with a ssid-name, without special characters

Bajajar, 2016-10-24:
Fixed by rework coding with PubSubClient. Nothing wrong with ESP8266WiFi. Thanks you guys.

closed 2016-10-24 by Bajajar