esp8266/Arduino #2596: Reading EEPROM issue

castro2487, 2016-10-11:
Hi to all.

    Thanks a lot for this library it's very usefull, i'm saving my SSID and PASS that i get from my browser and there is no problem saving the data because each value of the string that i save i print in serial to see if is the original or not, but when i try to read that information from the memory it appear in blank or with 0, i make a litlle sketch only with, clean, write and read from EEPROM to see if there is another instruction that have conflict with this part and still the same, if someone can help i appreciatte.

     here is the skecth that i made for test purpose ("i take some parts of the examples").

#include <EEPROM.h>

String A= "hola";

void setup() {
 
     EEPROM.begin(512);
     Serial.begin(9600);
     limpiarMemoria(); //clean EEPROM
     grabarMemoria(); // write EEPROM
     leerMemoria(); // read EEPROM
     
     
}

void loop() {

  }

void limpiarMemoria () {

  // write a 0 to all 4 bytes of the EEPROM
  for (int i = 0; i < 4; i++){
    EEPROM.write(i, 0);
    
  }

    EEPROM.end();
}

void grabarMemoria () {
  
  for (int i=0; i<A.length();i++){
    EEPROM.write(i,A[i]);  
    Serial.println(A[i]);
    delay (1000);
  }
  
    EEPROM.commit();
    
}

void leerMemoria () {
  
  String epass = "";
  
  for (int i=0; i<A.length(); i++){
  
    epass += char(EEPROM.read(i));
    
    }
    
  Serial.print("PASS: ");
  Serial.println(epass); 
   
  
}

pieman64, 2016-10-11:
I think you are missing EEPROM.commit();

castro2487, 2016-10-11:
thanks for your prompt response pieman64, in which one of the routines?

pieman64, 2016-10-11:
After EEPROM.write().

castro2487, 2016-10-11:
thanks pieman64, i made the corrections and it works, i put in my main code and i can retrieve my SSID but when i try to get the password back it's blank, i put in the example that i made the 2 thing that i want to save but give me an error, because i want to make a non specific quantity of space for SSID and PASS i want that can be dinamic to establish for any name and pass that you could have in you WIFI.

#include <EEPROM.h>

String A= "hola";
String B= "mundo";

void setup() {
 
     EEPROM.begin(512);
     Serial.begin(9600);
     limpiarMemoria();
     grabarEspacio1Memoria();
     grabarEspacio2Memoria();
     leerMemoria();
     
     
}

void loop() {

  }

void limpiarMemoria () {

  // write a 0 to all 512 bytes of the EEPROM
  for (int i = 0; i < B.length(); i++){
    EEPROM.write(i, 0);
    
  }

    EEPROM.commit();
}

void grabarEspacio1Memoria () {
  
  for (int i=0; i<A.length();i++){
    EEPROM.write(i,A[i]);  
    Serial.println(A[i]);
    delay (1000);
  }
  
    EEPROM.commit();
    
}

void grabarEspacio2Memoria() {

  for (int i>A.length(); i<B.length();i++){
    EEPROM.write(i,B[i]);  
    Serial.println(B[i]);
    delay (1000);
  }
  
    EEPROM.commit();
  
}

void leerMemoria () {
  
  String essid = "";
  String epass = "";
  
  for (int i=0; i<A.length(); i++){
  
    essid += char(EEPROM.read(i));
    
    }
    
  Serial.print("SSID: ");
  Serial.println(essid); 

  int C = A.length();

   for (int i>C; i<B.length(); i++){

    epass += char(EEPROM.read(i));
    
   }
   
   Serial.print("PASS:  ");
   Serial.println(epass);
  
}

WereCatf, 2016-10-12:
`for (int i>C; i<B.length(); i++){`

That's not how you use a for-clause, you're initializing i wrong. Also, the end-condition is also wrong for it to work. It should be:

`for (int i=C; i<(B.length()+C); i++){`

You have a similar error in multiple places, fix them all.

castro2487, 2016-10-12:
thanks for the comment, i made the correction and here is the fixed one, i can save only one of the 2 string. i check with serial to see what is saved in memory.

#include <EEPROM.h>

String A= "hola";
String B= "mundo";
int sizeofA;
int sizeofB;
int c; 

void setup() {
 
     EEPROM.begin(512);
     Serial.begin(9600);
     sizeofA = A.length();
     sizeofB = B.length();
     c = sizeofA + sizeofB;
     limpiarMemoria();
     grabarEspacio1Memoria();
     grabarEspacio2Memoria();
     leerMemoria();
     
     
}

void loop() {

  }

void limpiarMemoria () {

  for (int i = 0; i < 9; i++){
    EEPROM.write(i, 0);
    
  }

    EEPROM.commit();
}

void grabarEspacio1Memoria () {

  
  for (int i = 0; i < sizeofA; i++){
    EEPROM.write(i,A[i]);  
    Serial.print(A[i]);
    delay (1000);
  }

    EEPROM.commit();
    
}

void grabarEspacio2Memoria() {

  for (int i= sizeofA+1; i< c; i++){
    EEPROM.write(i,B[i]);  
    Serial.print(B[i]);
    delay (1000);
  }
  
    EEPROM.commit();
  
}

void leerMemoria () {
  
  String essid = "";
  String epass = "";
//  int C= A.length();
  
  for (int i= 0; i < sizeofA; i++){
  
    essid += char(EEPROM.read(i));
    
    }
    
  Serial.print("SSID: ");
  Serial.println(essid); 

   for (int i = sizeofA+1; i < c; i++){

    epass += char(EEPROM.read(i));
    
   }
   
   Serial.print("PASS:  ");
   Serial.println(epass);
  
}

WereCatf, 2016-10-12:
You introduced new bugs. Take a look at _grabarEspacio2Memoria()_: _B[0]_ would be the first character in the string _B_, but _i_ no longer begin at 0, now does it? You basically try to start writing to EEPROM from _B[sizeofA+1]_, which obviously just writes garbage in your EEPROM, not the value you wanted.

castro2487, 2016-10-12:
yes i want to write in the next space where the value of A finish. i want to write those 2 values and retrieve it

WereCatf, 2016-10-12:
No, you are writing to the correct place in EEPROM, but you are writing garbage in there.

castro2487, 2016-10-12:
i checked and you were right, i made the correction but i can only save the first letter of the second string. this is the fix that i made in the process  grabarEspacio2Memoria();

<void grabarEspacio2Memoria() {

  for (int i= sizeofA+1; i< c; i++){
    int j=0;
    EEPROM.write(i,B[j]);  
    Serial.println(B[j]);
    j++;
    Serial.println(j);
    delay (1000);
  }
  
    EEPROM.commit();
  
}
>

WereCatf, 2016-10-12:
`int j=0;` is inside your loop, so it sets j to 0 every single iteration of your for-loop. Move the line before your for-loop:
```
void grabarEspacio2Memoria() {
  int j=0;
  for (int i= sizeofA+1; i< c; i++){
    EEPROM.write(i,B[j]);
    Serial.println(B[j]);
    j++;
    Serial.println(j);
    delay (1000);
  }
  EEPROM.commit();
}
```

castro2487, 2016-10-12:
yes i see it. again thanks a lot, now it works how i want it, i'm going to apply all this change to the principal code that i made; that the length of each string received is different, if works ok i will tell you. again thanks a lot

devyte, 2017-10-11:
Resolved and/or off topic per #3655 , closing.

closed 2017-10-11 by devyte