esp8266/Arduino #2597: dead code in spiffs_api.h

everslick, 2016-10-12:
bool seek(uint32_t pos, SeekMode mode) override
    {
        CHECKFD();

        int32_t offset = static_cast<int32_t>(pos);
        if (mode == SeekEnd) {
            offset = -offset;
        }
        auto rc = SPIFFS_lseek(_fs->getFs(), _fd, pos, (int) mode);
        if (rc < 0) {
            DEBUGV("SPIFFS_lseek rc=%d\r\n", rc);
            return false;
        }

        return true;
    }

int32_t offset is calculated but not used afterwards.

devyte, 2017-10-11:
This issue looks like a bug to me, and is still present in latest git. Probably a typo:
`        auto rc = SPIFFS_lseek(_fs->getFs(), _fd, pos, (int) mode);
`
should read:
`        auto rc = SPIFFS_lseek(_fs->getFs(), _fd, offset, (int) mode);
`

devyte, 2018-02-04:
@everslick please try #4287 .

PR #4287 (2018-02-06): Fix spiffs lseek typo for seek_end
Fixes #2597 .
closed 2018-02-06 by devyte