esp8266/Arduino #2598: Update without content-length?

ecsfang, 2016-10-12:
The update doesn't support transfer encoding, so http 1.0 is used.

// use HTTP/1.0 for update since the update handler not support any transfer Encoding
    http.useHTTP10(true);

But, in in the cases I have tried, the server doesn't report any payload length (ie no content-length), and therefore the update fails ("Server not Report Size"). See example below.

Is there any chance that the library will be updated to support this, i.e. missing content-length in the GET-response?

Cheers,
Thomas
---------------------------

(If the example is done using http 1.1, transfer is done using transfer_encoding)

$ curl -0 -v --output sonoff.bin http://api.my_update_server.se/arduino/test.ino.bin
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
> **GET http://api.my_update_server.se/arduino/test.ino.bin HTTP/1.0**
> Host: api.my_update_server.se
> User-Agent: curl/7.47.0
> Accept: */*
> Proxy-Connection: Keep-Alive
> 
< HTTP/1.1 200 OK
< Server: nginx/1.9.9
< Date: Tue, 11 Oct 2016 06:58:48 GMT
< Content-Type: application/octet-stream
< Last-Modified: Sun, 09 Oct 2016 15:42:30 GMT
< magicmarker: 1
< X-Cacheable: YES
< X-Varnish: 782273366
< Via: 1.1 varnish
< X-Cache: HIT from Backend
< Connection: close
< Age: 244
< 
{ [6589 bytes data]
100  402k    0  402k    0     0  1348k      0 --:--:-- --:--:-- --:--:-- 1346k
* Closing connection 0

The result (using curl on my computer) is that the file is downloaded ok, but using ESP8266httpupdate the download fails with "Server not Report Size".

devyte, 2017-10-11:
@ecsfang No basic info, no IDE settings, no sketch. I assume you are referring to ESP8266httpupdate example?
Is this issue still valid with latest git?

arihantdaga, 2018-08-01:
On ESP8266 12S, Arduio, ESP8266httpupdate example
@devyte , I tried to download firmware using a simple link like this - 
http://myserver.com/firmwares/my_file_key

I am using some stream on my server to download from aws and pipe the data to response.. and i am not able to set content-length.
Is it mandatory to give content-length header in response or is there a way to skip that... ? 

Thanks you.

earlephilhower, 2019-07-25:
Closing due to lack of MCVE, release info, etc.  If there are ongoing problems, please open a new issue and fill out the requested fields so we can look into it.

closed 2019-07-25 by earlephilhower