esp8266/Arduino #2601: ESPhttpUpdate Fatal Exception 3 (LoadStoreErrorCause)

jcrockercheckfluid, 2016-10-12:
### Basic Infos

#### Hardware
Hardware:	        Adafruit ESP8266 Huzzah
Core Version:      	2.3.0

### Description

Upon trying to perform an OTA update using HTTPS, I receive exception 3 (LoadStoreErrorCause). The code provided below is a minimal modification of the "httpUpdate" sketch provided as an example for OTA through the Arduino IDE. I have only entered my Wifi credentials, file url and certificate fingerprint in the appropriate locations. Attempting OTA update using HTTP is successful. The update file is hosted on IBM's Object Storage.

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		4MB (3M SPIFFS)
CPU Frequency:      80Mhz
Flash Mode:	       dio
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		nodemcu

### Sketch

```
#include <Arduino.h>

#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>

#include <ESP8266HTTPClient.h>
#include <ESP8266httpUpdate.h>

#define USE_SERIAL Serial

ESP8266WiFiMulti WiFiMulti;

void setup() {

    USE_SERIAL.begin(115200);
    // USE_SERIAL.setDebugOutput(true);

    USE_SERIAL.println();
    USE_SERIAL.println();
    USE_SERIAL.println();

    for(uint8_t t = 4; t > 0; t--) {
        USE_SERIAL.printf("[SETUP] WAIT %d...\n", t);
        USE_SERIAL.flush();
        delay(1000);
    }

    WiFiMulti.addAP("SSID", "PASSWORD");

}

void loop() {
    // wait for WiFi connection
    if((WiFiMulti.run() == WL_CONNECTED)) {

        t_httpUpdate_return ret = ESPhttpUpdate.update("https://dal.objectstorage.open.softlayer.com/v1/AUTH_a483baa927ab4ec386c2ee5466ba32a5/Binaries/sketch_oct06a.ino.bin?temp_url_sig=075f43749b6d9876925ad36e8a19118bd8628eec&temp_url_expires=1507815047", "", "29 9C E1 CF AD 48 EF 2E 0E F5 1A 03 48 EC F5 70 2A E9 18 C6");
        //t_httpUpdate_return  ret = ESPhttpUpdate.update("https://server/file.bin");

        switch(ret) {
            case HTTP_UPDATE_FAILED:
                USE_SERIAL.printf("HTTP_UPDATE_FAILD Error (%d): %s", ESPhttpUpdate.getLastError(), ESPhttpUpdate.getLastErrorString().c_str());
                break;

            case HTTP_UPDATE_NO_UPDATES:
                USE_SERIAL.println("HTTP_UPDATE_NO_UPDATES");
                break;

            case HTTP_UPDATE_OK:
                USE_SERIAL.println("HTTP_UPDATE_OK");
                break;
        }
    }
}
```

### Debug Messages

Below is the output of the sketch with debug mode on:
```
[SETUP] WAIT 4...
[SETUP] WAIT 3...
wifi evt: 0
[SETUP] WAIT 2...
[SETUP] WAIT 1...
wifi evt: 3
[WIFI][APlistAdd] add SSID: CF33backup
[HTTP-Client][begin] url: https://dal.objectstorage.open.softlayer.com/v1/AUTH_a483baa927ab4ec386c2ee5466ba32a5/Binaries/sketch_oct06a.ino.bin?temp_url_sig=075f43749b6d9876925ad36e8a19118bd8628eec&temp_url_expires=1507815047
[HTTP-Client][begin] host: dal.objectstorage.open.softlayer.com port: 443 url: /v1/AUTH_a483baa927ab4ec386c2ee5466ba32a5/Binaries/sketch_oct06a.ino.bin?temp_url_sig=075f43749b6d9876925ad36e8a19118bd8628eec&temp_url_expires=1507815047
[HTTP-Client][begin] httpsFingerprint: 29 9C E1 CF AD 48 EF 2E 0E F5 1A 03 48 EC F5 70 2A E9 18 C6
[hostByName] request IP for: dal.objectstorage.open.softlayer.com
[hostByName] Host: dal.objectstorage.open.softlayer.com IP: 173.192.119.147
:ref 1
please start sntp first !
State:	sending Client Hello (1)
:sent 72
:rn 1440
:rd 5, 1440, 0
:rdi 1440, 5
:rd 74, 1440, 5
:rdi 1435, 74
State:	receiving Server Hello (2)
:rd 5, 1440, 79
:rdi 1361, 5
:rch 1440, 20
:rd 1356, 1440, 84
:rdi 1356, 1356
:c 1356, 1440, 1460
:rd 20, 20, 0
:rdi 20, 20
:c0 20, 20
:rn 1440
:rd 1440, 1440, 0
:rdi 1440, 1440
:c0 1440, 1440
:rn 1261
:rd 1252, 1261, 0
:rdi 1261, 1252
State:	receiving Certificate (11)
:rd 5, 1261, 1252
:rdi 9, 5
:rd 4, 1261, 1257
:rdi 4, 4
:c0 4, 1261
State:	receiving Server Hello Done (14)
State:	sending Client Key Exchange (16)
State:	sending Finished (16)
:sent 267
:sent 91
:rn 91
:rd 5, 91, 0
:rdi 91, 5
:rd 1, 91, 5
:rdi 86, 1
:rd 5, 91, 6
:rdi 85, 5
:rd 80, 91, 11
:rdi 80, 80
:c0 80, 91
State:	receiving Finished (16)
[HTTP-Client] connected to dal.objectstorage.open.softlayer.com:443
:ref 2
domain name: 'dal.objectstorage.open.softlayer.com'
:ur 2
:sent 613
:rn 1440
:rch 1440, 20
:rd 5, 1460, 0
:rdi 1440, 5
ssl->need_bytes=16448 > 6859
failed to grow plain buffer
check_poison_block is called for free block 0x3fff5110
Fatal exception 3(LoadStoreErrorCause):
epc1=0x4010011d, epc2=0x00000000, epc3=0x00000000, excvaddr=0x40037328, depc=0x00000000

Exception (3):
epc1=0x4010011d epc2=0x00000000 epc3=0x00000000 excvaddr=0x40037328 depc=0x00000000

ctx: cont 
sp: 3fff3090 end: 3fff34f0 offset: 01a0

>>>stack>>>
3fff3230:  3fff3280 00000000 3fff5110 401003f5  
3fff3240:  3fff23c0 00000280 00000280 4010020c  
3fff3250:  00000000 00000000 3fff5114 401008f4  
3fff3260:  00000000 3fff4c0c 3fff5034 402336b5  
3fff3270:  0000140e 3fff4c0c fffffefb 40203e41  
3fff3280:  00000000 3fff4c0c 00000000 40203e27  
3fff3290:  0000140e 3fff4cbc 00000000 40203ffd  
3fff32a0:  0000140e 00000000 00000000 4020404b  
3fff32b0:  3ffec278 3fff2400 3fff33f0 40207f02  
3fff32c0:  00000000 3fff2400 3fff33f0 40205324  
3fff32d0:  00000000 3ffea0ec 3fff3308 40206fbc  
3fff32e0:  00000000 00000001 00000000 40205098  
3fff32f0:  00000000 00000000 00000000 00000000  
3fff3300:  3fffa31c 0000000f 00000000 00000000  
3fff3310:  00000000 3fff4c9c 3fff4c9c 40207017  
3fff3320:  3ffea974 00000000 3fff4c9c 00000000  
3fff3330:  00000000 00000000 3fff33f0 40205448  
3fff3340:  3fff4ca8 00000000 3fff4c90 40207209  
3fff3350:  3ffead04 00000000 3fff33f0 40207984  
3fff3360:  3ffea974 000001b0 3fff33f0 40204cbb  
3fff3370:  3fff33ac 00000000 3fff4a94 3fff34a0  
3fff3380:  3fff220c 3ffead04 3fff33f0 4020547a  
3fff3390:  3fff220c 3ffead04 3fff33f0 40205b23  
3fff33a0:  00000000 00000000 00000000 3ffead04  
3fff33b0:  00000000 00000000 000001be 4010020c  
3fff33c0:  3fff34b8 3fff34ac 00000000 401008f4  
3fff33d0:  40230353 00000030 3fff346c 3fff34a0  
3fff33e0:  3fff34b8 3fff220c 3fff34ac 40205e2d  
3fff33f0:  3fff48a4 3fff4cbc 3fff4d5c 0000002f  
3fff3400:  00000024 530001bb 3f011f40 3fff4d9c  
3fff3410:  0000009f 0000009a 3fff4a7c 0000000f  
3fff3420:  00000005 3fff4e4c 0000011f 0000011e  
3fff3430:  3fff4abc 0000001f 00000013 3fff4aec  
3fff3440:  0000000f 00000000 3fff4c90 00000001  
3fff3450:  ffffffff ffffffff 3fff3400 00000000  
3fff3460:  00000000 00000000 00000000 00000000  
3fff3470:  00000000 00000000 3fff34a0 402070e2  
3fff3480:  3ffe93f8 3fff2158 3fff2400 3fff24bc  
3fff3490:  3fffdad0 00000000 3fff220c 40202259  
3fff34a0:  3fff4a2c 0000003f 0000003b 3fff4884  
3fff34b0:  0000000f 00000000 3fff47a4 000000cf  
3fff34c0:  000000c6 00000000 3fff2400 4020636d  
3fff34d0:  feefeffe 00000000 3fff24b4 40207b90  
3fff34e0:  feefeffe feefeffe 3fff24d0 40100984  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v00000000
~ld
```

jcrockercheckfluid, 2016-10-12:
The decoded stack is shown below:
`0x401003f5: check_poison_block at C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\cores\esp8266\umm_malloc/umm_malloc.c line 835
0x4010020c: _umm_free at C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\cores\esp8266\umm_malloc/umm_malloc.c line 1287
0x401008f4: free at C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\cores\esp8266\umm_malloc/umm_malloc.c line 1733
0x402336b5: ssl_free at /Users/igrokhotkov/e/axtls/e1/ssl/tls1.c line 260
0x40203e41: SSLContext::_readAll() at C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266WiFi\src/WiFiClientSecure.cpp line 587
0x40203e27: SSLContext::_readAll() at C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266WiFi\src/WiFiClientSecure.cpp line 587
0x40203ffd: SSLContext::available() at C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266WiFi\src/WiFiClientSecure.cpp line 587
0x4020404b: WiFiClientSecure::connected() at C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266WiFi\src/WiFiClientSecure.cpp line 587
0x40207f02: HTTPClient::connected() at C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266HTTPClient\src/ESP8266HTTPClient.cpp line 245
0x40205324: HTTPClient::handleHeaderResponse() at C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266HTTPClient\src/ESP8266HTTPClient.cpp line 130
0x40206fbc: String::~String() at C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\cores\esp8266/WString.cpp line 519
0x40205098: HTTPClient::sendHeader(char const*) at C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266HTTPClient\src/ESP8266HTTPClient.cpp line 130
0x40207017: String::changeBuffer(unsigned int) at C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\cores\esp8266/WString.cpp line 519
0x40205448: HTTPClient::sendRequest(char const*, unsigned char*, unsigned int) at C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266HTTPClient\src/ESP8266HTTPClient.cpp line 130
0x40207209: String::operator=(char const*) at C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\cores\esp8266/WString.cpp line 519
0x40207984: operator new[](unsigned int) at C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\cores\esp8266/abi.cpp line 57
0x40204cbb: HTTPClient::collectHeaders(char const**, unsigned int) at C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266HTTPClient\src/ESP8266HTTPClient.cpp line 130
0x4020547a: HTTPClient::GET() at C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266HTTPClient\src/ESP8266HTTPClient.cpp line 130
0x40205b23: ESP8266HTTPUpdate::handleUpdate(HTTPClient&, String const&, bool) at C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266httpUpdate\src/ESP8266httpUpdate.cpp line 36
0x4010020c: _umm_free at C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\cores\esp8266\umm_malloc/umm_malloc.c line 1287
0x401008f4: free at C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\cores\esp8266\umm_malloc/umm_malloc.c line 1733
0x40230353: wifi_station_get_connect_status at ?? line ?
0x40205e2d: ESP8266HTTPUpdate::update(String const&, String const&, String const&) at C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266httpUpdate\src/ESP8266httpUpdate.cpp line 36
0x402070e2: String::String(char const*) at C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\cores\esp8266/WString.cpp line 519
0x40202259: loop at C:\Users\jcrocker\AppData\Local\Temp\arduino_modified_sketch_123064/httpUpdate.ino line 43
0x4020636d: HardwareSerial::setDebugOutput(bool) at C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\cores\esp8266/HardwareSerial.cpp line 177
0x40207b90: loop_wrapper at C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\cores\esp8266/core_esp8266_main.cpp line 123
0x40100984: cont_norm at C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\cores\esp8266/cont.S line 109`

jcrockercheckfluid, 2016-10-12:
Any help would be very much appreciated! Thank you!

jcrockercf, 2016-10-13:
I am encountering the same exception trying to access a file on github. Compared to the example code I posted, the only modification I have made to the sketch is changing line 43 to:
`t_httpUpdate_return ret = ESPhttpUpdate.update("https://raw.githubusercontent.com/jcrockercf/CompiledBinary/master/sketch_sep14a.ino.bin", "", "21 99 13 84 63 72 17 13 B9 ED 0E 8F 00 A5 9B 73 0D D0 56 58");`

jcrockercf, 2016-10-28:
Is there any information that I could provide to better assist with this issue?

timlag1305, 2016-11-14:
I have also been having trouble with `umm_malloc` as well. I don't know if there's a recommended workaround or not.

jcrockercf, 2016-11-14:
@timlag1305 Am I correct in believing that when performing OTA, the arduino is running out of space in the heap?

timlag1305, 2016-11-14:
This might be the problem in your case but I am not using OTA. This happens when I connect to a server as a client with TLS 1.2 (specifically api.groupme.com/v3). I definitely think it has to be a problem with running out of heap space, but I would have hoped that it would gracefully fail instead of throwing a CPU exception.

chocol4te, 2017-01-13:
I am having this exact same issue and stack, but it only occurs after making a HTTPS request. Otherwise it will update fine.  Any ideas?

timlag1305, 2017-01-13:
Honestly, I just limited the number/size of the requests and then tried to keep my code off of the heap. I was able to get a limited version of my intended functionality working consistently when I did that.

I would love to figure out if malloc or free was implemented unsafely, but I lost my motivation.

chocol4te, 2017-01-14:
Okay, Ill try that. Thanks for your help! :)

devyte, 2017-10-11:
@jcrockercf is this issue still valid with latest git?

whyameye, 2017-12-03:
I had this problem with 2.4.0-rc2 but not with today's git (commit 7b09ae5f69f8dccb5152d128c8484ec9a18adb11)

devyte, 2017-12-03:
Closing as can't reproduce, and lack of feedback in over a month.

closed 2017-12-03 by devyte