esp8266/Arduino #2603: Switching to external usb power changes behavior GPIO0 (D3)

Imara90, 2016-10-13:
### Basic Infos
I'm using the NodeMCU 1.0 to detect movement with PIR and enable the user to press a button and turn on a light show with sounds. So I power a PIR, a button, a string of neopixels (39) and a 3W transducer. I've powered everything but the button with Vin, which is driven by a 5V usb cable. 

#### Hardware
Hardware:			NodeMCU 1.0  ESP12-E

### Description
Problem description

My entire system is working fine when connected to my laptop usb port. However once I try to power the setup using a usb charger (like a phone charger), the system is irresponsive to the button I connected. Both setups seem to have 3.3V running through the button once it's pressed, however when on external power, the sketch does not proceed. I tried to upload the standard Arduino Debounce sketch and it behaves exactly the same. It works while connected to my laptop but once I connect it to external power, it stops. 

### Settings in IDE

Module:  			NodeMCU 1.0 12-E
Flash Size: 		4MB
CPU Frequency:	80Mhz
Upload Using:		SERIAL
Reset Method:		nodemcu

I connected the button to D3, or GPIO0 and the LED to D4, or GPIO2.

### Sketch
```
/******************************************************
   De buitenbloem - prototype vs1
   Imara
*/

// dfplayer
#include <DFPlayer_Mini_Mp3.h>
#include <SoftwareSerial.h>

// adafruit neopixel
#include <Adafruit_NeoPixel.h>
#ifdef __AVR__
#include <avr/power.h>
#endif

// esp
#include <ESP8266WiFi.h>

// json streamer
#include <JsonListener.h>
#include <JsonStreamingParser.h>

// weather client librarier
#include "WundergroundClient.h"
#include "TimeClient.h"
#include <Time.h>
#include <TimeLib.h>
#include <TimeAlarms.h>
#include "ThingspeakClient.h"

// pin definities
// Serial uses UART0, WHICH IS MAPED TO GPIO1 (TX) and GPIO3 (RX)
#define KNOP D3    // GPIO0 needs a pullup resistor to 3.3 - Data Terminal Ready
#define KNOPLED D4  // GPIO2 IS  alternative TX TODO how to turn of alternative TX??
#define PIR D0      // GPIO
#define NEO D6      // GPIO04
#define BUSY D5     // GPIO05 TODO weghalen?

// GPIO15, GPIO0 en GPIO2 worden gebruikt voor resets

SoftwareSerial mySerial(D1, D2); // RX, TX

// constants
#define NUMPIXELS 39
#define KNOPPIXELS 9
#define LEAFPIXELS 3

[... 584 lines cut ...]
  if (WheelPos < 85) {
    return pixels.Color(255 - WheelPos * 3, 0, WheelPos * 3);
  }
  if (WheelPos < 170) {
    WheelPos -= 85;
    return pixels.Color(0, WheelPos * 3, 255 - WheelPos * 3);
  }
  WheelPos -= 170;
  return pixels.Color(WheelPos * 3, 255 - WheelPos * 3, 0);
}

uint32_t WheelValue(byte WheelPos, byte Value) {
  WheelPos = 255 - WheelPos;
  if (WheelPos < 85) {
    return pixels.Color((255 - WheelPos * 3) * Value / 255, 0, (WheelPos * 3) * Value / 255);
  }
  if (WheelPos < 170) {
    WheelPos -= 85;
    return pixels.Color(0, (WheelPos * 3) * Value / 255, (255 - WheelPos * 3) * Value / 255);
  }
  WheelPos -= 170;
  return pixels.Color((WheelPos * 3) * Value / 255, (255 - WheelPos * 3) * Value / 255, 0);
}

```

WereCatf, 2016-10-13:
How is your button connected? You have a pull-down resistor on D3? On the Nodemcu there is a pull-up resistor connected to D3, so by default digitalRead(D3) would return HIGH, and the button connected to it should pull it LOW, but your code seems to assume the reverse.

Imara90, 2016-10-13:
Hi @WereCatf. The button is indeed connected to a pull-down resistor. I thought the pull-up resistor was optional, so I chose to do a pull-down out of habit. However since the code is functional during USB connecting, I don't think that woul be the problem? I did end up finding [this](http://bbs.espressif.com/download/file.php?id=893&mode=view) file after some searching, which implies that the GPIO2 en GPIO0 should not be used. However, didn't find this in the initial [pinout](https://www.google.nl/search?q=nodemcu+pinout&espv=2&biw=1920&bih=1070&tbm=isch&imgil=8EwK33hzjrkNWM%253A%253BhbDmrOrL5qJudM%253Bhttp%25253A%25252F%25252Fwww.cnx-software.com%25252F2015%25252F10%25252F29%25252Fgetting-started-with-nodemcu-board-powered-by-esp8266-wisoc%25252F&source=iu&pf=m&fir=8EwK33hzjrkNWM%253A%252ChbDmrOrL5qJudM%252C_&usg=__rS6Lsiuyzh08FZIHf9h4BYqMuaE%3D&ved=0ahUKEwi6kuHU69fPAhVCWCwKHXIUDIgQyjcIJw&ei=Aob_V7qJMcKwsQHyqLDACA#imgrc=8EwK33hzjrkNWM%3A), which was the pinout I assumed while working with the breadboard. I'll try and see if the pullup will make any difference

WereCatf, 2016-10-13:
You're misunderstanding what I'm saying. There is already a pull-up resistor on the Nodemcu connected to D3. It is there already, it has to be, as GPIO0 controls whether the ESP8266 boots normally from flash or if it boots into flashing-mode. See https://zoetrope.io/tech-blog/esp8266-bootloader-modes-and-gpio-state-startup

Now, that means your pull-down resistor isn't working. If your pull-down resistor worked it'd pull GPIO0 down at boot and your board would boot into flashing-mode, not the normal boot. **You do not want a pull-down on GPIO0, unless you intend to flash a new sketch on the board.**

Imara90, 2016-10-13:
Yep @WereCatf definitely misunderstood that, thanks! So just to get it straight. I'm allowed to use GPIO0  as an input as long as I just use the pull-up provided and change the code to match it?

WereCatf, 2016-10-13:
Technically, you don't have to do anything to use the pull-up as it is there and you can't disable it without physically desoldering it. But yes, you can use it as an input and you have to remember that it's HIGH by default and if you connect a button to it or something the button should pull the pin LOW when pressed.

(You can also use it as output, HIGH or LOW, if you wish; there's no problem there, either way. Just thought to mention, if it was unclear.)

Imara90, 2016-10-13:
Yesssssss @WereCatf you are a lifesaver. This did the trick! There's so many things to keep into consideration with the NodeMCU, it's hard to keep track. Thanks!

closed 2016-10-13 by Imara90

WereCatf, 2016-10-13:
@Imara90 No problem, glad to be of help! :)

Imara90, 2016-10-13:
@WereCatf perhaps one more question as I think it's related to the subject. The Led on D4 (GPIO2) seems to be irresponsive to the code. It functions untill there's a serial communication (in this case with the DFplayer). I think it's because GPIO2 functions as an alternative TX, but I don't want it to. Is there any way to solve this? If not, I'll open a new issue.

WereCatf, 2016-10-14:
You aren't using Serial1 or otherwise enabling UART on D4, so I don't see how that could be the issue. If I had to guess I'd say the culprit is SoftwareSerial. I have never used it, so I am not sure, but you could test it: make a new sketch, move the LED to another pin, define SoftwareSerial on some pins and have some traffic on it -- if the LED becomes unresponsive again then you know it's not the pin, but SoftwareSerial that's the problem.

Imara90, 2016-10-17:
It appears so. It doesn't matter which pin I use, the blinking keeps
happening whenever the sketch sends a command to the DFplayer via Serial.
I've already tried to switch to using the regular Serial, but it has the
same effect. Guess I should post a new issue on why the serial output shows
on the IO pins and how to make sure it doesn't happen.

- Imara

www.imaraspeek.nl