esp8266/Arduino #2606: ESP8266WiFiMulti does not record the credentials of the last connected network

odilonafonso, 2016-10-13:
### Basic Infos

#### Hardware
Hardware:			ESP-12 NodeMCU
Core Version:      	2.3.0-rc2

### Description
I provided the credentials of some WiFi networks usually use with:
`WiFIMulti.addAP (sid, pass)`

And then I used:
`WiFiMulti.run ()`
to connect.

The connection went well and I figured the supplied credentials would be recorded on the chip in the same way that happens when I use:

`WiFi.begin (sid, passw)`

However, when trying to connect me through:

`WiFI.begin ()`
(without credentials)

I noted that previous credentials had not been recorded.

That's right this behavior? Is this how it works?

WereCatf, 2016-10-14:
As far as I know, it's not even supposed to store any other credentials than those supplied to WiFi.begin()

If you wish to store credentials for multiple networks for use with WiFiMulti then you should manually store those in EEPROM and read them at boot yourself from there.

odilonafonso, 2016-10-14:
Thanks @WereCatf 

Well, I'll make a method in ESP8266WiFiMulti to store the credentials, as in WiFi.begin ().

closed 2016-10-21 by odilonafonso