esp8266/Arduino #2608: AP Off

castro2487, 2016-10-14:
i was looking for but i haven't find it yet, it's possible to turn off the Wifi network that i create with the ESP8266? i try WiFi.mode(WIFI_OFF), and doesn't work.

because i use the AP for receive some data the first time that i turn of the hardware and when i finish the configuration process i want to shut down the AP and receive the colllect information VIA internet with a IP address that i have and then to the app that im doing.

MarcusWichelmann, 2016-10-17:
Might it work for your use case to store the settings in the EEPROM when you're finished with the configuration-process via the AP and then reboot the device? After the reboot the code could detect, that the configuration is already done, read it from the EEPROM and does it's stuff without starting the AP again.

copercini, 2016-10-18:
Using git version, mine ESP8266 create a Wifi network too, without I want it.

devyte, 2017-10-11:
@castro2487 If you want to turn off the Wifi, see #644 , you need something like:
  WiFi.disconnect();
  WiFi.mode(WIFI_OFF);
If you want to switch from StationAP to Station, then you need to play further with WiFi.mode().
If you need further help, please refer to a community forum, and/or look at the WiFi source code.
Off-topic, closing per #3655 .

closed 2017-10-11 by devyte