esp8266/Arduino #2610: AP mode gets unstable when SSID for STA can't be found

everslick, 2016-10-16:
when i configure the SSID wrong for the STA (it cannot connect to my accesspoint) the SOFTAP becomes totally unstable (ESP8266 hangs after first http request, no WD no exception, loop() just does not get executed anymore). which is really bad, because it can happen that somebody just switches off the accesspoint and then the ESP8266 cannot be reached and reconfigured anymore, what somehow defeats the purpose of the AP mode.

it also is the case when i reboot thus the ESP never was connected to the accesspoint in the first place.
it 'feels' as the SDK struggles with finding the SSID and gets confused somehow, when i connect to the SOFTAP. but as long as i do not connect to the SOFTAP (via sync webserver), the ESP runs fine.

krzychb, 2016-10-16:
Hi @everslick,
This is "expected" behavior since Soft-AP and STA share the same hardware channel - ref. [Espressif FAQ](https://espressif.com/sites/default/files/documentation/espressif_faq_en.pdf), point _"Why are there network connectivity and packet loss problems in ESP8266 SoftAP +
Station mode?"_

everslick, 2016-10-16:
thanx for the reply, this does indeed clear things up a little bit. i quote the FAQ here for reference and in case somebody has a good idea how to handle this case cleanly and wants to add something to the discussion...

----------

Why are there network connectivity and packet loss problems in ESP8266 SoftAP +
Station mode?

ESP8266 does support SoftAP + Station co-existing mode, however, ESP8266 has only one
hardware channel which is shared between the Station interface and the SoftAP interface. As a
result, ESP8266 SoftAP will dynamically adjust the channel value to be consistent with ESP8266
Station, which may cause the following cases.

Case 1
1. ESP8266 Station is connected to a router (assume the channel number is 6);
2. ESP8266 SoftAP mode is activated using wifi_softap_set_config;
3. The API will return true if configuration is correct, but the channel number will adjust
automatically to be the same as that of the ESP8266 Station interface (which is 6 in this case).

Case 2
1. In this case, wifi_softap_set_config is called and ESP8266 is configured to SoftAP mode
(assume the channel number is 5);
2. Other stations are connected to ESP8266 SoftAP;
3. ESP8266 Station is connected to the router (assume the router channel number is 6);
4. ESP8266 SoftAP will automatically adjust the channel number to be the same as that of the
ESP8266 Station (6 in this case);
5. As the channel changes, the station connected to the ESP8266 SoftAP will lose Wi-Fi
connectivity.

Case 3
1. Other stations are connected to the ESP8266 SoftAP.
2. If ESP8266 Station keeps trying to scan for or connect to a router, ESP8266 SoftAP may lose
connectivity, and UDP or ping packet loss may occur.

This is because ESP8266 Station will scan through all channels to find the target router. Therefore,
ESP8266 SoftAP keeps changing channel as well, leading to connectivity problems.
In such case, users can set the timer and call wifi_station_disconnect on expiry to stop ESP8266
Station from trying to connect to the router. The other way is to call
wifi_station_set_reconnect_policy and wifi_station_set_auto_connect at the beginning of the
configuration to prohibit ESP8266 Station from reconnecting to the router.

closed 2016-10-17 by everslick