esp8266/Arduino #2611: ESP8266WebServer parsing query part of URI

holgerlembke, 2016-10-16:
If I send a 

   http://xxxx/index.html?foo=baa

I get an args() result. If I send a 

   http://xxxx/index.html?foo

the foo is lost, no args(), nothing.

I'm not suure how to solve that. Perhaps add a foo-argname with an empty arg.

holgerlembke, 2016-10-16:
`  int hasSearch = url.indexOf('?');
  if (hasSearch != -1){
    searchStr = url.substring(hasSearch + 1);
	if (searchStr.indexOf('=')==-1) {
		searchStr="query="+searchStr;
	}
    url = url.substring(0, hasSearch);
  }
  _currentUri = url;
`

devyte, 2017-10-11:
@holgerlembke You are right, but I know that 
http://xxxx/index.html?foo=
works.
A quick google shows that the relevant RFC is not specific on this matter, meaning that it is implementation defined. ASP seems to choke on what you propose, but other parsers accept it.
I think this should be fixed, makes for better compatibility.

holgerlembke, 2017-10-12:
For concept compatibility it might be an approach to make it like php.
```
$_SERVER["argv"] = Array
(
    [0] => foo
)
```