esp8266/Arduino #2613: Repository for esp8266 library broken?

henrybcn, 2016-10-17:
I am trying to install the esp8266 library on the arduino ide and seems they are not available. Is the link broken?

I am following the instructions on the main page, but can not even see the library to install:
Enter http://arduino.esp8266.com/stable/package_esp8266com_index.json into Additional Board Manager URLs field. You can add multiple URLs, separating them with commas.

Is there any alternative to get it done?
Rgds,
henry

WereCatf, 2016-10-18:
Works just fine for me.

1-prop-head, 2016-10-20:
If you past the link into the preferences you may not get the required carriage return line feed characters and this will cause the repository not to load properly.  Open the explore icon beside the text input box on the preferences and add a carriage return.  This will repair the issue and allow repository to load

devyte, 2017-10-11:
Can't reproduce, board manager is currently working. Closing.

closed 2017-10-11 by devyte