esp8266/Arduino #2617: macro attributes should be used with brackets

bitluni, 2016-10-18:
Low-level macros are vulnerable to operator priorities which can cause problems. Took me some time to find why my code didn't work

Arduino/tools/sdk/include/eagle_soc.h rows:
#define GPIO_REG_READ(reg)                         READ_PERI_REG(PERIPHS_GPIO_BASEADDR + reg)
#define GPIO_REG_WRITE(reg, val)                 WRITE_PERI_REG(PERIPHS_GPIO_BASEADDR + reg, val)

should bechanged to:
#define GPIO_REG_READ(reg)                         READ_PERI_REG(PERIPHS_GPIO_BASEADDR + (reg))
#define GPIO_REG_WRITE(reg, val)                 WRITE_PERI_REG(PERIPHS_GPIO_BASEADDR + (reg), val)

devyte, 2017-10-11:
@bitluni I still see this in the latest code. Care to make a PR?

devyte, 2018-08-01:
Correcting milestone to 2.4.2

closed 2018-08-01 by devyte