esp8266/Arduino #2618: Why can't config file size be > 1024 bytes?

helpme1, 2016-10-19:
I am using ConfigFile.ino as a reference to store configuration settings.

https://github.com/esp8266/Arduino/blob/master/libraries/esp8266/examples/ConfigFile/ConfigFile.ino

I noticed this code.
```
size_t size = configFile.size();
  if (size > 1024) {
    Serial.println("Config file size is too large");
    return false;
  }
```
What is the reason configFile cannot be >1024? I have several config settings and I worry that the configFile size will exceed 1024. If this is a limitation, how can one work around this limitation?

zgoda, 2016-10-19:
This is just example, did you try to load/store larger json files?

devyte, 2017-10-11:
@helpme1 most likely because the entire file is read directly into a String, and you need a sanity check for that (limited RAM on the ESP).
Closing as off-topic, see #3655 .

closed 2017-10-11 by devyte