esp8266/Arduino #2620: spiffs - every file allocates 502 bytes memory even if the file is 8 bytes long

diffstorm, 2016-10-19:
Hi All,

I'm using es8266 arduino with spiffs (FS.h) (1M Flash with 256K SPIFFS) and I noticed that every file allocates 502 bytes memory even if the file is 8 bytes long.
I have 8 bytes long files but every single file covers 502 bytes.

How to reduce that offset or configure for the best?

> 2 files -> "usedBytes":"1004"
> 4 files -> "usedBytes":"2008"
> 6 files -> "usedBytes":"3012"
> 8 files -> "usedBytes":"4016"
> 10 files->"usedBytes":"5020"

The "usedBytes" comes from;
```
FSInfo fs_info;
SPIFFS.info(fs_info);
```

The other parameters of FSInfo with 10 files in FS is like below:
`{"totalBytes":"233681","usedBytes":"5020","blockSize":"4096","pageSize":"256","maxOpenFiles":"5","maxPathLength":"32","FreeSketchSpace":"458752"}`

igrr, 2016-10-19:
According to spiffs documentation, each file uses one page for index, and one or more pages for data. As you can see from output, page size is 256 bytes, so 512 bytes is the expected usage for small files.

diffstorm, 2016-10-19:
Thanks for fast answer. I guess there is nothing we can do unless change or improve the spiffs itself.

closed 2016-10-19 by diffstorm