esp8266/Arduino #2621: Using ESP8266_new_pwm to improve performance

diffstorm, 2016-10-19:
Hi,

Can you add the following library to improve pwm performance?
> https://github.com/StefanBruens/ESP8266_new_pwm

**Problem:**
I'm using 3 pwm outputs to drive an rgb led and sometimes I see kind of flickering (leds goes off/black then back to its color momentarily).

Init code:
```
  pinMode(PINR, OUTPUT);
  pinMode(PING, OUTPUT);
  pinMode(PINB, OUTPUT);
  digitalWrite(PINR, 0);
  digitalWrite(PING, 0);
  digitalWrite(PINB, 0);
  analogWriteRange(255);
  analogWriteFreq(100);
  analogWrite(PINR, 0);
  analogWrite(PING, 0);
  analogWrite(PINB, 0);
```

Drive code:
```
void ICACHE_RAM_ATTR SetRGB(int rr, int rg, int rb)
{
  os_intr_lock();
  analogWrite(PINR, rr);
  analogWrite(PING, rg);
  analogWrite(PINB, rb);
  os_intr_unlock();
  yield();
}
```

> Flash: QIO - 40MHz - 1M (256K spiffs)
> CPU: 160MHz

WereCatf, 2016-10-19:
Have you actually checked if it is any faster or not? That code is a replacement for the SDK PWM-code, but the ESP8266 Arduino-core doesn't use the SDK PWM-code.

diffstorm, 2016-10-19:
I didn't know that, I didn't test it. It claims that is better than SDK PWM.
So what can I do for the problem, please write of you have suggestions. I use only 3 pins for PWM.
Thanks

diffstorm, 2016-10-24:
Hi, I have tried the ESP8266_new_pwm on my system and indeed it works better. I don't see any flickering on LEDs.

WereCatf, 2016-10-24:
That's good to know. I'll give it a try myself, too, as soon as I have the time.

diffstorm, 2016-10-24:
You can get help by reviewing the link below. I have already asked about implementation and other useful questions;

https://github.com/StefanBruens/ESP8266_new_pwm/issues/5#issuecomment-254934514

jrobeson, 2016-10-31:
wouldn't it be better to get it included in the SDK if it is better? or do they not want it?

WereCatf, 2016-10-31:
@jrobeson The ESP8266 SDK isn't open-source, unlike the ESP32 SDK. You can, of course, ask them if they would like to improve on their PWM-functions, but it doesn't seem likely. Also, like I said, this Arduino-core for the ESP8266 doesn't use SDK-functions for PWM, so it'd have to be modified anyways.

jrobeson, 2016-10-31:
why doesn't it seem likely?

WereCatf, 2016-10-31:
@jrobeson Because they've been told about it several times. And their SDK has been sitting idle since March.

devyte, 2017-10-11:
@WereCatf did you ever get around to testing that new pwm?

devyte, 2017-10-11:
@diffstorm does the new pwm still have merit vs. the current pwm implementation?

WereCatf, 2017-10-11:
@devyte No, I completely forgot about this whole discussion. For my own purposes, the PWM-mechanism that's already included has worked fine, but then again, I haven't used it for anything demanding.

diffstorm, 2017-10-11:
@devyte I think current PWM implementation works better than the one in the link.
BTW, I still see flickering when brightness is lower than 100% in rainbow mode. I couldn't fix that issue, that leads me to think its a HW or SDK problem.

WereCatf, 2017-10-11:
To be honest, to properly gauge if it is better, worse or equal in performance, someone would have to capture multiple proper oscilloscope-output at different frequencies and duty-cycles on both implementations, and no, a single capture at a single frequency ain't enough. Until someone provides such proof one way or another there's no point in debating this topic.

closed 2017-10-11 by diffstorm