esp8266/Arduino #2623: Exception 29 on iPhone

Hassebas, 2016-10-20:
I am building a client-server app with the 'ESP8266 Thing Development Board'.  The software stems from this example: [Access Point (AP) Web Server.](https://learn.sparkfun.com/tutorials/esp8266-thing-development-board-hookup-guide/example-sketch-web-server)
In Safari it runs without problems on the iPhone.  Even if the refresh rate of the page is set to 2 seconds (which i need for the final application) 

However, if you add the webpage as an app to the homescreen and activate it, the app loses in a few minutes its connection with the server and the ESP8266 crashes. If the refresh rate is 15 sec or higher there is no problem.

This is the error code:

Exception (29):
epc1=0x4000e1c3 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000018 depc=0x00000000

ctx: sys 
sp: 3ffffc80 end: 3fffffb0 offset: 01a0

>>>stack>>>
3ffffe20:  40102ca1 3ffeff00 3fff0690 401043a6  
3ffffe30:  402125c9 40212506 3fff38a4 40102e30  
3ffffe40:  00000018 3fff0950 3fff08cc 00000000  
3ffffe50:  40212664 3ff20a00 3fff38a4 3ffeda8e  
3ffffe60:  3ffeda74 3ffecda8 3ffecda8 3ffeda74  
3ffffe70:  3ffeda8e 0000002c 00000004 401004f4  
3ffffe80:  3ffeda7e 3fff3c64 4020f3af 3ff20a00  
3ffffe90:  3fff38a4 3fff60cc 00000015 4020f281  
3ffffea0:  3ffecdb4 00000001 00000000 3fffdab0  
3ffffeb0:  40000f83 00000030 0000000d ffffffff  
3ffffec0:  40000f3d 00000023 00000000 3fffd9d0  
3ffffed0:  00000000 00000000 00000000 fffffffe  
3ffffee0:  ffffffff 3fffc6fc 00000001 3fffdab0  
3ffffef0:  00000000 3fffdad0 3fff1f08 00000030  
3fffff00:  401067a0 00000000 00000013 401004d8  
3fffff10:  4021681d ffffffb3 00000000 40107718  
3fffff20:  4020eb19 3ffeda74 3ffeff00 3fff38a4  
3fffff30:  00000000 3ffeda7e 3ffeda98 401067ba  
3fffff40:  40209f6c 00000000 3ffeda7e 3ffeda74  
3fffff50:  0000002e 00000000 3fff3c64 3ffeda68  
3fffff60:  00000000 00000013 40209b9e 3ffeff00  
3fffff70:  3ffeda68 3fffdcc0 3ffec5f8 3fff3b4c  
3fffff80:  3ffeda84 00000040 00000000 3fff1f10  
3fffff90:  402094e7 3fffdab0 00000000 4020744f  
3fffffa0:  3ffec5f8 40000f49 3fffdab0 40000f49  

Decoding 23 results
0x40102ca1: trc_NeedRTS at ?? line ?
0x401043a6: lmacProcessTxSuccess at ?? line ?
0x402125c9: ieee80211_alloc_proberesp at ?? line ?
0x40212506: ieee80211_alloc_proberesp at ?? line ?
0x40102e30: trc_NeedRTS at ?? line ?
0x40212664: ieee80211_send_proberesp at ?? line ?
0x401004f4: calloc at C:\Users\Harry\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266\umm_malloc/umm_malloc.c line 1684
0x4020f3af: hostap_input at ?? line ?
0x4020f281: hostap_input at ?? line ?
0x401067a0: igmp_timer at /Users/igrokhotkov/espressif/arduino/tools/sdk/lwip/src/core/timers.c line 217
0x401004d8: malloc at C:\Users\Harry\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266\umm_malloc/umm_malloc.c line 1664
0x4021681d: cnx_node_search at ?? line ?
0x40107718: pvPortMalloc at C:\Users\Harry\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/heap.c line 13
0x4020eb19: hostap_input at ?? line ?
0x401067ba: igmp_timer at /Users/igrokhotkov/espressif/arduino/tools/sdk/lwip/src/core/timers.c line 222
0x40209f6c: DefFreqCalTimerCB at ?? line ?
0x40209b9e: pp_tx_idle_timeout at ?? line ?
0x402094e7: ppPeocessRxPktHdr at ?? line ?
0x4020744f: loop_task at C:\Users\Harry\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_main.cpp line 56

Any idea how to fix this or have a work around?
.

devyte, 2017-10-11:
@Hassebas please try:
- latest git
- PR #3215  (SDK 2.1.0)
- PR #3362 (lwip2)

devyte, 2017-12-05:
Closing due to no feedback in almost 2 months, and due to age.

closed 2017-12-05 by devyte