esp8266/Arduino #2625: httpUpdateSPIFFS - how to get and save SPIFFS

odilonafonso, 2016-10-21:
### Basic Infos

#### Hardware
Hardware:		ESP-12 NodeMCU / Wemos D1
Core Version:      	2.3.0-rc2

### Description

httpUpdate is working well; now I want to update the SPIFFS too.
I have read the httpUpdateSPIFFS from the samples and look simple. But I do not know how to load and save the "spiffs.bin" file.

In the IDE we have:
`Tools -> "ESP8266 Sketch Data Upload"` - to load data from a file to the SPIFFS
and 
`Sketch -> Export Compiled Binary `- to save the sketch bin file for upload to my web server

I do not know if there is a tool to copy the contents of my spiffs ESP or I'll have to do it myself.
It is also unclear if copying only the data recorded on spiffs or if I provide a "spiffs.bin" with the total size of the available area for spiffs on my ESP.

Someone please know how to do this?
@krzychb, you being an expert in OTA, you know how to do this?
Thanks in advance !!

WereCatf, 2016-10-21:
There is a command-line tool called "mkspiffs" that is used to make those SPIFFS-images. I have never manually used it, so I can't give much more info on that, but call it and check what arguments it supports. The resulting image will have to be the size of your whole SPIFFS, not just the size of the files on it, so e.g. on a Nodemcu with 1MB/3MB split you'll end up with a 3MB SPIFFS-image.

odilonafonso, 2016-10-21:
Hi @WereCatf, thank you !!

Yes, I gave it a read and seems to be the right tool to use !!

I'll test it and post the results here.

odilonafonso, 2016-10-21:
Unfortunately I could not understand the workings of "mkspiffs" to make a copy of my ESP SPIFFS área to a file.

Please @igrr, 

It is possible to use "mkspiffs" to load an SPIFFS image of my ESP in a file?

For a next step using `httpUpdateSPIFFS ()` function from `ESP8266HttpUpdate` library?

I read mkspiffs help, but don't know how to make this.

Thank you for help

WereCatf, 2016-10-22:
To create a 3MB SPIFFS-image you'd use the following:
`mkspiffs -c directory_whose_contents_you_want_in_the_SPIFFS_image -b 8192 -s 3125248 image.spiffs`

odilonafonso, 2016-10-22:
Ahhhh OK @WereCatf.

But that's not what I want to do.

I would like to access the ESP and bring its SPIFFS contents in a directory. 

Like: `Sketch --> Export compiled Binary.`

Nevertheless it may be useful; I can for example create some other way a content equal to my ESP and (assuming I do it in the data directory of my sketch):

`mkspiffs -c date -b -s 8192 3125248 image.spiffs`

OK, thank you very much for the help!

WereCatf, 2016-10-22:
OTA-updates and httpUpdateSPIFFS are for sending content to the ESP, not for receiving content from it. If you want to receive the SPIFFS-filesystem that is already on your ESP you will have to write the code all by yourself.

odilonafonso, 2016-10-22:
OK

closed 2016-10-22 by odilonafonso

tonu42, 2016-11-11:
It is an interesting proposition to read the files back, as that could be used to get recorded data back, 3 MB is a lot a of data depending on frequency of data recording.

If you know the file names you can read them back by serving them up to an HTTPServer. The esp is not like a computer where you can have files existing unless you actually made them so I think it would be feasible. Could even make it a form to get back only what you want.

mrhurless, 2016-12-09:
@WereCatf  said:
> To create a 3MB SPIFFS-image you'd use the following:
mkspiffs -c directory_whose_contents_you_want_in_the_SPIFFS_image -b 8192 -s 3125248 image.spiffs

I was curious if that is the actual file you would point the .updateSpiffs() to, or does it specifically need to be a .bin file (output of mkspiffs is .image)?  Thanks in advance for any help!

HugoML, 2017-10-02:
I have the same puzzle. 
mkspiffs creats image.spiffs, but how to create the .bin file? 
I couldn't find the verbose mode line showing the process command. Please advice.
And, how to calculate the block size? 
8192 = 8*1024 byte  (can we change this to smaller size , such as 2K = 2048 bytes?)
3125248 != 3*1024*1024 byte (3MB) . Why is this?

captainreptile, 2017-12-08:
@HugoML did you figure it out how to make the bin file?

mrfaptastic, 2018-05-13:
I have never got Igor's mkspiffs tool to work properly, even with the correct parameters, the bin file when used with the ESPhttpUpdate.updateSpiffs function has resulted in a garbage SPIFFS file being loaded and the ESP then re-creating a blank file system.

A much simpler workaround is to simply use Arduino on Windows.

When you run the ESP8266 Sketch Data Upload tool, you will quickly see in the Arudino terminal output, something along the lines of:

/file.html

[SPIFFS] upload : C:\Users\USER\AppData\Local\Temp\build87a71177855ef0141b9b3758467a3f0f.spiffs/CryptoTicker_v1.8.spiffs.bin
[SPIFFS] reset  : nodemcu
[SPIFFS] port   : COM5
[SPIFFS] speed  : 921600
[SPIFFS] address: 0x300000

Take note of that file path and go immediately to it to extract a .bin file that works!

HugoML, 2018-05-14:
@mrfaptastic This works quite well. 
@captainreptile Command line I successfully tried: 
For nodeMCU with 4M flash (configured 3M File System)
esptool –cd nodemcu –cb 115200 –cp COM24 –ca 0x100000 –cf  yourArduino.spiffs.bin
For generic ESP8266 with 4M flash: (configured 3M File System)
esptool –cd ck	–cb 115200 –cp COM24 –ca 0x100000 –cf  yourArduino.spiffs.bin
Note:  0x100000 is the starting address of your file system. Depends on how you allocate it, the address is different.

HugoML, 2018-05-14:
And, according to document on file system, there is a reserved sector for EEPROM (4KB). So the calculation for SPIFFS starting address is better to be observed in Arduino output window.

ElectricBeat, 2022-11-21:
@captainreptile Did the actual file you would point the .updateSpiffs() and add/upload SPIFFS file system ?